whoenig/crazyflie_ros #79: Sending a new packet type to commander_generic

Malintha, 2017-12-09:
Hi,

How can we set the packet type value in the setpoint request? I am trying to modify the code in order to send a new packet type as mentioned in [cf firmware](https://github.com/bitcraze/crazyflie-firmware/blob/c55ed40b05b370e73ae85c957636aac78c87f20a/src/modules/src/crtp_commander_generic.c#L52).  and just couldn't figure out the place you have set the packet type.

Issue for the new commander_generic implementation
https://github.com/bitcraze/crazyflie-firmware/issues/113

whoenig, 2017-12-09:
There is some generic packet support, but it would be better to add native support:
1) add the packet type here: https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_cpp/include/crazyflie_cpp/crtp.h#L187-L206. 
2) add a function (could be overloaded): https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_cpp/src/Crazyflie.cpp#L118-L126
3) add ROS support in https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_driver/src/crazyflie_server.cpp

I'll do that eventually, but PRs are welcome!

Malintha, 2017-12-09:
Hi @whoenig,

More than happy to contribute. Some questions though;

1) If we are to implement a packet format that complies with the new cf generic packet format, I feel like all we have to do is to add another header param to the packet type. (Referring to this diagram - https://github.com/bitcraze/crazyflie-firmware/issues/113#issuecomment-234930509)

2) I didn't quite get the idea of 'native support'. Do we have to be more native to the ROS controller side or to the cf side?

Malintha, 2017-12-09:
@whoenig,

AFAIU, the current ROS cf packet calls to the port 0x03 and channel 0. But in order to comply with the new generic packetDecoder approach, we have to call channel 0x07. (as per here: https://github.com/bitcraze/crazyflie-firmware/blob/c55ed40b05b370e73ae85c957636aac78c87f20a/src/modules/src/crtp_commander.c#L58 ) 

So is it correct to say we need to overload the crtp [constructer](https://github.com/whoenig/crazyflie_ros/blob/e2ade2a3cbd1964a9f9618cc9e93dca07007b1ad/crazyflie_cpp/include/crazyflie_cpp/crtp.h#L8) if we are to add a new packet type with the new header values?

whoenig, 2018-04-18:
Most generic setpoints are implemented now (e.g., `cmd_hover`, `cmd_position`, and `cmd_full_state`).

closed 2018-04-18 by whoenig