whoenig/crazyflie_ros #81: communication problem

miaoyanan, 2018-01-12:
Hi,@whoenig 
When I use the teleop_xbox360.launch,it can communicate and fly normally.when I use optitrack.launch(which is the similar with the hover_vicon.launch ),everything shows normally,compiled correctly, and the command window shows waiting for takeoff service, but the radio's indicator is not bright, and cf can not communicate.I can also get data by  rostopic echo /Robot_1/pose.
That's all problem.Can you give the solution?Thank you in advance!

whoenig, 2018-01-14:
Where did `optitrack.launch` come from? This is not part of crazyflie_ros. For Optitrack, you should use `hover_vrpn.launch`. There is a tutorial available as well:  "Flying Multiple UAVs Using ROS", Chapter in Robot Operating System (ROS): The Complete Reference (Volume 2), Springer, 2017; you can find a pre-print free of charge at http://act.usc.edu/publications.html.

Also, if you are interested in flying multiple CFs, consider the crazyswarm project instead (see http://crazyswarm.readthedocs.io/en/latest/ for details).

miaoyanan, 2018-01-19:
Hello，@whoenig
Crazy flie can communicate normally, but when I point the handle on the takeoff operation, the propeller will spin, but it will not take off. And then again take off operation, the communication is still normal, but crazyflie did not respond. I recorded a video,and the content of ROS operation window ，but I don't know how to send them to you on this platform.I hope you can help me solve the problem, thank you in advance!

whoenig, 2018-01-19:
You could put it on youtube/google drive etc. and just put a link here.

s3xyxavy, 2018-02-11:
Hi @miaoyanan, 
I am also working on Optitrack with @whoenig crazyflie ros stack, you might want to try changing the mocap_optitrack file on mocap.yaml, from "Robot_1/base_link" to "crazyflie". As whoenig files respond to keyword "crazyflie". You will achieve view_frames as world>crazyflie>crazyflie_baselink. 
However, for me it will only take off but not hover. Lets see if we can share of what results we have. 
I have realized my orientation.y is inverted, but I have no idea how to swap it in the mocap file! Maybe you might have some insights. 
You should try checking the quarternions once you try changing the "crazyflie" on mocap.yaml and viewing on RVIZ graph whilst adding "TF" in the RVIZ GUI. You will see /world and crazyflie.

miaoyanan, 2018-03-05:
Hi @s3xyxavy 
You can get the answer from the mocap_datapackets.cpp in the file(mocap_optitrack-master). Hope that will help you.

s3xyxavy, 2018-03-05:
Hi @miaoyanan
I get that I can find it in that cpp file, but I do not know how to edit to add a negative sign to the orientation.