whoenig/crazyflie_ros #82: ImportError: dynamic module does not define module export function (PyInit__tf2)

theplaystar, 2018-01-18:
Hello,
I'm trying to launch hover_vicon.launch in a laptop with ubuntu 16.04 and an error occurs.
It could be launched in a raspberry Pi3 with Ubuntu Mate.
I'd like to know what could be the problem and how can I fix it.
I'm a beginner in ros.
Thanks.

whoenig, 2018-01-18:
What is the full error you are getting?

theplaystar, 2018-01-19:
Here is the error message

--
Traceback (most recent call last):
  File "/home/splinter/catkin_ws/src/crazyflie_ros/crazyflie_demo/scripts/publish_pose.py", line 4, in <module>
    import tf
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf/__init__.py", line 28, in <module>
    from tf2_ros import TransformException as Exception, ConnectivityException, LookupException, ExtrapolationException
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf2_ros/__init__.py", line 38, in <module>
    from tf2_py import *
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf2_py/__init__.py", line 38, in <module>
    from ._tf2 import *
ImportError: dynamic module does not define module export function (PyInit__tf2)
[crazyflie/crazyflie_add-3] process has finished cleanly
log file: /home/splinter/.ros/log/372e02ba-fcf2-11e7-93db-3ca9f4038250/crazyflie-crazyflie_add-3*.log
[crazyflie/pose-7] process has died [pid 3286, exit code 1, cmd /home/splinter/catkin_ws/src/crazyflie_ros/crazyflie_demo/scripts/publish_pose.py __name:=pose __log:=/home/splinter/.ros/log/372e02ba-fcf2-11e7-93db-3ca9f4038250/crazyflie-pose-7.log].
log file: /home/splinter/.ros/log/372e02ba-fcf2-11e7-93db-3ca9f4038250/crazyflie-pose-7*.log
--

whoenig, 2018-01-19:
This sounds like a generic ROS problem. Perhaps you somehow switched the default python interpreter to be python3 rather than python2 (check using `python --version`). ROS Kinetic requires python2 (see e.g., https://github.com/ros/geometry2/issues/259).