whoenig/crazyflie_ros #83: std::runtime_error when running roslaunch crazyflie_demo teleop_xbox360.launch

szx0112, 2018-01-22:
My CF is flashed with newest firmware with cfclient. And I can connect CF and crazyradio in the client.

When I do rosrun crazyflie_tools scan
I get LIBUSB_ERROR_TIMEOUT

[image]

When I do roslaunch crazyflie_demo teleop_xbox360.launch
I got following error:
[image]

My computer is ubuntu 14.04.5 with ros indigo 1.11.21. I have successfully implemented the CF and LPS with firmware in 2017.5. I updated the firmware of CF and LPS several days before, and then cannot connect through ROS. I think the problem may relate to ROS cannot connect CF with crazyradio for some reason, but have no clue to solve it.

I am a little bit of confusing if I need to update crazyradio as I think cfclient only updates the stm32 and nrf of CF. But since they can connect through client, the firmware should be good. Any comment is appreciate.

whoenig, 2018-01-22:
Did it work before on that machine using ROS, but stopped working when you updated the firmware? Or is it the first time you try the ROS interface? You should not need to update your Crazyradio.

The first error (using scan) indicates that there is some issue with libUSB. You can try removing the exception (perhaps from here: https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_cpp/src/Crazyradio.cpp#L173) and see if that gets you farther.

The second issue usually means that the uri of the CF is wrong; however, if the scan tool doesn't work it might also indicate that there is some issue communicating with the radio.

szx0112, 2018-01-23:
Thanks for the quick reply. Yes, I have been used the CF, ROS and LPS to do test of the path planning algorithm since last summer, and it worked really good! Recently, I noticed the update of LPS and try to re-flash the firmwares, then I found I cannot connect with ROS anymore. I reinstalled clients and tried different versions of crazyflie firmwares to flash, no good results show till now.    

[image]

I comment out the exception line and found the same error code shown (as above) in command line. The crazyflie_tools scan functions sometime can show the result without error, but still return errors if I call roslaunch.

[image]

whoenig, 2018-01-23:
When you use roslaunch, you attempt to connect to a CF with uri `radio://0/80/2M`, but the scanning tool finds the CF at `radio://0/80/250K`. Perhaps that is your issue?

I am not sure what's causing the trouble with the scanning tool exception. Are you using USB 2.0 or USB 3.0?

szx0112, 2018-01-23:
I think because I clone the repository this morning and forget to change it. But the error still showing after I made the change. 

[image]

The USB are 2.0s but those slots work fine in previous months. I uninstalled the cflib recently as I am not programming the client, but does it affect the ros connection?

whoenig, 2018-01-23:
The ROS interface is independent of cflib. Both directly work on top of libUSB. You can try getting rid of https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_cpp/src/Crazyradio.cpp#L154 as well and see if that helps. USB 2.0 is certainly less troublesome than 3.0.

I haven't seen this issue myself, but it could be an issue in libUSB (I usually test with 16.04 that comes with a slightly newer version of libUSB).

szx0112, 2018-01-23:
By adding some std::count in Crazyradio.cpp as below, I found it actually get several good connections and then the errors occurred. And these errors look like are not from this file. 
[image]

[image]

Not sure if this help for locating the problem. I did some updates, version change of libusb but looks not helpful.

whoenig, 2018-01-23:
But the error message changed now: before it said LIBUSB_ERROR_TIMEOUT, now it just says timeout. The lower case message indicates that data was sent, but no acknowledgement received. Please double check that the URI you specified is indeed the one you are using with the official python client (since the scan tool actually found two different ones). Also, do the LEDs on the CF indicate that it receives packets during the connection attempt?

szx0112, 2018-01-23:
I add the uri same as I connected through client when the command line as:
roslaunch crazyflie_demo teleop_xbox360.launch uri:=radio://0/80/250K
[image]
same timeout error throws

The green and red LED on crazyflie and crazyradio when I call roslaunch:
[image]
The only difference for the scan is the crazyradio only shows Red LED solid on for a second when no LIB_USB_ERROR throws. 

Everytime I replug in crazyradio I found the error message is timeout, after a while it turns to LIBUSB_USB_ERROR (after several roslaunch/rosrun tests).

It looks like under any situation, the crazyflie does receive packets (Green LED always OFF) when connect through ROS.

whoenig, 2018-01-24:
I noticed that the cfclient just ignores all LIBUSB errors (see https://github.com/bitcraze/crazyflie-lib-python/blob/master/cflib/drivers/crazyradio.py#L282-L283), which might or might not explain some of the behavior difference.
I would like to see if I can reproduce the issue you are having - which exact firmware version are you testing? Also, if you use the official client, could you print out which pyusb version is active on your system (see https://github.com/bitcraze/crazyflie-lib-python/blob/master/cflib/drivers/crazyradio.py#L65)?

szx0112, 2018-01-25:
I think I test the recent three versions (2017.5, 2017.6 and 2018.1) in [https://github.com/bitcraze/crazyflie-firmware/releases](https://github.com/bitcraze/crazyflie-firmware/releases) and update CF by bootloader through the most recent cfclient in [https://github.com/bitcraze/crazyflie-clients-python](https://github.com/bitcraze/crazyflie-clients-python). I also used update the crazyflie_ros by either git pull or git clone. 

I also test it on ubuntu 14.04 and 16.04 on the same computer and both got same error information. 

I tried to flash back the last work version which is 2017.5 I believe but looks like I cannot find the corresponding crazyflie_ros cloned at that time. 

The pyusb version in my system is 1.0.2.  I get it by: pip show pyusb. 

Not sure if it is a typical problem or just a special case. Let me know if you need any information.

szx0112, 2018-01-26:
I got the problem solved by reinstall everything on ubuntu 16.04. Now it works as usual. I believe it related to the USB setup but I am not an expert at OS. Thanks for the help.

closed 2018-01-26 by szx0112