whoenig/crazyflie_ros #84: teleop xbox360 launch issue

rayaan92, 2018-01-23:
Good day,

when running the teleop xbox360 launch, and I try to take off with the drone the trust keeps getting cut. I echoed the joy messages and they are being published constantly yet the drone keeps crashing due to the trust just being cut. 

Thanks in advance.

whoenig, 2018-01-23:
That sounds like some kind of bad connection. Does it work with the Bitcraze Python Client? Do you get any warnings about high package drop rates?

rayaan92, 2018-01-24:
it works perfectly with the cfclient, I managed to fix it by changing some stuff in your code. I will continue to test and provide feedback.

there was also no warnings of packet drops

whoenig, 2018-01-24:
It would be good to know what you had to change to make it work?

rayaan92, 2018-01-25:
void cmdVelChanged(
    const geometry_msgs::Twist::ConstPtr& msg)
  {
    if (!m_isEmergency) {
      roll = msg->linear.y + m_roll_trim;
      pitch = - (msg->linear.x + m_pitch_trim);
      yawrate = msg->angular.z;
      thrust = std::min<uint16_t>(std::max<float>(msg->linear.z, 0.0), 60000);

      // m_cf.sendSetpoint(roll, pitch, yawrate, thrust); // removed it here
      // m_sentSetpoint = true;
    }
  }

    ROS_INFO("Ready...");
    auto end = std::chrono::system_clock::now();
    std::chrono::duration<double> elapsedSeconds = end-start;
    ROS_INFO("Elapsed: %f s", elapsedSeconds.count());

    // Send 0 thrust initially for thrust-lock
    for (int i = 0; i < 100; ++i) {
       m_cf.sendSetpoint(0, 0, 0, 0);
    }
---------------------------------------------------------------------------------------------------------------------------------------------
    while(!m_isEmergency) {
      // make sure we ping often enough to stream data out
      if (m_enableLogging && !m_sentSetpoint && !m_sentExternalPosition) {
        m_cf.transmitPackets();
        m_cf.sendPing();
        if(m_enable_logging_packets) {
          this->publishPackets();
        }
      }
      m_sentExternalPosition = false;
      m_cf.sendSetpoint(roll, pitch, yawrate, thrust); // added the send command here
      m_sentSetpoint = true;                                      // and it did the trick 
      std::this_thread::sleep_for(std::chrono::milliseconds(1));
    }

-------------------------------------------------------------------------------------------------------------------------------------

I also set roll, pitch, yawrate and thrust as global variables.

whoenig, 2018-01-26:
Thanks! I was able to reproduce the issue. I fixed it slightly differently by using a constant cmd_vel update rate. In your fix, it wouldn't be easily possible to switch to other control modes (such as trajectory following).

closed 2018-01-26 by whoenig

rayaan92, 2018-01-26:
I am working on a hovering mode using the realsense camera, something
similar to what you did with the vicon system when I'm done I'll send you
the code, if you interested