whoenig/crazyflie_ros #85: Optitrack, on mocap_optitrack with Crazyflie 2.0

s3xyxavy, 2018-01-29:
Hi whoenig, 

I am currently working on the hover launch using mocap_optitrack. After creating the folder with mocap, in it, I am able to run the hover_vrpn.launch changing to mocap files accordingly. I have also been keep close check on issue 15 and issue 17. I am able to visuallize the tracking of crazyflie on the RVIZ but the problem comes in when I try to execute the taking off. Please do note that I am pretty new in ROS and if possible guide me through any tutorials I should take a look at. The files, view_frames, the error log file and the launch file, relevant are attached in the Dropbox link: https://www.dropbox.com/sh/uf3ryxdbvlfma03/AAAlkzkckq6dVgnsOyRNtiaoa?dl=0 
I hope you can assist me in it, please let me know if You still require any more log or issuing of commands. 
Thank you very much

Warm regards, 
Xavier

whoenig, 2018-01-29:
According to your view frames result, the mocap is not capturing anything. 

* "I am able to visuallize the tracking of crazyflie on the RVIZ" Does that mean you can manually move the crazyflie and its updated position is shown in rviz? If that's the case, the view_frames output must have been created during a different experiment.
* I personally don't know much about `mocap_optitrack`. If you intend to use VRPN, you won't need that package - the Motive software has a VRPN server built-in. If you want to use `mocap_optitrack`, make sure that it works correctly without using crazyflie_ros first by looking at the tf.

There is a tutorial style book chapter for crazyflie_ros which has some more details about VRPN:  "Flying Multiple UAVs Using ROS", Chapter in Robot Operating System (ROS): The Complete Reference (Volume 2), Springer, 2017. A free preprint is available at act.usc.edu. 

Finally, there is the crazyswarm project (see http://crazyswarm.readthedocs.io/en/latest/), if you are interested in flying many CFs. The OptiTrack integration is different there.

s3xyxavy, 2018-01-30:
Hi whoenig, 

Thanks for the reply. 

1. I am able to receive tf data from mocap_node with coordinates and quaternions real time which in that case automatically transform the optitrack left hand coordinate system to ROS standards of right hand coordinate system. This reflects ROS standards coordinate when I rostopic echo /tf .
When manually moving the crazyflie with reflector balls, the RVIZ w/ TF is able to see world and crazyflie. 
Problem comes in when I execute the take off, it gives me the error: 
[ WARN] [1517232696.592782947]: Link Quality low (0.650000)
[ WARN] [1517232698.593181248]: Link Quality low (0.590000)
[ INFO] [1517232698.637396554]: Takeoff requested!
[ERROR] [1517232698.637669841]: Exception thrown while processing service call: Could not find a connection between 'world' and 'crazyflie' because they are not part of the same tree.Tf has two or more unconnected trees.
[ERROR] [WallTime: 1517232698.639493] bad callback: <bound method Controller._joyChanged of <__main__.Controller instance at 0x7fef907fdb48>>
Traceback (most recent call last):
File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/topics.py", line 720, in _invoke_callback
cb(msg)
File "/home/xavier/fypcrazyflie_ws/src/crazyflie_ros/crazyflie_demo/scripts/controller.py", line 46, in _joyChanged
self._takeoff()
File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__
return self.call(*args, **kwds)
File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 515, in call
responses = transport.receive_once()
File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 696, in receive_once
p.read_messages(b, msg_queue, sock)
File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 353, in read_messages
self._read_ok_byte(b, sock)
File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 336, in _read_ok_byte
raise ServiceException("service [%s] responded with an error: %s"%(self.resolved_name, str))
ServiceException: service [/crazyflie/takeoff] responded with an error: Could not find a connection between 'world' and 'crazyflie' because they are not part of the same tree.Tf has two or more unconnected trees.

terminate called after throwing an instance of 'tf2::ConnectivityException'
what(): Could not find a connection between 'world' and 'crazyflie' because they are not part of the same tree.Tf has two or more unconnected trees.
[crazyflie/controller-6] process has died [pid 14946, exit code -6, cmd /home/xavier/fypcrazyflie_ws/devel/lib/crazyflie_controller/crazyflie_controller __name:=controller __log:=/home/xavier/.ros/log/a05ecc5c-04f8-11e8-8bef-f48c50ec2615/crazyflie-controller-6.log].
log file: /home/xavier/.ros/log/a05ecc5c-04f8-11e8-8bef-f48c50ec2615/crazyflie-controller-6*.log

2. Whenever I put in the static_transformation from node tf, and I run /tf, it overlaps 2 results one that mocap is tracking. Which in this case my view frames just captured when it switches to static transformation. 

3. How does the view_frames suppose to look like? world > crazyflie > crazyflie_baselink 
Does this mean I have create a node from mocap to call out for arg: frame?

whoenig, 2018-01-30:
1. How does the view_frames output look like in this case though? You can also visualize the tf using rviz. It might be that your frame names simply do not match what the crazyflie_ros package expects by default.
3. There is an example output for 2 CFs in the book chapter I referenced. Essentially there needs to be a node "world" (or something else), and an arrow from "world" to "crazyflie".

s3xyxavy, 2018-01-31:
Hi whoenig,

1. I have edited the mocap_optitrack file on mocap.yaml, from Robot_1/base_link to crazyflie. And have view_frames as 
world>crazyflie>crazyflie_baselink 
I am able to execute the take off command.

2. The crazyflie is able to take off but not hover, it might be due to weight of the reflective balls. The arg frame of x=0 y=0 z=0.5 does it mean it hovers at coordinate of 0,0,0.5? Or it hovers an additional 0.5 from its original position?

Thank you very much!

Warm regards,
Xavier

whoenig, 2018-01-31:
I am glad the takeoff works now! The coordinates are absolute, so it should hover around 0,0,0.5 in the ROS world frame. Note that the controller is not very good - it should be fine for hovering, but it is not great for trajectory tracking. You achieve better performance if the position controller is running on-board. This works with the Crazyswarm I referenced earlier, and we have some preliminary work that allows to use crazyflie_ros for trajectory tracking with a controller running on-board (see https://github.com/bitcraze/crazyflie-firmware/commit/2a87f5ac84242ac9956d2828a30928bea327d079).

s3xyxavy, 2018-02-05:
Hi whoenig, 
My hover has not been successful till now, I have checked the quaternions with the tf, by tilting 90 degree on pitch, roll and yaw axis and it is turning positive and negative respectively, it works good. Coordinates, quarternions work good, but the Crazyflie still slides to the same side for the same 5 test I have been running.
 
1. I have been to the trajectory tracking, update and clone the repo on the github to my Ubuntu that you are working on. But I am still unsure of what steps to take to start up the position controller. 
I am unsure of how to go on after the header Compiling, on the link you sent me, how do you go about
 " compile the firmware with ESTIMATOR set to "kalman" and CONTROLLER set to "mellinger"? "

Warm Regards,
Xavier

whoenig, 2018-02-05:
You would need to clone the firmware and build it using `make`. Before you do so, adjust the following lines in the Makefile: https://github.com/bitcraze/crazyflie-firmware/blob/master/Makefile#L30-L31. (Replace "any" with "kalman" and "pid" with "mellinger").

s3xyxavy, 2018-02-06:
Hi whoenig,

Thank you for being so patient with your replies. I have ‘make’ the crazyflie-firmware file, compiled with the rest of the files, but still unfortunatelyun able to hover. I have recorded 2 instances where I execute hovering, hope you can give me more insight into getting it to hover. 
https://youtu.be/2siXJARX_xk 
https://youtu.be/r6pqEkCJS7M 

I have watched many videos on crazyflie hovering smoothly and hopefully envision it with my crazyflie some day. 
Appreciate your guidance, many thanks.

Warm Regards,
Xavier

whoenig, 2018-02-06:
1. In your videos it looks like you are using the "old" method, since you are pressing a button? In order to test the on-board controller, you will need to launch [external_position_vicon](https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_demo/launch/external_position_vicon.launch) [Which needs to be slightly changed to support Optitrack]. Then, in another terminal, execute [python execute_trajectoy.py figure8withTakeoffAndLanding.csv](https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_demo/scripts/execute_trajectory.py).
2. If you prefer the "old" method for now, you should probably go back to the old firmware (the kalman filter does not work as well if there is no mocap information sent to the CF). Also, please pull the latest crazyflie_ros: I just fixed a critical race condition yesterday.

In all cases, double check the coordinate system and that it matches the ROS convention. What you showed in the video might very well be an issue with an axis flip, wrong yaw etc.

s3xyxavy, 2018-02-06:
Hi whoenig,

1. Yes, got it. I figured if I am not able to hover I would not be able to test the onboard controller. 

2. Will pull the latest crazyflie_ros. 

Figured that I couldn’t use the crazyswarm software, as it requires at least 1.10 version for motive (am stuck with 1.7). The onboard controller on the other hand still do work with the ‘old’ crazyflie_ros right?
This might very well be that if the TF works on RVIZ it does not mean that the yaw angle is simply correct, yes? 
I would be probably be working back and relying on VRPN in that case.. 
Thanks!

Warm Regards, 
Xavier

whoenig, 2018-02-06:
Crazyswarm: It should be possible to add other backends, for example by copying the logic from `mocap_optitrack`. Also, if you only want to fly with unique marker arrangements, it would be trivial to add a VRPN backend.

The onboard controller works with the "old" crazyflie_ros. The goal is to merge both CF firmwares and crazyflie_ros eventually.

I have seen many issues with OptiTrack because users can apparently choose in Motive which coordinate system to use. If that choice doesn't match what the ROS side expects, the coordinate transformations are all wrong.

s3xyxavy, 2018-03-13:
Hi whoenig, 

I am closing this topic as I have finally found the issue of it. For the benefit of the people who comes across this thread, 

The mocap_optitrack only works IF

1. Your optitrack setup is the same axis coordinate setup as the one that it’s been set. 

2. There is no apparent way to find and edit the codes that exist in mocap_optitrack. 

I reccomend using VRPN, as it directly streams the whatever coordinates without any rotations or edit fresh from Optitrack. 

You could look into the other users editing the VRPN changing its axis by transforming such example codes are as  from crigroup from NTU (Nanyang Technological University) or eth-zasl (uses a PI/2 rotation). 

For beginners that just started researching on Optitrack and ROS integration, could take note that when editing cpp files in VRPN and allocating the VRPN file with whoenig crazyflie_demo files where it contains nodes code editing please remember to: 

1. Save the .cpp file 
2. catkin_make (make sure that the red lines are showing updates of the nodes) 
3. roslaunch crazyflie_demo hover_vrpn.launch 

Please don’t hesitate to direct message me! A University of Glasgow student, Xavier Leong. 

Warm Regards, 
Xavier

whoenig, 2018-03-21:
I am currently in the process to integrate a new high-level commander into the official firmware and crazyflie_ros. The "old" way is described in W. Hönig and N. Ayanian. "Flying Multiple UAVs Using ROS", Chapter in Robot Operating System (ROS): The Complete Reference (Volume 2), Springer, 2017. (see http://act.usc.edu/publications.html for a free pre-print). The "new" way will use an on-board controller and planner and is still work-in-progress.

LZMHIT, 2018-03-22:
Oh, I am so glad to get your response, and I get the paper yesterday. And I have another question, if I don't have the  joystick， I want to use gazebo to simulate the flying, use the ground_truth value in gazebo replacing the motion capture system, what should I do?

whoenig, 2018-03-22:
Unfortunately, we don't have gazebo support yet. I recommend testing high-level algorithms without Gazebo first. If you want to test with UAV dynamics, you can use a package such as RotorS to simulate a UAV within Gazebo.

closed 2018-04-10 by s3xyxavy

shmpwk, 2019-03-21:
Hi, @s3xyxavy ;

I cannot execute the take off command, when launch hover_vrpn.launch.
Error occurred like
`Exception thrown while processing service call: Could not find a connection between 'world' and 'crazyflie' because they are not part of the same tree.Tf has two or more unconnected trees.`

You seem to have solved this problem by changing the program like
> I have edited the mocap_optitrack file on mocap.yaml, from Robot_1/base_link to crazyflie. And have view_frames as
world>crazyflie>crazyflie_baselink
I am able to execute the take off command.

Does it mean you didn't change crazyflie_demo to connect  'world' and 'crazyflie', but changed other package file?

Please see #114 

Best regards, Shumpei

shmpwk, 2019-03-23:
OK, I've done, thank you. Looking at#126.