whoenig/crazyflie_ros #86: Timeout error

bklein18, 2018-02-06:
When I try to run ``roslaunch hover_vicon.launch`` I get the following set of errors:
``
terminate called after throwing an instance of 'std::runtime_error'
  what():  timeout
[crazyflie_server-1] process has died [pid 7536, exit code -6, cmd /home/ubuntu/crazy/devel/lib/crazyflie_driver/crazyflie_server __name:=crazyflie_server __log:=/home/ubuntu/.ros/log/7e974784-0b56-11e8-8dd6-e89eb42d3fd7/crazyflie_server-1.log].
log file: /home/ubuntu/.ros/log/7e974784-0b56-11e8-8dd6-e89eb42d3fd7/crazyflie_server-1*.log
``

Is this an issue with my having setup something incorrectly, or is it a latency issue, or what?

Thanks

whoenig, 2018-02-06:
This means that no connection could be established. Either your CF is turned off, or you specified the wrong uri.

bklein18, 2018-02-08:
I tried running it again after ensuring that my URI was correct and that my CF was turned on, and now it's warns me about a low link quality, then it throws a ``LIBUSB_ERROR_TIMEOUT``. Is this because of the low link quality? I have the rules set up for both the CF and the Crazyradio dongle.

whoenig, 2018-02-08:
This seems to happen on some machines. It shouldn't be related to the low link quality (although that means you should try a different channel and/or switch to the 2M data rate). As test, you can remove the error throwing [here](https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_cpp/src/Crazyradio.cpp#L154) and  [here](https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_cpp/src/Crazyradio.cpp#L173).
If that works I'll likely push a change to ignore LIBUSB_ERROR_TIMEOUT's and only throw errors in other cases.

bklein18, 2018-02-08:
I tried commenting out that line, and I'm no longer getting ``LIBUSB_ERROR_TIMEOUT``, though it does seem to endlessly print ``Received unrequested data for block: 6``. That is, it does so until another ``timeout()`` is thrown. Using scan and the CF client, I confirmed that I had the correct address for the CF 2.0 and for the Crazyradio, as well as seeing the CF turned on. Is there some other error throwing section that I'm missing?

whoenig, 2018-02-08:
I just pushed a [change](https://github.com/whoenig/crazyflie_ros/commit/68905e31387ad7b117adb7207dfc481d5b6e3140) that makes the changes I had in mind. I suspect that the unrequested data for block was created by data corruption that could happen if you simply comment out the two lines.

Please check if that helps. If not, please specify the firmware, the OS, and ROS versions you are using.

bklein18, 2018-02-12:
It sadly didn't work. The firmware I have on the CF is the 2017.06 firmware, my OS is Ubuntu 14.04, and I run ROS Indigo. Thanks.

whoenig, 2018-02-13:
Does `rosrun crazyflie_tools scan` find your CF? Do you still see the `Received unrequested data for block: 6` warning messages?

bklein18, 2018-02-14:
It was finding my CF, at least when I run it with the `-v` tag. I'm no longer seeing the `Received unrequested data for block: 6` error either, however. Now, it alternates between timing out, and getting to a certain point, `[INFO] [WallTime: 1518652127:708272] found takeoff service` before just stopping. My CF doesn't do anything, and I'm not seeing any more information in any of my console windows pertaining to any other errors.

whoenig, 2018-02-15:
What do you mean by "just stopping"? If you run `vicon_hover.launch`, nothing will happen until you press a button on the joystick for takeoff or run a user-script. Does the teleoperation work with ROS?

bklein18, 2018-02-20:
Sorry for the long delay before a response. I run `hover_vicon.launch`, and even when I press the takeoff button and see `Takeoff requested!`, nothing happens. Teleoperation does work, as well.

rayaan92, 2018-03-05:
I have the same timeout issue when trying to launch the lps dwm_loc_ekf_hover.launch.

I keeps saying time, I am able to connect to the crazyflie via the cfclient, and i changed the channel.

The strange part is everything was working perfectly on Friday. I am running ubuntu 16.04.3 with ROS Kinetic.

please advise on how to resolve this issue.

whoenig, 2018-03-05:
What is the output of `rosrun crazyflie_tools scan`? How did you specify your URI when running the launch file?

rayaan92, 2018-03-05:
Yes i specified yhe uri and everything, the issue was with ubuntu, after running a apt-get update it stopped working so i just reverted the update and it worked again

closed 2018-06-05 by whoenig