whoenig/crazyflie_ros #87: "what():  timeout" issue in regard with Radio Address

srnetsr, 2018-02-09:
Hi,

I am using crazyflie and crazyflie_ros.
When I execute hovering launch file, then it worked with Radio address  **0xE7E7E7E7E7**.
($ roslaunch crazyflie_demo hover_opti_temp.launch uri:=radio://0/100/2M/ frame:=cf2 x:=0 y:=0 z:=1)

However, I switch Radio address to **0xE7E7E7E701** as shown in the following figure.
[image]

Then, when run the following command, the error message happened.
$ roslaunch crazyflie_demo hover_opti_temp.launch uri:=radio://0/100/2M/ frame:=cf2 x:=0 y:=0 z:=1
or
$ roslaunch crazyflie_demo hover_opti_temp.launch uri:=radio://0/100/2M/E7E7E7E701 frame:=cf2 x:=0 y:=0 z:=1
The error message is like that
**terminate called after throwing an instance of 'std::runtime_error'
  what():  timeout**

I read the issue about this error message, but I think it is not applicable to this issue.
Please help me.

Thank you.

P.S. In this package, there is only one yaml file, crazyflie_controller/config/**crazyflie2.yaml**. But it does not seem to be related with radio setting, but PID gains.

whoenig, 2018-02-09:
Does the Python Client find the CF using the specified address? Also, when reconfiguring, it is best to use a USB cable for connection (in my experience it doesn't always work to reconfigure while connected over the radio.)

Your usage of crazyflie_ros seems correct. You don't need to edit any config file.

srnetsr, 2018-02-09:
When I switch Radio Address, then I turned it off and tuned it on.
After that, I ran Python Client again.
At the first time, Python Client did not scan anything.
So I change "Address" to 0xE7E7E7E701 in the following figure. Then, it can scan cf2 and connect to it.

[image]

whoenig, 2018-02-09:
That looks good. Does `rosrun crazyflie_tools scan --address 0xE7E7E7E701` find it, too?

srnetsr, 2018-02-09:
The result is that
$ rosrun crazyflie_tools scan --address 0xE7E7E7E701
Found Crazyradio with version 0.53
radio://0/100/2M/E7E7E7E701
No Crazyflie over USB found.

whoenig, 2018-02-10:
That looks like the configuration is correct. How does `hover_opti_temp.launch` look like?

srnetsr, 2018-02-10:
Yes, I think so.
I also used usb cable as well, but the results are the same.

The launch file is the same with the original version except the motion capture camera parts.
I am using optitrack, so I just changed 

< arg name="opti_ip" default="!@#$%"/>
< arg name="frame" default="!@#$%" />
.
.
 
  <include file="$(find vrpn_client_ros)/launch/sample_temp.launch">
	<arg name="server" value="$(arg opti_ip)" />
  </include>

That is all I changed.

whoenig, 2018-02-11:
What do you mean by "original version"? The snippets you show look very different from https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_demo/launch/hover_vrpn.launch

My guess is that maybe the uri doesn't get passed in correctly to the crazyflie_add call. If you post the full log it should contain the uri it tries to connect to.

srnetsr, 2018-02-12:
I mean the original version is the link what you gave me.

From the link you give me, https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_demo/launch/hover_vrpn.launch,
I only changed,
1.
`<arg name="ip" default="vicon" />` to `<arg name="opti_ip" default=<..optitrack ip address../>`
2.
`<arg name="frame" default="crazyflie" />` to `<arg name="frame" default="cf2" />`
3. 
```
<!-- run vrpn client -->
    <node pkg="ros_vrpn_client" name="vrpn" output="screen" type="ros_vrpn_client" required="true">
      <param name="ip" value="$(arg ip)" />
      <param name="port" value="$(arg port)" />
      <param name="frame_id" value="/world" />
      <param name="child_frame_id" value="$(arg frame)" />
</node>
```
to
```
  <!-- run OptiTrack bridge -->
  <include file="$(find vrpn_client_ros)/launch/sample_temp.launch">
	<arg name="server" value="$(arg opti_ip)" />
  </include>
```
Because I am using "vrpn_client_ros" package (http://wiki.ros.org/vrpn_client_ros), I changed as like the above.
Actually, the parameter information is defined in "sample_temp.launch"
I attach the "sample_temp.launch" 
```
<launch>
  <arg name="server" default="localhost"/>
  <node pkg="vrpn_client_ros" type="vrpn_client_node" name="vrpn_client_node" output="screen">
    <rosparam subst_value="true">
      server: $(arg server)
      port: 3883

      update_frequency: 100.0
      frame_id: world
      child_frame_id: cf2
      # Use the VRPN server's time, or the client's ROS time.
      use_server_time: false
      broadcast_tf: true

      # Must either specify refresh frequency > 0.0, or a list of trackers to create
      refresh_tracker_frequency: 1.0
      #trackers:
      #- FirstTracker
      #- SecondTracker
    </rosparam>
  </node>
</launch>
```

I hope what I show you is helpful for resolving the issue.
I will give you the full log soon.

srnetsr, 2018-02-12:
With 0xE7E7E7E7E7, I got the following log.

```
ubuntu@-ubuntu:~/catkin_ws$ roslaunch crazyflie_demo hover_opti_temp.launch uri:=radio://0/100/2M frame:=cf2 x:=0 y:=0 z:=1
... logging to /home/ubuntu/.ros/log/85ed4b96-1015-11e8-bc02-74d435137038/roslaunch-asrl-ubuntu-23376.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
xacro.py is deprecated; please use xacro instead
started roslaunch server http://asrl-ubuntu:40333/

SUMMARY
========

PARAMETERS
 * /crazyflie/controller/PIDs/X/integratorMax: 0.1
 * /crazyflie/controller/PIDs/X/integratorMin: -0.1
 * /crazyflie/controller/PIDs/X/kd: 20.0
 * /crazyflie/controller/PIDs/X/ki: 2.0
 * /crazyflie/controller/PIDs/X/kp: 40.0
 * /crazyflie/controller/PIDs/X/maxOutput: 10.0
 * /crazyflie/controller/PIDs/X/minOutput: -10.0
 * /crazyflie/controller/PIDs/Y/integratorMax: 0.1
 * /crazyflie/controller/PIDs/Y/integratorMin: -0.1
 * /crazyflie/controller/PIDs/Y/kd: -20.0
 * /crazyflie/controller/PIDs/Y/ki: -2.0
 * /crazyflie/controller/PIDs/Y/kp: -40.0
 * /crazyflie/controller/PIDs/Y/maxOutput: 10.0
 * /crazyflie/controller/PIDs/Y/minOutput: -10.0
 * /crazyflie/controller/PIDs/Yaw/integratorMax: 0.0
 * /crazyflie/controller/PIDs/Yaw/integratorMin: 0.0
 * /crazyflie/controller/PIDs/Yaw/kd: -20.0
 * /crazyflie/controller/PIDs/Yaw/ki: 0.0
 * /crazyflie/controller/PIDs/Yaw/kp: -200.0
 * /crazyflie/controller/PIDs/Yaw/maxOutput: 200.0
 * /crazyflie/controller/PIDs/Yaw/minOutput: -200.0
 * /crazyflie/controller/PIDs/Z/integratorMax: 1000.0
 * /crazyflie/controller/PIDs/Z/integratorMin: -1000.0
 * /crazyflie/controller/PIDs/Z/kd: 6000.0
 * /crazyflie/controller/PIDs/Z/ki: 3500.0
 * /crazyflie/controller/PIDs/Z/kp: 5000.0
 * /crazyflie/controller/PIDs/Z/maxOutput: 60000.0
 * /crazyflie/controller/PIDs/Z/minOutput: 10000.0
 * /crazyflie/controller/frame: cf2
 * /crazyflie/controller/worldFrame: world
 * /crazyflie/crazyflie_add/enable_logging: False
 * /crazyflie/crazyflie_add/enable_logging_battery: True
 * /crazyflie/crazyflie_add/enable_logging_imu: True
 * /crazyflie/crazyflie_add/enable_logging_magnetic_field: True
 * /crazyflie/crazyflie_add/enable_logging_packets: True
 * /crazyflie/crazyflie_add/enable_logging_pressure: True
 * /crazyflie/crazyflie_add/enable_logging_temperature: True
 * /crazyflie/crazyflie_add/pitch_trim: 0
 * /crazyflie/crazyflie_add/roll_trim: 0
 * /crazyflie/crazyflie_add/tf_prefix: crazyflie
 * /crazyflie/crazyflie_add/uri: radio://0/100/2M
 * /crazyflie/crazyflie_add/use_ros_time: True
 * /crazyflie/joy/dev: /dev/input/js0
 * /crazyflie/joystick_controller/use_crazyflie_controller: True
 * /crazyflie/pose/name: goal
 * /crazyflie/pose/rate: 30
 * /crazyflie/pose/x: 0
 * /crazyflie/pose/y: 0
 * /crazyflie/pose/z: 1
 * /robot_description: <?xml version="1....
 * /rosdistro: kinetic
[... 52 lines cut ...]
[INFO] [1518454527.433525]: found update_params service
[INFO] [1518454527.433932]: waiting for emergency service
[INFO] [1518454527.435062]: found emergency service
[INFO] [1518454527.435290]: waiting for land service
[ INFO] [1518454527.917810029]: Found new sender: cf2
[ INFO] [1518454527.917968770]: Creating new tracker cf2
[INFO] [1518454528.340457]: found land service
[INFO] [1518454528.340863]: waiting for takeoff service
[INFO] [1518454528.341977]: found takeoff service
^C[rviz-10] killing on exit
[vrpn_client_node-9] killing on exit
[crazyflie/baselink_broadcaster-8] killing on exit
[crazyflie/pose-7] killing on exit
[crazyflie/joystick_controller-5] killing on exit
[crazyflie/controller-6] killing on exit
[crazyflie/joy-4] killing on exit
[crazyflie_server-2] killing on exit
vrpn_Connection::~vrpn_Connection: Connection was deleted while 1 references still remain.
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

whoenig, 2018-02-12:
Thanks for the details.
1. Which OS are you running. In particular, is it 64 or 32-bit?
2. Can you try if one of the command line tools connects, e.g. `rosrun crazyflie_tools listParams --uri radio://0/100/2M/E7E7E7E701`

srnetsr, 2018-02-12:
I am sorry.
I found that I got the flight with "0xE7E7E7E7E7" address.
However, after the flight I executed crazyflie-clients-python with the command "python3 bin/cfclient", I got the same error when I run "roslaunch crazyflie_demo hover_opti_temp.launch uri:=radio://0/100/2M/E7E7E7E701 frame:=cf2 x:=0 y:=0 z:=1".
So I replugged CrazyRadio PA, but the same error happened.
Finally, I reboot CF2, then hovering launch file worked....weird...
I think that was related with code issue, but usb or crazyflie-client-python issue...

Thank you so much for your help anyway.

closed 2018-02-12 by srnetsr