whoenig/crazyflie_ros #89: Hovering, controller.cpp

s3xyxavy, 2018-03-13:
Hi whoenig, 

I was checking out how the hovering (hover_vrpn.launch) controller works and realized that you have input in the orientation and position as inverse to allow it enter back to the target set point.

I switched target point at (0,0,0.5), the Thrust to 0, and ROS_INFO the messages of  linear x,y,z and angular velocity. I held the crazyflie 2.0 on my hand and tilt slightly away from the target point.
I realized that for Linear.X in increases and decreases normally. (slow increase between -10 to 10 linearly, and is correct for which X axis  ) 
Linear.Y however keeps increases or decreases exactly at either -10 or 10. with no linear motivation.
Linear.Z keeps thrusting at 10000, at no matter which Z axis position the CF 2.0 is at. 

1. Is linear.Y and linear.Z suppose to happen for the hover? 
2. How would you suggest to put the markers on the CF for a Vicon/Opitrack? 
3. Other than this github/crazyflie_ros, Is crazyswarm the only option to hover? This is because the optitrack version of mine is only suitable for this. 

Thanks in advance.

Warm Regards,
Xavier

whoenig, 2018-03-21:
1. That sounds weird. What is the current position of your CF as measured by your mocap?
2. As far away as possible from each other and asymmetric. You can find example pictures in the tutorial book chapter references in the repo and on the Crazyswarm webpage.
3. I am in the process of integrating the crazyswarm functionality into the official firmware and crazyflie_ros. The goal is that in a few weeks you'll be able to a) use the latest official firmware with both crazyswarm and crazyflie_ros, b) the official firmware will support planning and trajectory execution on-board, c) crazyflie_ros will support takeoff/landing and waypoint navigation with the same good accuracy as the crazyswarm, d) the off-board controller (crazyflie_controller) will be removed.

s3xyxavy, 2018-03-22:
Hi whoenig,

I am happy to announce to you that I have finally made it hover and utilise the waypoint movements.
A) mocap_node can be used
B) vrpn node can be use, but 
Requires a lil tweaking, since the optitrack frames are a lil funky.. when a x-z ground plane is calibrated it gives 
X as north
Y as up 
Z as east (facing you) 
Which is actually still right hand axis coordinate system. 

For 
X(0) = X_ROS (0)
Y(1) = -Y_ROS (2)
Z (2) = Z_ROS (1)
Effectively just a 90 degree rotation and invert it a negative sign. 

More importantly is that, even when I set the 4 markers the same as how crazyswarm was configured. I realise it has to be set in the correct direction. 
For example, for my case that dint work it’s when the crazyflie facing away from me and next setting 4 markers as rigid body (blue led furtherest away and red nearest ). 
For the case that finally worked out for hover and waypoint, is where blue led on the left and red led on the right and next setting 4 markers as rigid body.

1. If the hover happens, does it totally makes sure that my orientation is correct? Or more like my transformation by using this illegal way of swapping y and z coordinates with a negative sign on y, proves that it can be done this way safely? Am I required to use quaternions even though the CF still hovers and waypoint nicely? 

Hoepfully my observation could help anyone that is using Optitrack. 

I look forward to the new update and hopefully still be able to use it on a Ubuntu 14.04 indigo platform! 

Thank you for your guidance and making ROS an educational platform.

whoenig, 2018-03-22:
You can swap the quaternion components and sign in a similar way in order to get the correct pose feedback. The controller attempts to correct yaw and uses the pose information for that purpose.

yotamgani, 2018-04-17:
Hello Wolfgang,
I am also doing a project on ROS+OptiTrack+Crazyflie system to enable a swarm waypoint navigation hovering.
Currently, we are able to perform a single crazyflie hovering and waypoint navigation.
We are using your crazyflie_ros package and the vrpn_client_ros package, on ROS-indigo, OptiTrack with Motive 2.0, crazyflie 2.0, and running the entire system on one workstation using the bitcraze VM, running Ubuntu 14.04.

While we are able to successfully hover, the controller is still quite slow, and it takes a few seconds (sometimes even around 10 seconds) to reach hovering point, and the flight is not very nice.

I realize that the controller is not the new on-board controller that you are currently using on the CrazySwarm project.
In this part of the project, we don't want to switch to a different project, for we are already "invested" in the current one, and we only have few more weeks to finish and submit our project.

So, I have noticed to you mentioned in this forum around a month ago that you are in the process of integrating the new better on-board controller into the crazyflie_ros project.
Can you tell us what is the current status and if there is a way already (even if your work around it has not been completed) to integrate the new controller into our project?

Thank you very much
Yotam

whoenig, 2018-04-17:
Hi Yotam,
Waypoint navigation and nice takeoff/landing is already supported; Trajectory following is on its way. To use it, use the following steps:

1. Compile the latest official stm32 firmware using ESTIMATOR=kalman and CONTROLLER=mellinger
2. Use the latest (master-branch) of crazyflie_ros
3. I use the following launch file: https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_demo/launch/external_position_vicon.launch. You would need to change it for VRPN (along with a VRPN version of https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_demo/scripts/publish_external_position_vicon.py)
4. Waypoint following can be done using a goTo method in the high-level script (see https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_demo/scripts/test_high_level.py#L15-L21 for an example). You can specify a desired duration when the CF should reach its goal - planning, state estimation, and non-linear control are all running on-board.

yotamgani, 2018-04-19:
Hi Wolfgang,
Thanks for your reply.

1)
I went over your example codes and I wan't to make sure that I understand things right before i'm getting started:
So, currently, the way things work with the off-board controller is as follows:
- The vrpn_client node (in my case, OptiTrack) is publishing it's position into /tf topic.
- The /pose node is publishing the required position into /crazyflie/goal topic.
- The controller node is subsribed to both /tf and /crazyflie/goal and gets both data.
- The controller is publishing velocity commands to /crazyflie/cmd_vel.
- The crazyflie server is subscribed to /crazyflie/cmd_vel and receives flight commands.
- Another joystick (Xbox360) node is publishing into /joy topic take_off/land/emergency commands, and the joystick_controller script translate those into crazyflie_server services (takeOff, Land etc.).

Now, I want to make sure that I understood the new settings (with the on-board controller) correctly:
- The vrpn_client is publishing into /tf position data.
- The publish_external_position node is subscribed to the vrpn_client data (can be set simply to /tf).
- The publish_external_position node is publishing position (just x,y,z, no orientation) into /crazyflie/external_position. If I got it right, other than the firstTransform procedure that sets the kalman parameters, the rest of the data is being published as is (and could have been actually be sent directly by vrpn_client node).
- The crazyflie_server is subscribed to /crazyflie/external_position and the on-board controller does all the work from there on.
- Take off, Landing, and goal position is now taken care of by crazyflie_server services called by the high level script node (so, no need for joystick controller).

If that is correct, the changes that needs to be done are:
a) Adjust the launch file to the vrpn_client
b) Adjust topics names in publish_external_position (could simply be set to /tf)
c) change frame names (according to motive names) in the high level script.

I would be glad if you can confirm my understanding or correct me where I am wrong.

2)
After completing this stage, to proceed to the Swarm stage, will the changes be similar and relatively simple as the moving from hover_vrpn into multi_hover_vrpn? i.e. mainly to duplicate the group of crazyflie in the launch file to several groups (as the number of crazyflie's I want to use) and add more instances of crazyflie's in the high level script according to frame names?

3) 
A more general question:
What is the "Delta" between the crazylie_ros project that we are working with and the CrazySwarm project? i.e. what are the advantages and/or disadvantages of working with each one of them?

Thank you very much,
Yotam

whoenig, 2018-04-19:
1. Yes, that is correct. One note: The official firmware EKF does not support fusing the full pose information, that's why we only send the position, but not orientation. In practice, this will limit your ability to hover at one place for a long time because the yaw might drift. For most practical application it works fine (as long as you do move a little in x/y), and there is an open issue for the firmware to fix that. 

2. I believe so, but I did not test that yet. In general, this approach will only scale to a few (something in the order of 2 to 4 CFs per radio). For larger swarms, you would still need the CrazySwarm project instead (see below).

3. The delta is consistently getting smaller as we move functionality in the official firmware:-) We have a few outstanding firmware PRs submitted and once those are accepted we can actually use the CrazySwarm with the latest crazyflie_ros and official firmware. The only difference will be that the CrazySwarm uses a "crazyswarm_server" instead of the "crazyflie_server". The crazyswarm_server is similar to crazyflie_server, but keeps a thread per "Crazyflie Group" (that is, CFs that share the same channel and can be used with one radio). Each group uses broadcasts instead of direct communication whenever possible (e.g., for takeoff commands), to reduce bandwidth and achieve synchronization. This allows you to control 15-30 CFs (depending on if you send position or pose) per radio. There is a very similar python interface to control all of them (almost identical to the new high-level script in crazyflie_ros). Additionally, it tightly integrates our custom frame-to-frame tracking mechanism to allow identical marker configurations.

The advantage of crazyflie_ros is that it works with the official firmware. The downside is that you can only use it for "few" (maybe up to 6) CFs. The advantage of the CrazySwarm is that it allows very large swarms. We will keep reducing the gap between the two in the future, but it will likely stay two separate projects because they are meant for different applications.

yotamgani, 2018-04-30:
Hello Wolfgang,
So, I have successfully managed to perform the required changes for the new on-board controller with the high-level script using the vrpn (with OptiTrack). 
In addition to the changes in topics and frame-names, there was also a need to adjust the message type to the vrpn code.
I can share the working codes, if you want.
Indeed, the flight now is very fast and accurate, so thanks!

However, unfortunately, when I wrote a new program to fly a swarm of 2 crazyflies I get the following weird behavior after executing the launch file:
During the creation of all nodes, I can see that both crazyflies are added to the server. I can actually see all the data coming (such as: ESTIMATOR type 2, I2C OK etc.) for both crazyflies.
But in some point, all of a sudden I get a timeout error without an apparent reason.
Actually, in few different runs the timeout pops in changing timings (but quite similar).

I did managed to perform swarm hovering using 2 crazyflies with the old multi_hover_vrpn program, using the old server, firmware and off-board controller, so it is not an issue of addresses or radio configuration, and I assume it is not an issue in the codes, for they are quite similar.

I hope you can give me some insight on this new issue, so eventually I will be able to perform swarm navigation using the crazyflie_ros pkg with the on-board controller.

Thank you,
Yotam

whoenig, 2018-04-30:
Hi,
Great! I would be certainly interested in your changes. I prefer Pull Requests, but I am also fine with attached code etc.

For your new problem, I recommend the following:
1.) Disable logging. Logging requires a lot of bandwidth and might hinder other packets to get transmitted.
2.) Increase the timeout: https://github.com/whoenig/crazyflie_cpp/blob/master/include/crazyflie_cpp/Crazyflie.h#L346-L347.

On another note, the Crazyswarm is now using the official crazyflie_ros. As discussed earlier, the main difference is that it is using broadcasts instead of direct communication (which avoids timeout errors effectively).
Best,
Wolfgang

yotamgani, 2018-05-06:
Hello Wolfgang,
I am happy to inform, that we are able to perform swarm waypoint navigation using the on-board controller.
Timeout was needed to be increased in order to make it work.
All 3 codes (for single and swarm) were uploaded as pull requests (2 launch file and 1 script) to your git project. 
It will be a honor to contribute to your project after all the hard work for future OptiTrack and other VRPN users.
Thanks for all your help!

By the way, do you have any updates regarding Trajectory ?

Thanks again,
Yotam

whoenig, 2018-05-07:
Hi Yotam,
Thank you very much for your contribution! I merged or commented on all your PRs - thanks for keeping them clean and the code well documented!

Trajectory upload and following is now fully supported, see https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_demo/scripts/test_high_level.py#L23-L43 for an example. We support 8th degree polynomial trajectories at the moment, that can, for example, be generated using the tools in https://github.com/whoenig/uav_trajectories.
We also have a prototype of the Crazyswarm that can operate with the latest official firmware - the only difference being that the external state update is position only (while it is position and orientation if you use the Crazyswarm firmware.)

Thanks again,
Wolfgang

closed 2018-06-05 by whoenig