whoenig/crazyflie_ros #92: No Stop.msg file, cannot use Hover.py

fro5, 2018-04-19:
Hello,

I am trying to run the demo for just Hovering a crazyflie and the Hover.py script is importing a message from the crazyflie_driver msg directory called Stop.msg that doesn't exist currently in this repo. I have the Stop.srv file but I don't want to assume what lines were in the msg file. Please help!

Respectfully,
Falon Ortega

whoenig, 2018-04-20:
If you simply want to hover, I recommend using the new high-level interface instead, see https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_demo/scripts/test_high_level.py#L11-L21 for an example. I'll take a look at Hover.py, but this currently demonstrates on how to use low-level commands.

adaldo, 2018-05-21:
I also have this issue. There is a `Stop.srv` file, but not a `Stop.msg` file.

closed 2018-06-05 by whoenig