whoenig/crazyflie_ros #97: Compiling the project fails with gcc 7.3.x

adaldo, 2018-05-21:
The package `crazyflie_cpp` seems to fail compilation (for example, with `catkin_make` or `catkin build`) with gcc 7.3.x. I haven't saved the (long) error message, but it seems to being with a `std::function` not being defined. The issue seems to vanish if `#include <functional>` is added in `Crazyflie.h` somewhere at the top, for example after `#pragma once`.

whoenig, 2018-05-22:
Thanks! Does gcc 7.3.x come with a specific Ubuntu version or did you just manually install it? So far, we only test with Ubuntu 16.04 and ROS Kinetic, but we plan to support more recent (LTS) versions as well.

adaldo, 2018-05-25:
It came with Ubuntu 18.04, I was using Melodic. Once the library was included I could compile and connect (read temperature, battery, etc.) but I couldn't test flight from that machine.

closed 2018-06-05 by whoenig