whoenig/crazyflie_ros #100: Error when clicking X button to take off

rgoodling3, 2018-06-19:


rgoodling3, 2018-06-19:
Not sure what happened to fix this, I just started up everything again after leaving overnight and it worked

rgoodling3, 2018-06-19:
This is actually a problem again

whoenig, 2018-06-19:
This seems a ROS version mismatch. Do you have multiple ROS versions installed on your machine? Is everything running on one machine or do you use two different machines?

rgoodling3, 2018-06-20:
I only have ROS indigo installed on my machine. I am using a virtual machine to run Ubuntu 14.04

whoenig, 2018-06-21:
I think the problem here is that there are potentially two takeoffs:
1.) High-level command takeoff. In this case, controller.py is currently not supported.
2.) Custom controller, running on the basestation. In this case, things should work, although the flight performance won't be very good.

Which launch file are you using? Note that the manual is for an older version of crazyflie_ros, so there are some mismatches. I am planning to write updated documentation, but I do not have a timeline for it right now.

rgoodling3, 2018-06-21:
I am launching the vicon_hover launch file. It sometimes works but not always. It usually will have three outcomes: not responding to the command at all even though my machine shows that the takeoff was requested, the props spinning but not enough for it to take off, and then sometimes it flies off crazily. I rarely have the problem that I originally posted about, and if I do I can kill the program and restart it and that will typically work.

closed 2018-07-02 by rgoodling3

cjbanks, 2018-07-17:
Hi, I am also having a checksum issue. I only have ROS Kinetic installed on Ubuntu 16.04. Do you suggest we don't use controller.py for joystick controls?