whoenig/crazyflie_ros #101: Hector quadrotor teleop

mlpadilla03, 2018-06-21:
Hi whoenig,

I have seen your solution from issue #52 and I did the same process. Although what happened is it required me to have controller_interface and I got this error 

****
CMake Error at hector_quadrotor_teleop/CMakeLists.txt:22 (add_executable):
  add_executable cannot create target "quadrotor_teleop" because another
  target with the same name already exists.  The existing target is an
  executable created in source directory
  "/home/mark/crazyflie_ws/src/crazyflie_ros/crazyflie_demo".  See
  documentation for policy CMP0002 for more details.

CMake Error at hector_quadrotor_teleop/CMakeLists.txt:23 (target_link_libraries):
  Attempt to add link library "hector_quadrotor_interface" to target
  "quadrotor_teleop" which is not built in this directory.

****
This is just a part of it. Thought this is the start of the error.

mlpadilla03, 2018-06-21:
It's okay already. What I did is remove the crazyflie_ros folder then do catkin_make. Then put back the folder. It's working now.

closed 2018-06-21 by mlpadilla03

Huzayfaa, 2019-02-14:
> It's okay already. What I did is remove the crazyflie_ros folder then do catkin_make. Then put back the folder. It's working now.

Did this actually work and where u then able to run the crazyflie demos for example? I am having the same issue and was wondering if this is the best or most appropriate fix. 

Thanks

whoenig, 2019-02-15:
Do you have hector_quadrotor in the same ROS workspace? If yes, then remove it.

Huzayfaa, 2019-02-15:
At the moment I have, however having removed it I am able to successfully 'catkin_make' the workspace with the crazyflie_ros and lps-ros packages in the 'src' folder

This allows me to use certain commands to identify the Crazyflie using the radio, however I am looking at running the 'roslaunch crazyflie_demo teleop_xbox360.launch uri:=radio://0/80/2M' command and this does not produce any outcome. I understand this maybe because I don't now have the hector_teleop package, however, even having the package in the 'src' folder I get the same response. The following code represents what I see in the command prompt:  

`huzayfaa@huzayfaa:~/crazyflie_ws$ roslaunch crazyflie_demo teleop_xbox360.launch uri:=radio://0/80/2M
... logging to /home/huzayfaa/.ros/log/85218188-313d-11e9-acc2-503eaa8a202f/roslaunch-huzayfaa-7424.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.
`

Following this code it then times out and states that there may be some networking issue and that I would need to try and ping my PC, which when pinged shows perfect behavior with minimal packet loss.  

Any help, or reference to a guide would be highly appreciated. I am currently reading and following the book **Flying Multiple UAVs Using ROSWolfgang H ̈onig and Nora Ayanian**. 

I am currently working towards being able to represent the LPS set-up and the Crazyflie visually using ROS.

whoenig, 2019-02-15:
The latest crazyflie_ros does not need hector_teleop anymore (it comes with its own teleoperation node). The reference you refer to is unfortunately outdated at this point. You can take a look at https://crazyswarm.readthedocs.io/en/latest/, which supports multiple CFs and also works with LPS (in TDOAv3 mode). 

The command you ran should work for teleoperation of a single CF. Please post the exact error message you get (preferably in a new issue actually, since this issue is technically closed.)

Huzayfaa, 2019-02-17:
Thank you for your guidance. I have spent some time familiarising myself with the 'crazyswarm' package and have had to start a fresh following the latest instructions and this has so far worked very well for me. 

I'll raise the error I am/was getting in a new issue, just waiting to repair my CF2.0 to replicate and provide further details on the error. Again, thank you!