whoenig/crazyflie_ros #103: hover_vicon is not hovering but drifiting instead

rgoodling3, 2018-07-02:
I am using Ubuntu 14.04 and a Vicon system. We can get the Crazyflie off the ground, but it will always drift away from the center point. It doesn't usually follow a pattern of where it drifts to.

whoenig, 2018-07-03:
This sounds like the tracking is not working properly. Does the tracking work if you run the launch file and move it around manually (check by looking at the visualization in rviz)?

rgoodling3, 2018-07-03:
Yes. We see the vicon image mirrored by rViz

whoenig, 2018-07-03:
For new development I recommend using the high-level command mode, where the controller is running on-board, instead of hover_vicon.launch, see also https://github.com/whoenig/crazyflie_ros/issues/102.

As for your current problem: How do you actually take-off? How fast does it drift away? Is the CF working otherwise (i.e., are you able to fly it manually using a joystick)?

andresbhoo, 2018-07-09:
I was working with rgoodling3, but she has left the project, so it is just me now. I took your suggestion to move to the new high level command mode. I am currently trying to flash the precompiled firmware to the crazyflie. However, all I get in return is the error: LIBUSB_ERROR_ACCESS. I am attempting to flash to to the crazyflie via crazyradio.

I am using ROS Kinetic in Ubuntu 16.04 on a VMware Workstation.

Any suggestions? And thank you so much for the help thus far.

whoenig, 2018-07-09:
You need to set the USB permissions, see section 3.1 in http://act.usc.edu/publications/Hoenig_Springer_ROS2017.pdf.

andresbhoo, 2018-07-09:
Oh right! Thank you.

closed 2018-12-07 by whoenig