whoenig/crazyflie_ros #104: no matching function for call to LogBlock CrazyflieROS

flavia13, 2018-07-13:
Hi, I would like to modify the crazyflie_server.cpp code a little bit for the comunication with MATLAB, in particular, this are the changes: I simply added onMotorData method, logmotor struct but when I try to add this part of the script:

```
logBlockMotor.reset(new LogBlock<logMotor>(
                &m_cf,{
                        {"motor", "m1"},
                        {"motor", "m2"},
                        {"motor", "m3"},
                        {"motor", "m4"},
        }, cbmotor));

logBlockMotor->start(1); // 10ms`
```
The error is:

` /home/bitcraze/catkin_ws/src/crazyflie_ros/crazyflie_driver/src/crazyflie_server.cpp: In member function ‘void CrazyflieROS::run()’:
/home/bitcraze/catkin_ws/src/crazyflie_ros/crazyflie_driver/src/crazyflie_server.cpp:530:9: error: no matching function for call to ‘LogBlock<CrazyflieROS::logMotor>::LogBlock(Crazyflie*, <brace-enclosed initializer list>, std::function<void(unsigned int, CrazyflieROS::logImu*)>&)’
    }, cb));
         ^
In file included from /home/bitcraze/catkin_ws/src/crazyflie_ros/crazyflie_driver/src/crazyflie_server.cpp:50:0:
/home/bitcraze/catkin_ws/src/crazyflie_ros/crazyflie_cpp/include/crazyflie_cpp/Crazyflie.h:462:3: note: candidate: LogBlock<T>::LogBlock(Crazyflie*, std::__cxx11::list<std::pair<std::__cxx11::basic_string<char>, std::__cxx11::basic_string<char> > >, std::function<void(unsigned int, T*)>&) [with T = CrazyflieROS::logMotor]
   LogBlock(
   ^
/home/bitcraze/catkin_ws/src/crazyflie_ros/crazyflie_cpp/include/crazyflie_cpp/Crazyflie.h:462:3: note:   no known conversion for argument 3 from ‘std::function<void(unsigned int, CrazyflieROS::logImu*)>’ to ‘std::function<void(unsigned int, CrazyflieROS::logMotor*)>&’
/home/bitcraze/catkin_ws/src/crazyflie_ros/crazyflie_cpp/include/crazyflie_cpp/Crazyflie.h:459:7: note: candidate: LogBlock<CrazyflieROS::logMotor>::LogBlock(const LogBlock<CrazyflieROS::logMotor>&)
 class LogBlock
       ^
/home/bitcraze/catkin_ws/src/crazyflie_ros/crazyflie_cpp/include/crazyflie_cpp/Crazyflie.h:459:7: note:   candidate expects 1 argument, 3 provided
crazyflie_ros/crazyflie_driver/CMakeFiles/crazyflie_server.dir/build.make:62: recipe for target 'crazyflie_ros/crazyflie_driver/CMakeFiles/crazyflie_server.dir/src/crazyflie_server.cpp.o' failed
make[2]: *** [crazyflie_ros/crazyflie_driver/CMakeFiles/crazyflie_server.dir/src/crazyflie_server.cpp.o] Error 1
CMakeFiles/Makefile2:2995: recipe for target 'crazyflie_ros/crazyflie_driver/CMakeFiles/crazyflie_server.dir/all' failed
make[1]: *** [crazyflie_ros/crazyflie_driver/CMakeFiles/crazyflie_server.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j1 -l1" failed`

I don't know if this is the right section to ask for this kind of questions, anyway I hope you can help me. I really don't know what I'm doing wrong, I simply added the modified part as similar as possible to the OnImuData.

Thank you!  is The part of the script that I'm modifying

````
private:
  struct logImu {
    float acc_x;
    float acc_y;
    float acc_z;
    float gyro_x;
    float gyro_y;
    float gyro_z;
  } __attribute__((packed));
private:
struct logMotor {
        int m1;
        int m2;
        int m3;
        int m4;
    } __attribute__((packed));
  void onImuData(uint32_t time_in_ms, logImu* data) {
    if (m_enable_logging_imu) {
      sensor_msgs::Imu msg;
      if (m_use_ros_time) {
        msg.header.stamp = ros::Time::now();
      } else {
        msg.header.stamp = ros::Time(time_in_ms / 1000.0);
      }
      msg.header.frame_id = m_tf_prefix + "/base_link";
      msg.orientation_covariance[0] = -1;
//new
        msg.orientation.x = cos(data->gyro_z/2)*cos(data->gyro_y/2)*cos(data->gyro_x/2) + sin(data->gyro_z/2)*sin(data->gyro_y/2)*sin(data->gyro_x/2);
    msg.orientation.y = sin(data->gyro_z/2)*cos(data->gyro_y/2)*cos(data->gyro_x/2) - cos(data->gyro_z/2)*sin(data->gyro_y/2)*sin(data->gyro_x/2);
    msg.orientation.z = cos(data->gyro_z/2)*sin(data->gyro_y/2)*cos(data->gyro_x/2) + sin(data->gyro_z/2)*cos(data->gyro_y/2)*sin(data->gyro_x/2);
    msg.orientation.w = cos(data->gyro_z/2)*cos(data->gyro_y/2)*sin(data->gyro_x/2) - sin(data->gyro_z/2)*sin(data->gyro_y/2)*cos(data->gyro_x/2);

//endnew

      // measured in deg/s; need to convert to rad/s
      msg.angular_velocity.x = degToRad(data->gyro_x);
      msg.angular_velocity.y = degToRad(data->gyro_y);
      msg.angular_velocity.z = degToRad(data->gyro_z);

      // measured in mG; need to convert to m/s^2
      msg.linear_acceleration.x = data->acc_x * 9.81;
      msg.linear_acceleration.y = data->acc_y * 9.81;
      msg.linear_acceleration.z = data->acc_z * 9.81;

      m_pubImu.publish(msg);
    }
}

void onMotorData(logMotor* data) {
        std_msgs::Int32MultiArray array;
        array.data.clear();
        array.data.push_back(data->m1);
        array.data.push_back(data->m2);
        array.data.push_back(data->m3);
        array.data.push_back(data->m4);

        m_pubMotors.publish(array);
    }

std::function<void(uint32_t, logImu*)> cb = std::bind(&CrazyflieROS::onImuData, this, std::placeholders::_1,std::placeholders::_2);

    logBlockImu.reset(new LogBlock<logImu>(
      &m_cf,{
        {"acc", "x"},
        {"acc", "y"},
        {"acc", "z"},
        {"gyro", "x"},
        {"gyro", "y"},
        {"gyro", "z"},
      }, cb));
    logBlockImu->start(1); // 10ms
std::function<void(logMotor*)> cbmotor = std::bind(&CrazyflieROS::onMotorData, this, std::placeholders::_1);
 logBlockMotor.reset(new LogBlock<logMotor>(
                &m_cf,{
                        {"motor", "m1"},
                        {"motor", "m2"},
                        {"motor", "m3"},
                        {"motor", "m4"},
        }, cbmotor));

        logBlockMotor->start(1); // 10ms

}`

whoenig, 2018-07-13:
onMotorData needs to get `uint32_t time_in_ms` as first argument.

Instead of this strongly typed version, you can also use the generic log blocks, e.g., https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_demo/launch/customLogBlocks.launch.

flavia13, 2018-07-13:
Should I comment also this method?
Because I did it and I did also what you suggested and now it compile
```
void onLogCustom(uint32_t time_in_ms, std::vector<double>* values, void* userData) {

    ros::Publisher* pub = reinterpret_cast<ros::Publisher*>(userData);

    crazyflie_driver::GenericLogData msg;
    if (m_use_ros_time) {
      msg.header.stamp = ros::Time::now();
    } else {
      msg.header.stamp = ros::Time(time_in_ms / 1000.0);
    }
    msg.header.frame_id = m_tf_prefix + "/base_link";
    msg.values = *values;

    pub->publish(msg);
  }
```

whoenig, 2018-07-13:
What do you mean by "comment"? If it compiles you should be good to go.

flavia13, 2018-07-13:
Sorry, I mean If delete that method it compiles!!

closed 2018-07-23 by flavia13