whoenig/crazyflie_ros #105: Checksum md5sum mismatch when using Joystick for takeoff and landing

cjbanks, 2018-07-19:
I am having a similar issue to this: #100, my program works when I run it for one quadcopter but as soon as I add more than one or two I get the following error:  
[ERROR] [1532039584.691694366]: client wants service /crazyflie0/takeoff to have md5sum d41d8cd98f00b204e9800998ecf8427e, but it has b665b6c83a196e4774268cc26329b159. Dropping connection.
[ERROR] [1532039584.692239858]: client wants service /crazyflie3/takeoff to have md5sum d41d8cd98f00b204e9800998ecf8427e, but it has b665b6c83a196e4774268cc26329b159. Dropping connection.
[ERROR] [1532039584.692772]: bad callback: <bound method Controller._joyChanged of <__main__.Controller instance at 0x7fa88cca1a70>>
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/quadcopteruser/crazyflie_ws/src/crazyflie_ros/crazyflie_demo/scripts/controller.py", line 46, in _joyChanged
    self._takeoff()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 505, in call
    raise ServiceException("unable to connect to service: %s"%e)

I have tried reinstalling the crazyflie_ros package and updating my ROS distribution (Kinetic) but neither have had any effect. I run all my nodes on one linux machine running Ubuntu 16.04 and connect to a windows server to communicate with the vicon system.  Any advice on what may be causing this issue?

whoenig, 2018-07-20:
Which launch file are you using? (Or provide some steps to reproduce the problem, please).

cjbanks, 2018-07-22:
I am using a custom launch file where I hover more than one quadcopter, they each have their own crazyradio and unique URI. Usually the launch file will run without problem until I press the joystick button for land or takeoff which causes the checksum error. I never get the issue with a single quadcopter. The problem occurs when I have more than one crazyflie node in the launch file and the more crazyflies I have, the more often the issue occurs. However, rolling back to the previous version of the crazyflie_ros package mentioned in "Flying Multiple UAVs Using ROS" fixed my problem.

lorenzol92, 2018-10-15:
Hi, I'm facing the same issue!
I'm using ROS on a virtual machine of linux 16.04
I would like to autonomously drive 4 CF's with the help of LPS (6 anchors) in ROS environment, but I got stuck.
I've followed the: https://www.bitcraze.io/getting-started ... yflie-2-0/ to update the firmware of all the 4 crazyflies.
I've followed the: https://www.bitcraze.io/getting-started ... ng-system/ for setting the Loco Positioning System in the right way.
I've launch the: dwm_loc_ekf_hover.launch with a 6 anchors setup in both TWR and TdOA mode, and it works!
In TdOA I've even tryed with 2 CF's and it works!
Now I'm trying to launch the: dwm_loc_ekf_swarm_hover.launch with 4 CF's but I found necessary to add another Crazyradio.
So I' ve put 2 CF's on one Crazyradio and 2 CF's on the other Crazyradio changing the dwm_loc_ekf_swarm_hover.launch with the appropriate URI:

uri0 radio://0/80/2M/E7E7E7E701
uri1 radio://1/70/2M/E7E7E7E702
uri2 radio://1/70/2M/E7E7E7E703
uri3 radio://0/80/2M/E7E7E7E704

Unexpectedly the result was something like this:
[image]
Could anyone help me in find what could be the problem?
Thank you very much.

P.S:
I'm pretty sure that the previous URI setup is correct because implementing it in the multi_teleop_xbox360.launch (https://github.com/whoenig/crazyflie_ros) I could manually drive the 4 CF's.

shmpwk, 2019-03-19:
Hi, @cjbanks and @lorenzol92 
I'm facing the similar issue. #124 
Have you already done with the problem?

shmpwk, 2019-03-23:
@cjbanks @lorenzol92 
OK, I've done, by changing the firmware.

mahdi65, 2019-10-02:
@shmpwk could you please tell me which version ? 

I tested with all 2019.09 , 2019.01 and 2018.01 versions and still experiencing same issue.

mahdi65, 2019-10-02:
Experiencing Similar issue !
I am using motion Analysis tracking system. successfully  hovered one Crazyflie 2.1 and Also trajectory movements after some changes to code and making it working with motion analysis system. 
However, when I am trying to hover two devices. I run into checksum Error ( Attached ScreenShot) 
I have tested with firmware versions :

- 2019.01

- 2018.01

- 2019.09

All shows same error messages. ( after pushing take off too many times JUST one quad start hovering) 

[image]

full detail log : https://pastebin.com/E7atN8pA
my modified multi_hover_vicon.launch : https://pastebin.com/s4vGP0hb
my rqt_graph : 
[image]

@whoenig  Please give me some suggestions to see whether works.