whoenig/crazyflie_ros #111: Cannot build a ROS workspace for crazyflie

lkyfly0765, 2018-10-30:
When I type the $ catkin_make,I received some errors.
Could anyone tell me what's wrong with crazyflie_cpp?
I have correctly installed the ROS Driver for Bitcraze Crazyflie.

eric@eric-ThinkPad-S2:~/crazyflie_ws$ catkin_make
Base path: /home/eric/crazyflie_ws
Source space: /home/eric/crazyflie_ws/src
Build space: /home/eric/crazyflie_ws/build
Devel space: /home/eric/crazyflie_ws/devel
Install space: /home/eric/crazyflie_ws/install
####
#### Running command: "cmake /home/eric/crazyflie_ws/src -DCATKIN_DEVEL_PREFIX=/home/eric/crazyflie_ws/devel -DCMAKE_INSTALL_PREFIX=/home/eric/crazyflie_ws/install -G Unix Makefiles" in "/home/eric/crazyflie_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/eric/crazyflie_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/kinetic
-- This workspace overlays: /opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/eric/crazyflie_ws/build/test_results
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.14
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 5 packages in topological order:
-- ~~  - crazyflie (metapackage)
-- ~~  - crazyflie_description
-- ~~  - crazyflie_demo
-- ~~  - crazyflie_controller
-- ~~  - crazyflie_driver
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin metapackage: 'crazyflie'
-- ==> add_subdirectory(crazyflie_ros/crazyflie)
-- +++ processing catkin package: 'crazyflie_description'
-- ==> add_subdirectory(crazyflie_ros/crazyflie_description)
-- +++ processing catkin package: 'crazyflie_demo'
-- ==> add_subdirectory(crazyflie_ros/crazyflie_demo)
-- +++ processing catkin package: 'crazyflie_controller'
-- ==> add_subdirectory(crazyflie_ros/crazyflie_controller)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- +++ processing catkin package: 'crazyflie_driver'
-- ==> add_subdirectory(crazyflie_ros/crazyflie_driver)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by "crazyflie_cpp"
  with any of the following names:

    crazyflie_cppConfig.cmake
    crazyflie_cpp-config.cmake

  Add the installation prefix of "crazyflie_cpp" to CMAKE_PREFIX_PATH or set
  "crazyflie_cpp_DIR" to a directory containing one of the above files.  If
  "crazyflie_cpp" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  crazyflie_ros/crazyflie_driver/CMakeLists.txt:7 (find_package)

-- Could not find the required component 'crazyflie_cpp'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "crazyflie_cpp"
  with any of the following names:

    crazyflie_cppConfig.cmake
    crazyflie_cpp-config.cmake

  Add the installation prefix of "crazyflie_cpp" to CMAKE_PREFIX_PATH or set
  "crazyflie_cpp_DIR" to a directory containing one of the above files.  If
  "crazyflie_cpp" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  crazyflie_ros/crazyflie_driver/CMakeLists.txt:7 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/eric/crazyflie_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/eric/crazyflie_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

lkyfly0765, 2018-10-30:
How can I figure it out? Thanks!

whoenig, 2018-10-31:
You need to initialize and update the submodules, see https://github.com/whoenig/crazyflie_ros#installation.

lkyfly0765, 2018-11-01:
> You need to initialize and update the submodules, see https://github.com/whoenig/crazyflie_ros#installation.

I reinstalled the Ubuntu 16.04 and I have followed the instruction of the installation.When I initialize the submodules,it showed the following:

wangyubin@wangyubin-VLT-WX0:~$ git clone https://github.com/whoenig/crazyflie_ros.git
Cloning into 'crazyflie_ros'...
remote: Enumerating objects: 2, done.
remote: Counting objects: 100% (2/2), done.
remote: Compressing objects: 100% (2/2), done.
remote: Total 1494 (delta 0), reused 1 (delta 0), pack-reused 1492
Receiving objects: 100% (1494/1494), 742.94 KiB | 261.00 KiB/s, done.
Resolving deltas: 100% (873/873), done.
Checking connectivity... done.
wangyubin@wangyubin-VLT-WX0:~$ cd crazyflie_ros
wangyubin@wangyubin-VLT-WX0:~/crazyflie_ros$ git submodule init
Submodule 'crazyflie_cpp' (https://github.com/whoenig/crazyflie_cpp.git) registered for path 'crazyflie_cpp'
Submodule 'crazyflie_tools' (https://github.com/whoenig/crazyflie_tools) registered for path 'crazyflie_tools'

whoenig, 2018-11-01:
Seems like you forgot `git submodule update`?

lkyfly0765, 2018-11-02:
> Seems like you forgot `git submodule update`?

It seems that I clone the code to the wrong folder.
The issue had been solved,and Thank you again !

closed 2018-11-02 by lkyfly0765