whoenig/crazyflie_ros #113: receive  an error while catkin_make to your package

mandelyoni, 2018-11-28:
Hi Wolfgang,

I'm following your article "Flying Multiple UAVs Using ROS" and receive  an error while catkin_make to your package:

CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "crazyflie_cpp"
  with any of the following names:

    crazyflie_cppConfig.cmake
    crazyflie_cpp-config.cmake

Any idea?

Thanks,  Yoni.

mandelyoni, 2018-11-28:
Sorry,

Just saw your new installation instructions here..

mandelyoni, 2018-12-01:
Hi, yes it does.
By the way, we're trying to pass VRPN messages from Motive to a virtual machine running ROS, but it's not quite working. Are you familiar with a short tutorial on this issue (configurations of the motive and the VM)?
Thank you,  Yoni.

whoenig, 2018-12-02:
I don't have motive myself, so I cant comment on that problem. One general hint might be to run some network tests (ping each other, traceroute, etc.).

closed 2018-12-02 by whoenig