whoenig/crazyflie_ros #114: Roslaunch vrpn_hover.launch occurs some errors

lkyfly0765, 2018-12-13:
Hi Whoenig:
We are going to do some research on crazyflies, and we have the Optitrack as the mocap system.
I following the instructions which provide in Flying Multiple UAVs Using ROS, and we are trying to let the single crazyflie to hover at a specified coordinate.
I have correctly configured the network on my laptop which runs the Ubuntu 14.05 caring the ROS system, and I have established the vrpn connections between the server computer which runs the Motive and my computer.The rigid body data can be attained by the ROS system.
Then I use eric@eric-ThinkPad-S2:~$ roslaunch crazyflie_demo hover_vrpn.launch the Rviz open as expect, and the crazyflie connected to the Crazyradio successfully. The I press the green button on the Xbox360 joystick, it occurs some errors, and the connection between Crazyflie and crazyradio lost.

HERE IS THE LOG I COPIED FROM THE TERMINAL
eric@eric-ThinkPad-S2:~$ roslaunch crazyflie_demo hover_vrpn.launch 
... logging to /home/eric/.ros/log/3d764118-feaf-11e8-a4ec-34f64bc13130/roslaunch-eric-ThinkPad-S2-18643.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
xacro.py is deprecated; please use xacro instead
started roslaunch server http://eric-ThinkPad-S2:46519/

SUMMARY
========

PARAMETERS
 * /crazyflie/controller/PIDs/X/integratorMax: 0.1
 * /crazyflie/controller/PIDs/X/integratorMin: -0.1
 * /crazyflie/controller/PIDs/X/kd: 20.0
 * /crazyflie/controller/PIDs/X/ki: 2.0
 * /crazyflie/controller/PIDs/X/kp: 40.0
 * /crazyflie/controller/PIDs/X/maxOutput: 10.0
 * /crazyflie/controller/PIDs/X/minOutput: -10.0
 * /crazyflie/controller/PIDs/Y/integratorMax: 0.1
 * /crazyflie/controller/PIDs/Y/integratorMin: -0.1
 * /crazyflie/controller/PIDs/Y/kd: -20.0
 * /crazyflie/controller/PIDs/Y/ki: -2.0
 * /crazyflie/controller/PIDs/Y/kp: -40.0
 * /crazyflie/controller/PIDs/Y/maxOutput: 10.0
 * /crazyflie/controller/PIDs/Y/minOutput: -10.0
 * /crazyflie/controller/PIDs/Yaw/integratorMax: 0.0
 * /crazyflie/controller/PIDs/Yaw/integratorMin: 0.0
 * /crazyflie/controller/PIDs/Yaw/kd: -20.0
 * /crazyflie/controller/PIDs/Yaw/ki: 0.0
 * /crazyflie/controller/PIDs/Yaw/kp: -200.0
 * /crazyflie/controller/PIDs/Yaw/maxOutput: 200.0
 * /crazyflie/controller/PIDs/Yaw/minOutput: -200.0
 * /crazyflie/controller/PIDs/Z/integratorMax: 1000.0
 * /crazyflie/controller/PIDs/Z/integratorMin: -1000.0
 * /crazyflie/controller/PIDs/Z/kd: 6000.0
 * /crazyflie/controller/PIDs/Z/ki: 3500.0
 * /crazyflie/controller/PIDs/Z/kp: 5000.0
 * /crazyflie/controller/PIDs/Z/maxOutput: 60000.0
 * /crazyflie/controller/PIDs/Z/minOutput: 10000.0
 * /crazyflie/controller/frame: crazyflie
 * /crazyflie/controller/worldFrame: world
 * /crazyflie/crazyflie_add/enable_logging: False
 * /crazyflie/crazyflie_add/enable_logging_battery: True
 * /crazyflie/crazyflie_add/enable_logging_imu: True
 * /crazyflie/crazyflie_add/enable_logging_magnetic_field: True
 * /crazyflie/crazyflie_add/enable_logging_packets: True
 * /crazyflie/crazyflie_add/enable_logging_pressure: True
 * /crazyflie/crazyflie_add/enable_logging_temperature: True
 * /crazyflie/crazyflie_add/pitch_trim: 0
 * /crazyflie/crazyflie_add/roll_trim: 0
 * /crazyflie/crazyflie_add/tf_prefix: crazyflie
 * /crazyflie/crazyflie_add/uri: radio://0/99/2M
 * /crazyflie/crazyflie_add/use_ros_time: True
 * /crazyflie/joy/dev: /dev/input/js0
 * /crazyflie/joystick_controller/use_crazyflie_controller: True
 * /crazyflie/pose/name: goal
 * /crazyflie/pose/rate: 30
 * /crazyflie/pose/x: 0
[... 116 lines cut ...]
[INFO] [1544693919.680637]: found takeoff service
[ INFO] [1544693949.963402421]: Takeoff requested!
[ERROR] [1544693949.963801181]: Exception thrown while processing service call: Could not find a connection between 'world' and 'crazyflie' because they are not part of the same tree.Tf has two or more unconnected trees.
[ERROR] [1544693949.966671]: bad callback: <bound method Controller._joyChanged of <__main__.Controller instance at 0x7f01a26333f8>>
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/eric/crazyflie_ws/src/crazyflie_ros/crazyflie_demo/scripts/controller.py", line 46, in _joyChanged
    self._takeoff()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 515, in call
    responses = transport.receive_once()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 727, in receive_once
    p.read_messages(b, msg_queue, sock)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 353, in read_messages
    self._read_ok_byte(b, sock)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 336, in _read_ok_byte
    raise ServiceException("service [%s] responded with an error: %s"%(self.resolved_name, str))
ServiceException: service [/crazyflie/takeoff] responded with an error: Could not find a connection between 'world' and 'crazyflie' because they are not part of the same tree.Tf has two or more unconnected trees.

terminate called after throwing an instance of 'tf2::ConnectivityException'
  what():  Could not find a connection between 'world' and 'crazyflie' because they are not part of the same tree.Tf has two or more unconnected trees.
[crazyflie/controller-5] process has died [pid 18696, exit code -6, cmd /home/eric/crazyflie_ws/devel/lib/crazyflie_controller/crazyflie_controller __name:=controller __log:=/home/eric/.ros/log/3d764118-feaf-11e8-a4ec-34f64bc13130/crazyflie-controller-5.log].
log file: /home/eric/.ros/log/3d764118-feaf-11e8-a4ec-34f64bc13130/crazyflie-controller-5*.log

whoenig, 2018-12-14:
The error message 

> Could not find a connection between 'world' and 'crazyflie' because they are not part of the same tree.Tf has two or more unconnected trees.

indicates that there is an issue with your transformations. You can visualize the transformations using `rosrun tf view_frames && evinces frames.pdf`. Is there a world frame and a crazyflie frame that are connected by an arrow?

lkyfly0765, 2018-12-17:
I have used the command you provided and I got a pdf file which contains a arrow connect the world and cf1.

whoenig, 2018-12-18:
You add the crazyflie using the name "crazyflie", so your rigid body should be named "crazyflie" and not "cf1".

lkyfly0765, 2018-12-23:
I have figure it out. The key is that I need to change the channel and the brand width in all related launchfiles.
Another problem I am facing it that when the  crazyflie take off using the **hover-vrpn.launch** file, it is not stable. In the launch file, the hover point is set to (0,0,0.5), but actually the crazyflie float at about (0,0,0.3) to (0,0,1). It’s altitude is not stable. What’s wrong with it?  Does this problem relate to the PID parameters?

whoenig, 2019-01-04:
The crazyflie now has an onboard position controller that can follow waypoints or trajectories at a much higher accuracy than the off-board controller you are using now. There is no documentation in crazylie_ros for that yet (although it is described in some of the older issues), but you can check out the crazyswarm project: https://crazyswarm.readthedocs.io (which also works if you are interested in just using a few CFs).

shmpwk, 2019-03-20:
Hi I have a same error. 

> Could not find a connection between 'world' and 'crazyflie' because they are not part of the same tree.Tf has two or more unconnected trees.

[image]

How did you change 

>  to change the channel and the bandwidth in all related launchfiles.

?

lkyfly0765, 2019-03-20:
Hi Shumpei:

During my test, the solution is that in the launch file, in line 8, the ‘<arg name=“frame” default =“crazyflie” />’like, you should confirm that the <default =“ xxxx” >match the rigid body name in Optitrack Motive application.

If you have any questions please contact me and I am very glad to solve it together.

Best regards,
Eric Liu

shmpwk, 2019-03-20:
Thank you for your quick response, Eric.
I'll try it tomorrow.

Best regards, Shumpei

lkyfly0765, 2019-03-20:
It’s my pleasure.

From: Shumpei Wakabayashi
Sent: Wednesday, March 20, 2019 9:58 PM
To: whoenig/crazyflie_ros
Cc: lkyfly0765; Mention
Subject: Re: [whoenig/crazyflie_ros] Roslaunch vrpn_hover.launch occurs someerrors (#114)

Thank you for your quick response, Eric.
I'll try it tomorrow.
Best regards, Shumpei

shmpwk, 2019-03-20:
Hi, Eric,

I changed as you said, but cannot connect 'world' and 'crazyflie1'.
Looking at tf view_frame, there are still two 'crazyflie1'.
Did you change other program?

lkyfly0765, 2019-03-21:
I cannot confirm where the error comes from.
I will check my launch file and give you a reply a few hours later, I am out now. 
It seems that your view_frame pdf file is now same as mine. And I will check it at noon. 

Best Regard,
Eric Liu

lkyfly0765, 2019-03-21:
Hi,
I have check the frames.pdf and I send it in the attachment.
It seems that your connection between world and crazyflie1 has someting wrong.
Please check: 1.You have already run the vrpn client using  roslaunch ros-vrpn-client sample.launch server:xxx.xxx.xxx.xxx:3883
                       2. In the hover_vrpn.launch file, line 4 and line 5 ,the IP and the port has been correctly configured. In line 7,the crazyflies's uri has been correctly configured.

                       3. The firewall both on your conputer and the Optitrack server computer has been shutdown completely.
                       4. In the Optitrack Motive, at the data streaming board, you have change the axis to Z-up.

That's all I can infer.

shmpwk, 2019-03-21:
Thank you very much for super thorough explaining!

I chacked 1 by 1, and found all things correct. So, the cause of the problem is not what you told me.

Actually, I use Cortex as the motion capture system. It can also use VRPN connection. 
I'll chack its program carefully again.