whoenig/crazyflie_ros #115: Access estimates of the Complementary filter

JohnSilverFarmer, 2018-12-14:
Hello,

I'm trying to access the predicted state of the complementary filter (that's running on the firmeware) using crazyflie_ros. It's possible to access the output of the kalman filter using log variables i.e.:

>         genericLogTopics: ["log_kfpos", "log_kfqt"]
>          genericLogTopicFrequencies: [30, 30]
>          genericLogTopic_log_kfpos_Variables: ["kalman.stateX", "kalman.stateY", "kalman.stateZ"]
>          genericLogTopic_log_kfqt_Variables: ["kalman.q0", "kalman.q1", "kalman.q2", "kalman.q3"]

The kalman filter seems to implement the log protocol i.e. https://github.com/bitcraze/crazyflie-firmware/blob/a203d748e139b82854059423e04bed7b0992ebd8/src/modules/src/estimator_kalman.c#L1499 however the complementary filter does not?

What would I have to do access the values of the complementary filter on a ros topic?

Thanks!

whoenig, 2018-12-14:
For the position state estimate, you can use "stateEstimate.x" etc., which will work no matter what filter is currently active (see https://github.com/bitcraze/crazyflie-firmware/blob/master/src/modules/src/stabilizer.c#L488-L492). For the other states (such as quaternion), you will need to add entries in the firmware accordingly (I suggest putting it in stabilizer.c).

JohnSilverFarmer, 2018-12-17:
That seems to work - thanks for getting back to me!

closed 2018-12-18 by whoenig