whoenig/crazyflie_ros #117: crazyflie 2.0 about hover using optitrack

Marcosz-zhuhe, 2019-01-06:
Hello,I have launch the vrpn_client_ros and there is no error.I have also set up a connection between ROS and Optitrack. But crazyflie can't hover and there is no reaction no matter how I press the key on the Xbox360. What should I do?Thanks.
[image]

whoenig, 2019-01-08:
This might indicate that the CF does not receive its pose update. Do you have the LED ring installed (it should show green if the communication works and red otherwise).

Marcosz-zhuhe, 2019-01-09:
Thanks,Mr whoenig.The CF has responded，but it doesn't hover and flies aslantly.When I use X360 controller to control the CF directly,one of the propellers always stops to rotate.When I change another CF,it happens too.Is there something wrong with the crazyflies?Or it is the problem of algorithm.

Marcosz-zhuhe, 2019-01-09:
@whoenig When I launch the multi_hover_vrpn.launch.There is an error which I don't kown how to correct.
[image]
[image]

Marcosz-zhuhe, 2019-01-11:
I find that no matter how many crazyflies,they couldn't hover. And they all had a obviously horizontal move.When I changed the PID parameters,it still happened.Although I tried my best to find out the reason, it failed.Welcome to comment on it.My English is not good,and if there is anything wrong,please forgive me.

whoenig, 2019-01-13:
> When I use X360 controller to control the CF directly,one of the propellers always stops to rotate.

This is not normal. Does this happen if you fly using the official cfclient or when using crazyflie_ros?

The movement issues you describe indicate that there is an issue with the motion capture. Try logging the state estimate, does it update correctly when you manually move the CF?

whoenig, 2019-01-18:
@Marcosz-zhuhe were you able to solve your problem?

Marcosz-zhuhe, 2019-01-19:
@whoenig It's kind of you.But I didn't solve my problem.My classmates in another group made crazyflie hover successfully.I will ask him about it when I go back to the university in the next semester.Thank you again.