whoenig/crazyflie_ros #118: rotation while landing

wydmynd, 2019-02-05:
I am using the high level controller with flow deck. 
using the goTo service for motion (all motion relative) and land service for landing.
example - 
cf.goTo(goal=[0.0, 0.2, 0.0], yaw=0, duration=def_duration, relative=True)

when I use  yaw commands, upon landing, **the drone tries to rotate to initial yaw during landing.**
this does not apply to x,y motion (lands in place)
so it is landing in "corkscrew" motion.
any suggestion if this is a problem in the crazyflie_server , I did not encounter this problem when using high level commands in cfclient.
thanks!

whoenig, 2019-02-06:
The high-level land command doesn't take a desired yaw as input (and rather uses the current yaw as goal orientation), see https://github.com/bitcraze/crazyflie-firmware/blob/master/src/modules/src/crtp_commander_high_level.c#L324.

Could you perhaps share the following:
1. Which firmware version are you using?
2. Could you provide how you tested the functionality in the cfclient? Did you try that using the same firmware version?
3. I assume you are referring here to Flow V2, like in issue #119?

wydmynd, 2019-02-07:
1. using bitcraze firmware.
2. functionality via cfclient was tested the ranging tutorial script [https://www.bitcraze.io/getting-started-with-stem-ranging-bundle/](url)  (in retrospect need to re-check that it uses high level, will do it after weekend)
3. this happens with flow deck V1 and V2

whoenig, 2019-02-08:
The motionCommander uses the low-level commands (velocity commands), it seems. So this is likely a bug in the high-level mode. I'll need to test this (hopefully sometime next week).