whoenig/crazyflie_ros #120: `what(): LIBUSB_ERROR_TIMEOUT ` Error on Roadrunner (Only on USB)

ashwinsushil, 2019-02-15:
Hello Everyone,

I'm trying to run this launch file,
```
roslaunch bitcraze_lps_estimator dwm_loc_ekf_hover.launch uri:=usb://0
```
(With all the copter parts removed ofcourse)

But this error is being thrown **every now and then**,
```
terminate called after throwing an instance of 'std::runtime_error'                                                                │                                                                                                                                   
  what():  LIBUSB_ERROR_TIMEOUT                                                                                                    │                                                                                                                                   
[crazyflie_server-6] process has died [pid 24065, exit code -6, cmd /home/unmanned/Workspace/devel/lib/crazyflie_driver/crazyflie_s│                                                                                                                                   
erver __name:=crazyflie_server __log:=/home/unmanned/.ros/log/b02b0af4-3082-11e9-8c3f-faa083e2307f/crazyflie_server-6.log].        │                                                                                                                                   
log file: /home/unmanned/.ros/log/b02b0af4-3082-11e9-8c3f-faa083e2307f/crazyflie_server-6*.log     
```
After couple of tries of running the same launch file, another error is being thrown,
```
[INFO] [1550167626.114321]: Anchor 0 at [0.0, 0.0, 1.93]                                                                           │                                                                                                                                   
terminate called after throwing an instance of 'std::runtime_error'                                                                │                                                                                                                                   
  what():  LIBUSB_ERROR_TIMEOUT                                                                                                    │                                                                                                                                   
Traceback (most recent call last):                                                                                                 │                                                                                                                                   
  File "/home/unmanned/Workspace/src/lps-ros/scripts/lps_ekf_bridge.py", line 77, in <module>                                      │                                                                                                                                   
    update_params([name + 'x', name + 'y', name + 'z'])                                                                            │                                                                                                                                   
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__                          │                                                                                                                                   
    return self.call(*args, **kwds)                                                                                                │                                                                                                                                   
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 525, in call                              │                                                                                                                                   
    raise ServiceException("transport error completing service call: %s"%(str(e)))                                                 │                                                                                                                                   
rospy.service.ServiceException: transport error completing service call: unable to receive data from sender, check sender's logs fo│                                                                                                                                   
r details 
```

**And after couple more tries, it WORKS and I'm able to get the pose on the Roadrunner! This is a USB realated issue i presume as I don't get this error when Roadrunner is connected over crazyradio.** 
I have uploaded the same issue in the bitcraze forum:
https://forum.bitcraze.io/viewtopic.php?f=2&t=3350

**Firmware version on Roadrunner:** 
SYS: Build 4:1762deae7f25 (2019.02 +4) CLEAN
SYS: I am 0x3934323530374718003F0023 and I have 1024KB of flash!
CFGBLK: v1, verification [OK]
DECK_INFO: DECK_FORCE=bcDWM1000 found
DECK_INFO: compile-time forced driver bcDWM1000 added
DECK_CORE: 1 deck(s) found
DECK_CORE: Calling INIT on driver bcDWM1000 for deck 0
IMU: BMI088 Gyro I2C connection [OK].
IMU: BMI088 Accel I2C connection [OK]
ESTIMATOR: Using Kalman (2) estimator
CONTROLLER: Using PID (1) controller
EEPROM: I2C connection [OK].
DECK_CORE: Deck 0 test [OK].
SYS: Free heap: 11432 bytes
TDOA3: 2D positioning enabled at 0.300000 m height
DWM: Automatic mode: detected TDoA3

**LPS node firmware**: lps-node-firmware-2018.10.dfu

**System details:**
Ubuntu 16.04
ROS Kinetic
ROS driver for roadrunner/crazyflie from https://github.com/whoenig/crazyflie_ros

Thanks in advance,
Ash

ashwinsushil, 2019-02-15:
comment from @whoenig  on bitcraze forum

> Feel free to open an issue in github for crazyflie_ros or crazyflie_cpp. I have seen this issue before with USB and a regular Crazyflie. Here is a change that reduces the probability of issues occurring, but does not entirely fix the problem: https://github.com/whoenig/crazyflie_cp ... dd6e48b851.

whoenig, 2019-02-15:
From the forum, the fix only changes the kind of error message, and does not improve reliability drastically.

> Like you mentioned it has not solved the issue and 7 out of 10 times (approx :D), I am not able to get the pose.
> 
> However, now I don't have `what(): LIBUSB_ERROR_TIMEOUT` error. Instead its the below given warning and no pose..
> 
> [ WARN] [1550222925.240533190]: Link Quality low (0.000000)

whoenig, 2019-02-19:
Just as update here: I'll receive a roadrunner from Bitcraze to investigate. There are also two firmware bugs related to this issue open now: https://github.com/bitcraze/crazyflie-firmware/issues/393 and https://github.com/bitcraze/crazyflie-firmware/issues/394.

AlexisTM, 2019-03-08:
This definitely improved the behaviour. I disabled the Watchdog in the firmware and I do not remember having the TIMEOUT again after applying this patch.

The rest of the issue likely lies in Crazyflie Firmware.

evoggy, 2019-05-03:
The issues bitcraze/crazyflie-firmware#393 and bitcraze/crazyflie-firmware#394 should now be fixed so hopefully this should also work now. Could someone give it a try and let me know if it's fixed?

wydmynd, 2019-06-02:
I had success / workaround by re-typing the udev commands - 
sudo udevadm control --reload-rules
sudo udevadm trigger

ashwinsushil, 2019-11-07:
Hello everyone, recently I tried to used the roadrunner to get the position from it using the crazyflie_ros package by running 

`roslaunch crazyflie_demo position.launch`

without of course setting URI to `usb://0`.

Unfortunately, I have the same error 
`
terminate called after throwing an instance of 'std::runtime_error'                                   │
  what():  LIBUSB_ERROR_TIMEOUT                                                                       │
[crazyflie_server-2] process has died [pid 11760, exit code -6, cmd /home/unmanned/Workspace/devel/lib│
/crazyflie_driver/crazyflie_server __name:=crazyflie_server __log:=/home/unmanned/.ros/log/5f32f6f6-01│
43-11ea-8767-185e0f1c20e6/crazyflie_server-2.log].                                                    │
log file: /home/unmanned/.ros/log/5f32f6f6-0143-11ea-8767-185e0f1c20e6/crazyflie_server-2*.log 
`
I tried disabling the watchdog timer as @AlexisTM mentioned, but it didn't help.
@evoggy I don't really think this issued is fixed.

I am using the latest firmware from the master branch of [bitcraze-firmware](https://github.com/bitcraze/crazyflie-firmware)

AlexisTM, 2019-11-07:
Did you update the crazyflie to the last master branch?