whoenig/crazyflie_ros #121: multi_waypoint_vicon.launch. takeoff service question for 4 crazyflies

Flaret, 2019-02-25:
Hi Whoenig,

I'm trying to launch 4 crazyflies (using two crazyradios) based on multi_waypoint_vicon.launch

I can launch 1 or 2 crazyflies, but when I go to 3 and 4, takeoff service parameters change to 1-2 of them:
-from
`rosservice call /crazyflie/takeoff "{}"`

-to
`rosservice call /crazyflie3/takeoff "groupMask: 0
height: 0.0
duration:
  secs: 0
  nsecs: 0"`

And crazyflies takeoff only with the first one.

How this can be solved?

whoenig, 2019-02-26:
If you have 4 CFs, I suggest using the Crazyswarm (https://crazyswarm.readthedocs.io/en/latest/), which comes with some additional features. For example, it will allow you to execute the takeoff synchronously using broadcasts.

As for this problem: You should pass in a height and duration - a zero duration might even trigger some strange behavior in the firmware.

Flaret, 2019-02-27:
Thank you Whoenig.

Is it possible to use Crazyswarm without firmware modifications?

whoenig, 2019-02-28:
Yes, as long as you run a fairly recent official firmware, you are good to go.

Flaret, 2019-03-04:
Thank you Whoenig.

closed 2019-03-04 by Flaret