whoenig/crazyflie_ros #122: Nothing appeared on rviz

shmpwk, 2019-03-03:
Hi, I'm confusing about nothing appearing on rviz. 

I use ros kinetic on ubuntu.
After ```roscore```,
```roslaunch crazyflie_demo teleop_xbox360.launch uri:=radio://0/2/2M```
I set uri based on scanning.
It booted rviz and rqt_plot, but nothing appeared. It seemed it didn’t connect to crazyfile.
So I booted ```cfclient```, and It successfully reflect on the tilt of the real crazyflie. 
The error is below.

```
roslaunch crazyflie_demo teleop_xbox360.launch uri:=radio://0/2/2M
Successfully booted rviz and 2 rqt_plot, but it seemed it didn’t connect to cf.
WARNING: Package "cfclient" does not follow the version conventions. It should not contain leading zeros (unless the number is 0).
... logging to /home/drone/.ros/log/44e9810c-3b28-11e9-a5c0-28c63f262be5/roslaunch-drone-GL502VSK-4074.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

WARNING: Package "cfclient" does not follow the version conventions. It should not contain leading zeros (unless the number is 0).
started roslaunch server http://localhost:46293/

SUMMARY
========

PARAMETERS
 * /crazyflie/crazyflie_add/enable_logging: True
 * /crazyflie/crazyflie_add/enable_logging_battery: True
 * /crazyflie/crazyflie_add/enable_logging_imu: True
 * /crazyflie/crazyflie_add/enable_logging_magnetic_field: True
 * /crazyflie/crazyflie_add/enable_logging_packets: True
 * /crazyflie/crazyflie_add/enable_logging_pressure: True
 * /crazyflie/crazyflie_add/enable_logging_temperature: True
 * /crazyflie/crazyflie_add/pitch_trim: 0
 * /crazyflie/crazyflie_add/roll_trim: 0
 * /crazyflie/crazyflie_add/tf_prefix: crazyflie
 * /crazyflie/crazyflie_add/uri: radio://0/2/2M
 * /crazyflie/crazyflie_add/use_ros_time: True
 * /crazyflie/joy/dev: /dev/input/js0
 * /crazyflie/quadrotor_teleop/x_axis: 5
 * /crazyflie/quadrotor_teleop/x_velocity_max: 30
 * /crazyflie/quadrotor_teleop/y_axis: 4
 * /crazyflie/quadrotor_teleop/y_velocity_max: -30
 * /crazyflie/quadrotor_teleop/yaw_axis: 1
 * /crazyflie/quadrotor_teleop/yaw_velocity_max: -200
 * /crazyflie/quadrotor_teleop/z_axis: 2
 * /crazyflie/quadrotor_teleop/z_velocity_max: 60000
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /crazyflie/
    crazyflie_add (crazyflie_driver/crazyflie_add)
    crazyflie_demo_controller (crazyflie_demo/controller.py)
    joy (joy/joy_node)
    quadrotor_teleop (crazyflie_demo/quadrotor_teleop)
  /
    crazyflie_server (crazyflie_driver/crazyflie_server)
    rqt_plot2 (rqt_plot/rqt_plot)
    rqt_plot3 (rqt_plot/rqt_plot)
    rviz (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

WARNING: Package "cfclient" does not follow the version conventions. It should not contain leading zeros (unless the number is 0).
process[crazyflie_server-1]: started with pid [4091]
process[crazyflie/crazyflie_add-2]: started with pid [4092]
process[crazyflie/joy-3]: started with pid [4103]
process[crazyflie/quadrotor_teleop-4]: started with pid [4113]
process[crazyflie/crazyflie_demo_controller-5]: started with pid [4132]
process[rviz-6]: started with pid [4155]
process[rqt_plot2-7]: started with pid [4168]
[ INFO] [1551339409.079220665]: wait_for_service /add_crazyflie
[ INFO] [1551339409.080276490]: found /add_crazyflie
process[rqt_plot3-8]: started with pid [4183]
[ INFO] [1551339409.085844694]: Adding radio://0/2/2M as crazyflie with trim(0.000000, 0.000000). Logging: 1, Parameters: 1, Use ROS time: 1
[ INFO] [1551339409.274221556]: Opened joystick: /dev/input/js0. deadzone_: 0.050000.
[crazyflie/crazyflie_add-2] process has finished cleanly
log file: /home/drone/.ros/log/44e9810c-3b28-11e9-a5c0-28c63f262be5/crazyflie-crazyflie_add-2*.log
terminate called after throwing an instance of 'std::runtime_error'
  what():  timeout
[crazyflie_server-1] process has died [pid 4091, exit code -6, cmd /home/drone/catkin_ws/devel/lib/crazyflie_driver/crazyflie_server __name:=crazyflie_server __log:=/home/drone/.ros/log/44e9810c-3b28-11e9-a5c0-28c63f262be5/crazyflie_server-1.log].
log file: /home/drone/.ros/log/44e9810c-3b28-11e9-a5c0-28c63f262be5/crazyflie_server-1*.log
QXcbWindow: Unhandled client message: "_COMPIZ_TOOLKIT_ACTION"
```

How can I fix it?

whoenig, 2019-03-04:
There is an error `Exception thrown while processing service call: LIBUSB_ERROR_BUSY`, which indicates that you have another process running that uses the Crazyradio? Make sure you disconnect from the cfclient first. Also, it seems like you installed cfclient inside your ROS workspace (there are a bunch of warnings: `WARNING: Package "cfclient" does not follow the version conventions.`). Note that cfclient has nothing to do with ROS and should be installed separately, outside of any ROS workspace.

shmpwk, 2019-03-05:
Thank you for your reply. 
I've finished what you advised. But it appeared another error.

```
drone@drone-GL502VSK:~$ roslaunch crazyflie_demo teleop_xbox360.launch uri:=radio://0/80/2M
... logging to /home/drone/.ros/log/013f3330-3edd-11e9-a296-28c63f262be5/roslaunch-drone-GL502VSK-5675.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:35461/

SUMMARY
========

PARAMETERS
 * /crazyflie/crazyflie_add/enable_logging: True
 * /crazyflie/crazyflie_add/enable_logging_battery: True
 * /crazyflie/crazyflie_add/enable_logging_imu: True
 * /crazyflie/crazyflie_add/enable_logging_magnetic_field: True
 * /crazyflie/crazyflie_add/enable_logging_packets: True
 * /crazyflie/crazyflie_add/enable_logging_pressure: True
 * /crazyflie/crazyflie_add/enable_logging_temperature: True
 * /crazyflie/crazyflie_add/pitch_trim: 0
 * /crazyflie/crazyflie_add/roll_trim: 0
 * /crazyflie/crazyflie_add/tf_prefix: crazyflie
 * /crazyflie/crazyflie_add/uri: radio://0/80/2M
 * /crazyflie/crazyflie_add/use_ros_time: True
 * /crazyflie/joy/dev: /dev/input/js0
 * /crazyflie/quadrotor_teleop/x_axis: 5
 * /crazyflie/quadrotor_teleop/x_velocity_max: 30
 * /crazyflie/quadrotor_teleop/y_axis: 4
 * /crazyflie/quadrotor_teleop/y_velocity_max: -30
 * /crazyflie/quadrotor_teleop/yaw_axis: 1
 * /crazyflie/quadrotor_teleop/yaw_velocity_max: -200
 * /crazyflie/quadrotor_teleop/z_axis: 2
 * /crazyflie/quadrotor_teleop/z_velocity_max: 60000
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /crazyflie/
    crazyflie_add (crazyflie_driver/crazyflie_add)
    crazyflie_demo_controller (crazyflie_demo/controller.py)
    joy (joy/joy_node)
    quadrotor_teleop (crazyflie_demo/quadrotor_teleop)
  /
    crazyflie_server (crazyflie_driver/crazyflie_server)
    rqt_plot2 (rqt_plot/rqt_plot)
    rqt_plot3 (rqt_plot/rqt_plot)
    rviz (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

process[crazyflie_server-1]: started with pid [5694]
process[crazyflie/crazyflie_add-2]: started with pid [5695]
process[crazyflie/joy-3]: started with pid [5701]
process[crazyflie/quadrotor_teleop-4]: started with pid [5714]
process[crazyflie/crazyflie_demo_controller-5]: started with pid [5729]
process[rviz-6]: started with pid [5753]
process[rqt_plot2-7]: started with pid [5766]
process[rqt_plot3-8]: started with pid [5778]
[ INFO] [1551745895.414099749]: wait_for_service /add_crazyflie
[ INFO] [1551745895.415708567]: found /add_crazyflie
[ INFO] [1551745895.419800879]: Adding radio://0/80/2M as crazyflie with trim(0.000000, 0.000000). Logging: 1, Parameters: 1, Use ROS time: 1
[ INFO] [1551745895.434745866]: Opened joystick: /dev/input/js0. deadzone_: 0.050000.
[crazyflie/crazyflie_add-2] process has finished cleanly
log file: /home/drone/.ros/log/013f3330-3edd-11e9-a296-28c63f262be5/crazyflie-crazyflie_add-2*.log
terminate called after throwing an instance of 'std::runtime_error'
  what():  timeout
[crazyflie_server-1] process has died [pid 5694, exit code -6, cmd /home/drone/catkin_ws/devel/lib/crazyflie_driver/crazyflie_server __name:=crazyflie_server __log:=/home/drone/.ros/log/013f3330-3edd-11e9-a296-28c63f262be5/crazyflie_server-1.log].
log file: /home/drone/.ros/log/013f3330-3edd-11e9-a296-28c63f262be5/crazyflie_server-1*.log
```

I've  checked radio setting in the crazyflie2.yaml file, teleop_xbox360.launch file and, crazyflie_add.launch file.

whoenig, 2019-03-05:
Does `rosrun crazyflie_tools scan` show your quadrotor?

shmpwk, 2019-03-05:
I successfully showed arrow on rviz.
I tried scan command again, and found that the uri changed. So, I recorrect it on the files and command line. 
I don't know why uri changed, but thank you  again!

closed 2019-03-05 by whoenig