whoenig/crazyflie_ros #124: runtime_error when running roslaunch crazyflie_demo hover_vrpn.launch

shmpwk, 2019-03-13:
Hi Whoenig:
I'm trying your Crazyflie_demo. When running `roslaunch crazyflie_demo hover_vrpn.launch`,  it shows `'std::runtime_error'`.  
When I checked whether it received the data from motion capture system by `rostopic echo /vrpn_client_node/drone2/pose`, it successfully received. So VRPN client is not the problem, but looking at rqt_graph, it fails to connect crazyflie_server.
[image]
How can I fix it? I alreadyconfigured IP address.

`drone@drone-GL502VSK:~$ roslaunch crazyflie_demo hover_vrpn.launch uri:=radio://0/80/250K/E7E7E7E701 frame:=crazyflie1 x:=0 y:=0 z:=0.5
... logging to /home/drone/.ros/log/7e59b5d4-4545-11e9-92af-2cfda1853ff1/roslaunch-drone-GL502VSK-5109.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
xacro.py is deprecated; please use xacro instead
started roslaunch server http://localhost:35851/

SUMMARY
========

PARAMETERS
 * /crazyflie/controller/PIDs/X/integratorMax: 0.1
 * /crazyflie/controller/PIDs/X/integratorMin: -0.1
 * /crazyflie/controller/PIDs/X/kd: 20.0
 * /crazyflie/controller/PIDs/X/ki: 2.0
 * /crazyflie/controller/PIDs/X/kp: 40.0
 * /crazyflie/controller/PIDs/X/maxOutput: 10.0
 * /crazyflie/controller/PIDs/X/minOutput: -10.0
 * /crazyflie/controller/PIDs/Y/integratorMax: 0.1
 * /crazyflie/controller/PIDs/Y/integratorMin: -0.1
 * /crazyflie/controller/PIDs/Y/kd: -20.0
 * /crazyflie/controller/PIDs/Y/ki: -2.0
 * /crazyflie/controller/PIDs/Y/kp: -40.0
 * /crazyflie/controller/PIDs/Y/maxOutput: 10.0
 * /crazyflie/controller/PIDs/Y/minOutput: -10.0
 * /crazyflie/controller/PIDs/Yaw/integratorMax: 0.0
 * /crazyflie/controller/PIDs/Yaw/integratorMin: 0.0
 * /crazyflie/controller/PIDs/Yaw/kd: -20.0
 * /crazyflie/controller/PIDs/Yaw/ki: 0.0
 * /crazyflie/controller/PIDs/Yaw/kp: -200.0
 * /crazyflie/controller/PIDs/Yaw/maxOutput: 200.0
 * /crazyflie/controller/PIDs/Yaw/minOutput: -200.0
 * /crazyflie/controller/PIDs/Z/integratorMax: 1000.0
 * /crazyflie/controller/PIDs/Z/integratorMin: -1000.0
 * /crazyflie/controller/PIDs/Z/kd: 6000.0
 * /crazyflie/controller/PIDs/Z/ki: 3500.0
 * /crazyflie/controller/PIDs/Z/kp: 5000.0
 * /crazyflie/controller/PIDs/Z/maxOutput: 60000.0
 * /crazyflie/controller/PIDs/Z/minOutput: 10000.0
 * /crazyflie/controller/crazyflies: [{'id': 1, 'chann...
 * /crazyflie/controller/frame: crazyflie1
 * /crazyflie/controller/worldFrame: world
 * /crazyflie/crazyflie_add/enable_logging: False
 * /crazyflie/crazyflie_add/enable_logging_battery: True
 * /crazyflie/crazyflie_add/enable_logging_imu: True
 * /crazyflie/crazyflie_add/enable_logging_magnetic_field: True
 * /crazyflie/crazyflie_add/enable_logging_packets: True
 * /crazyflie/crazyflie_add/enable_logging_pressure: True
 * /crazyflie/crazyflie_add/enable_logging_temperature: True
 * /crazyflie/crazyflie_add/pitch_trim: 0
 * /crazyflie/crazyflie_add/roll_trim: 0
 * /crazyflie/crazyflie_add/tf_prefix: crazyflie
 * /crazyflie/crazyflie_add/uri: radio://0/80/250K...
 * /crazyflie/crazyflie_add/use_ros_time: True
 * /crazyflie/joy/dev: /dev/input/js0
 * /crazyflie/joystick_controller/use_crazyflie_controller: True
 * /crazyflie/pose/name: goal
 * /crazyflie/pose/rate: 30
 * /crazyflie/pose/x: 0
 * /crazyflie/pose/y: 0
 * /crazyflie/pose/z: 0.5
 * /crazyflie/vrpn_client_node/child_frame_id: crazyflie1
 * /crazyflie/vrpn_client_node/frame_id: /world
 * /crazyflie/vrpn_client_node/ip: 169.254.91.127
 * /crazyflie/vrpn_client_node/port: 3883
 * /robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /crazyflie/
    baselink_broadcaster (tf/static_transform_publisher)
    controller (crazyflie_controller/crazyflie_controller)
    crazyflie_add (crazyflie_driver/crazyflie_add)
    joy (joy/joy_node)
    joystick_controller (crazyflie_demo/controller.py)
    pose (crazyflie_demo/publish_pose.py)
    vrpn_client_node (vrpn_client_ros/vrpn_client_node)
  /
    crazyflie_server (crazyflie_driver/crazyflie_server)
    rviz (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

process[crazyflie_server-1]: started with pid [5129]
process[crazyflie/crazyflie_add-2]: started with pid [5130]
process[crazyflie/joy-3]: started with pid [5131]
process[crazyflie/joystick_controller-4]: started with pid [5146]
process[crazyflie/controller-5]: started with pid [5165]
process[crazyflie/pose-6]: started with pid [5182]
[ INFO] [1552456497.368791450]: wait_for_service /add_crazyflie
process[crazyflie/baselink_broadcaster-7]: started with pid [5200]
[ INFO] [1552456497.370301235]: found /add_crazyflie
[ INFO] [1552456497.373163206]: Adding radio://0/80/250K/E7E7E7E701 as crazyflie with trim(0.000000, 0.000000). Logging: 1, Parameters: 0, Use ROS time: 1
process[crazyflie/vrpn_client_node-8]: started with pid [5214]
process[rviz-9]: started with pid [5235]
[ INFO] [1552456497.396014668]: Connecting to VRPN server at localhost:3883
[ INFO] [1552456497.567432395]: Opened joystick: /dev/input/js0. deadzone_: 0.050000.
[crazyflie/crazyflie_add-2] process has finished cleanly
log file: /home/drone/.ros/log/7e59b5d4-4545-11e9-92af-2cfda1853ff1/crazyflie-crazyflie_add-2*.log
[ INFO] [1552456498.396419693]: Connection established
terminate called after throwing an instance of 'std::runtime_error'
  what():  timeout
[crazyflie_server-1] process has died [pid 5129, exit code -6, cmd /home/drone/catkin_ws/devel/lib/crazyflie_driver/crazyflie_server __name:=crazyflie_server __log:=/home/drone/.ros/log/7e59b5d4-4545-11e9-92af-2cfda1853ff1/crazyflie_server-1.log].
log file: /home/drone/.ros/log/7e59b5d4-4545-11e9-92af-2cfda1853ff1/crazyflie_server-1*.log
`

whoenig, 2019-03-15:
The timeout error indicates that there is a problem with the connection to the Crazyflie, not to VRPN. Is your uri (`radio://0/80/250K`) correct? Can you connect to the CF with the uri using, for example, the cfclient?

shmpwk, 2019-03-15:
Thank you for your reply. Yes, I checked uri, using cfclient, and found that it is correct.

whoenig, 2019-03-15:
Can you try `rosrun crazyflie_tools scan`? If that finds the same URI, can you try `rosrun crazyflie_tools reboot --uri radio://0/80/250K` and verify that you can reboot the CF?

shmpwk, 2019-03-15:
Yes, I successfully reboot the CF.

shmpwk, 2019-03-15:
But still the same error.

whoenig, 2019-03-15:
Which firmware version are you using? Which version of crazyflie_ros? Does `rosrun crazyflie_tools listParams --uri radio://0/80/250K` work for you?

Note that in indoor scenarios 2M datarate (instead of 250K) should provide better results (although this shouldn't be the issue here unless you have a very high packet drop rate).

shmpwk, 2019-03-15:
> Does rosrun crazyflie_tools listParams --uri radio://0/80/250K work for you?

Yes, it works.

shmpwk, 2019-03-16:
I tried crazyflie 2.1 also, which is 2M datarate, but same error occurred.

whoenig, 2019-03-18:
Actually I misread your initial post. Your uri in the initial launch file is `radio://0/80/250K/E7E7E7E701`. Given that your uri seems to be `radio://0/80/250K` instead, use `roslaunch crazyflie_demo hover_vrpn.launch uri:=radio://0/80/250K frame:=crazyflie1 x:=0 y:=0 z:=0.5` instead of `roslaunch crazyflie_demo hover_vrpn.launch uri:=radio://0/80/250K/E7E7E7E701 frame:=crazyflie1 x:=0 y:=0 z:=0.5`.

shmpwk, 2019-03-18:
Thank you, runtime_error disappeared.
But another error occurred.

`drone@drone-GL502VSK:~$ roslaunch crazyflie_demo hover_vrpn.launch uri:=radio://0/80/2M frame:=crazyflie1 x:=0 y:=0 z:=0.5 
... logging to /home/drone/.ros/log/64ccc114-496a-11e9-b01b-28c63f262be5/roslaunch-drone-GL502VSK-4454.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
xacro.py is deprecated; please use xacro instead
started roslaunch server http://localhost:39091/

SUMMARY
========

PARAMETERS
 * /crazyflie/controller/PIDs/X/integratorMax: 0.1
 * /crazyflie/controller/PIDs/X/integratorMin: -0.1
 * /crazyflie/controller/PIDs/X/kd: 20.0
 * /crazyflie/controller/PIDs/X/ki: 2.0
 * /crazyflie/controller/PIDs/X/kp: 40.0
 * /crazyflie/controller/PIDs/X/maxOutput: 10.0
 * /crazyflie/controller/PIDs/X/minOutput: -10.0
 * /crazyflie/controller/PIDs/Y/integratorMax: 0.1
 * /crazyflie/controller/PIDs/Y/integratorMin: -0.1
 * /crazyflie/controller/PIDs/Y/kd: -20.0
 * /crazyflie/controller/PIDs/Y/ki: -2.0
 * /crazyflie/controller/PIDs/Y/kp: -40.0
 * /crazyflie/controller/PIDs/Y/maxOutput: 10.0
 * /crazyflie/controller/PIDs/Y/minOutput: -10.0
 * /crazyflie/controller/PIDs/Yaw/integratorMax: 0.0
 * /crazyflie/controller/PIDs/Yaw/integratorMin: 0.0
 * /crazyflie/controller/PIDs/Yaw/kd: -20.0
 * /crazyflie/controller/PIDs/Yaw/ki: 0.0
 * /crazyflie/controller/PIDs/Yaw/kp: -200.0
 * /crazyflie/controller/PIDs/Yaw/maxOutput: 200.0
 * /crazyflie/controller/PIDs/Yaw/minOutput: -200.0
 * /crazyflie/controller/PIDs/Z/integratorMax: 1000.0
 * /crazyflie/controller/PIDs/Z/integratorMin: -1000.0
 * /crazyflie/controller/PIDs/Z/kd: 6000.0
 * /crazyflie/controller/PIDs/Z/ki: 3500.0
 * /crazyflie/controller/PIDs/Z/kp: 5000.0
 * /crazyflie/controller/PIDs/Z/maxOutput: 60000.0
 * /crazyflie/controller/PIDs/Z/minOutput: 10000.0
 * /crazyflie/controller/crazyflies: [{'id': 1, 'chann...
 * /crazyflie/controller/frame: crazyflie1
 * /crazyflie/controller/worldFrame: world
 * /crazyflie/crazyflie_add/enable_logging: False
 * /crazyflie/crazyflie_add/enable_logging_battery: True
 * /crazyflie/crazyflie_add/enable_logging_imu: True
 * /crazyflie/crazyflie_add/enable_logging_magnetic_field: True
 * /crazyflie/crazyflie_add/enable_logging_packets: True
 * /crazyflie/crazyflie_add/enable_logging_pressure: True
 * /crazyflie/crazyflie_add/enable_logging_temperature: True
 * /crazyflie/crazyflie_add/pitch_trim: 0
 * /crazyflie/crazyflie_add/roll_trim: 0
 * /crazyflie/crazyflie_add/tf_prefix: crazyflie
 * /crazyflie/crazyflie_add/uri: radio://0/80/2M
 * /crazyflie/crazyflie_add/use_ros_time: True
 * /crazyflie/joy/dev: /dev/input/js0
 * /crazyflie/joystick_controller/use_crazyflie_controller: True
 * /crazyflie/pose/name: goal
 * /crazyflie/pose/rate: 30
 * /crazyflie/pose/x: 0
 * /crazyflie/pose/y: 0
 * /crazyflie/pose/z: 0.5
 * /crazyflie/vrpn_client_node/child_frame_id: crazyflie1
[... 95 lines cut ...]
[ INFO] [1552905879.372340562]: Ready...
[ INFO] [1552905879.372390251]: Elapsed: 0.293019 s
[INFO] [1552905879.586298]: found update_params service
[INFO] [1552905879.586780]: waiting for emergency service
[INFO] [1552905879.588099]: found emergency service
[INFO] [1552905879.588399]: waiting for land service
[INFO] [1552905879.589543]: found land service
[INFO] [1552905879.589876]: waiting for takeoff service
[INFO] [1552905879.590927]: found takeoff service
[ INFO] [1552905880.052012129]: Connection established
[ERROR] [1552905883.302422963]: client wants service /crazyflie/takeoff to have md5sum d41d8cd98f00b204e9800998ecf8427e, but it has b665b6c83a196e4774268cc26329b159. Dropping connection.
[ERROR] [1552905883.311716]: bad callback: <bound method Controller._joyChanged of <__main__.Controller instance at 0x7fcfbc0c2fc8>>
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/drone/catkin_ws/src/crazyflie_ros/crazyflie_demo/scripts/controller.py", line 46, in _joyChanged
    self._takeoff()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 505, in call
    raise ServiceException("unable to connect to service: %s"%e)
ServiceException: unable to connect to service: remote error reported: client wants service /crazyflie/takeoff to have md5sum d41d8cd98f00b204e9800998ecf8427e, but it has b665b6c83a196e4774268cc26329b159. Dropping connection.

`
Is it a problem of computer itself?

whoenig, 2019-03-18:
This might be a mixed ROS installation or a bug. Also see issue #105, which seems to be a similar error.

Oliver-12138, 2019-03-19:
@whoenig  Hi whoenig,I'm now facing a problem,  I launch hover_vrpn.launch, all the nodes are started, and there is no error, when I press the button on the xbox360, the crazyflie2.0 takes off ,but doesn't hover. What should I do? Thank you inadvance!

whoenig, 2019-03-19:
How did you get rid of the error?

What does the CF do if not hovering?

Oliver-12138, 2019-03-20:
Thank for your reply. The quadcopter just flies randomly, and can't get to the desired point showed in the rviz.

shmpwk, 2019-03-23:
@whoenig 
By changing the firmware, (for me changing to 2018.01 version,) md5sum mismatch error often disappear. (Sometimes the error occurred, but I don't mind)

shmpwk, 2019-03-23:
@Oliver-12138 
1. If you use Optitrack, is it OK the direction of axis? (maybe  Z-up).
2, Can Motion capture system always recognize your crazyflie?
3, If you attatch camera on crazyflie, doesn't it obstract the view?

podnerogato, 2019-03-29:
I just want to add we are facing the same issue. We use ROS Kinetic on Ubuntu 16.04., LPS, 5 Crazyflies 2.0 with latest firmware, and lps_ros_ekf_hover lunch files. On first machine everything works well, but on recently installed new machine (Ubuntu 16.04 and ROS Kinetic) we get following error when we press take off/landing button on joystick. It seems to be the issue with these two actions since emergency button seems to be functioning and references in RVIZ are changing as expected when joystick sticks are moved. 
`[ERROR] [1553857490.480143]: bad callback: <bound method Controller._joyChanged of <__main__.Controller instance at 0x7f514611d710>>
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/lariat/crazy_ws/src/crazyflie_ros/crazyflie_demo/scripts/controller.py", line 42, in _joyChanged
    self._land()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 505, in call
    raise ServiceException("unable to connect to service: %s"%e)
ServiceException: unable to connect to service: remote error reported: client wants service /crazyflie0/land to have md5sum d41d8cd98f00b204e9800998ecf8427e, but it has b665b6c83a196e4774268cc26329b159. Dropping connection.
`

shmpwk, 2019-03-29:
Hi, @mikehamer;
When I used the latest firmware, I had md5sum error. But when I changed to the older firmware (for me 2018.01 version), strangely enough, the error disappeared.