whoenig/crazyflie_ros #125: Tuning PID parameters

DennisMelamed, 2019-03-14:
Hi Wolfgang - I'm running into a bit of a problem using the crazyflie_controller code in simulation. I know it is old, non-recommended code, but I'm intending to train a network to control a crazyflie which cannot run onboard, so I'm forced to train it to output the `cmd_vel` components (i.e would take the place of `crazyflie_controller`, but I need `crazyflie_controller` as a ground truth). I've been tuning the PID parameters, and can get the crazyflie to take off and get a stable hover pretty consistently in simulation. However, after changing the position specified in `/goal` a few times (or by too large of an amount), it starts to oscillate wildly and cannot recover. Do you have any suggestions for tuning the parameters? Maybe the procedure you used to tune them? A better way to approach this problem?

Thanks for your help! -Dennis

whoenig, 2019-03-15:
Hi Dennis,

I think for large step inputs you would need to limit the maximum contributing portion of the P-part. Similar to the bound here: https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_controller/src/pid.hpp#L53, you would need to add a bound on p and potentially d.

Some thoughts on alternative approaches:
1. Some networks can actually run on-board, see https://sites.google.com/view/sim-to-multi-quad.
2. You could add additional hardware and connect over USB (or as a deck)
3. It sounds like you try to do supervised learning? Since the crazyflie_controller itself is "bad", your network won't do any better. Perhaps RL would be a better choice (in which case you won't need the crazyflie_controller code, unless you want to compare to it).

DennisMelamed, 2019-04-03:
Hi Wolfgang- Thank you so much for your help. Limiting the P and D parameters worked perfectly. Thank you again for your work on this project and support.

closed 2019-04-03 by DennisMelamed