whoenig/crazyflie_ros #127: Subscriber in publish_external_position.py is not working in native Ubuntu

mandelyoni, 2019-03-23:
Dear Wolfgang,

We're masters students from TAU university.
We have been working with your great crazyflie_ros library for quite some time now on a VM (from Bitcraze) and an Optitrack system, and everything worked fine.
Lately, we moved to use Ubuntu native 16.04 with ROS Kinetic and installed all libraries according to this GitHub and to the "Flying Multiple UAVs Using ROS" paper. 
For some reason, the subscriber in publish_external_position_vicon.py is not working (the messages are sent but onNewTransform is not activated). Any idea?

Thank you,  Yoni

mandelyoni, 2019-03-23:
I'm sorry, it's working now. 
Thanks,  Yoni

whoenig, 2019-03-24:
Perhaps you can describe how/what you changed to get it working and close this issue? Thanks for reaching out!

mandelyoni, 2019-03-24:
I don't remember but I'll install it on another computer later this week and describe the process here.

shmpwk, 2019-03-24:
Hi, 
I have a similar issue.
When I execute
`roslaunch crazyflie_demo external_position_vrpn.launch`
Error is
`ERROR: cannot launch node of type [crazyflie_demo/publish_external_position_vrpn.py]: can't locate node [publish_external_position_vrpn.py] in package [crazyflie_demo]
`
I read #85 #89 as a reference.
I followed the explanation below
> you will need to launch external_position_vicon [Which needs to be slightly changed to support Optitrack]. Then, in another terminal, execute python execute_trajectoy.py figure8withTakeoffAndLanding.csv.

> a) Adjust the launch file to the vrpn_client
b) Adjust topics names in publish_external_position (could simply be set to /tf)
c) change frame names (according to motive names) in the high level script.

What should I do to fix it?

mandelyoni, 2019-03-28:
#shmpwk - try   chmod +x    on the file

mandelyoni, 2019-03-28:
Dear Wolfgang, it was just a topic name problem. We switched to the correct name and it was ok.
THANKS FOR ALL YOUR HELP !!!

shmpwk, 2019-03-28:
@mandelyoni 
Thank you for your reply. 
I tried 
`chmod +x on the file`
but still have an error.
In the first place, I'm not sure whether the process I've done is correct.

whoenig, 2019-03-28:
You don't need to build a custom firmware anymore. The scrips switch to the kalman filter automatically; the kalman filter is part of the official firmware. Otherwise your steps seemed ok to me.