whoenig/crazyflie_ros #129: Nothin happened when take off requested.

GalBrandwine, 2019-03-31:
Hey!

Until last month we worked with your repo, and everything worked grate.

Today I've forked your repo, and when launched my launch file (simple crazyflie server and 1 CF with logging set to true), nothing happened when I request takeoff.

I've did some debugging, and there's a good connection.
All topics flying fluently between CF and the laptop. Sensor readings are great. Just there's no movement.

I've tried takeoff via python script, and takeoff via service call. Yet nothing.

Can you please help?
Thanks, Gal.

wydmynd, 2019-03-31:
does takeoff using /cmd_hover work?

hanzheteng, 2019-03-31:
Please check if you have enable highlevel command parameter

whoenig, 2019-04-01:
I found a critical error in crazyflie_cpp that was recently introduced that caused some packets to be total garbage. This (and crazyflie_ros) is fixed now and I did some preliminary flight tests with the flow deck. Please let me know if that fixes your problem.

GalBrandwine, 2019-04-04:
Hey, again.
Iv'e updates your submodules, at my repo.

everything working fine so far,
thanks!

Gal.

whoenig, 2019-04-04:
Great - thanks for the update! Feel free to open a new issue if you run into more trouble.

closed 2019-04-04 by whoenig