whoenig/crazyflie_ros #132: Hover with VRPN issues position data is not correct

ChrisYang0830, 2019-04-11:
Hi, whoenig
I'm using the crazyflie to do the path following through the VRPN. I launch the hover_vrpn.launch to retrieve the position data from the Optitrack Track. Everything seems working well, the crazyflie is able to fly and hover above a certain point, I also can get position information from the Optitrack Track through the VRPN. The position data is CORRECT when the yaw angle is fixed.
But the issue is the retrieved position information is NOT correct when I rotate the crazyflie (change the yaw angle). The x, y, and z position are varying with the changing of the yaw angle in a fixed point. I move the crayflie by hand to test whether the retrieved position data is correct or not.
I'm thinking the issue is about the tf frame, but I can't find a solution.
I appreciate if you can give me some suggestions.
I'm sorry for my bad English.
Best
Chris

whoenig, 2019-04-11:
There is some setting in the Motive software that you need to change in order to let OptiTrack operate in the default ROS coordinate frame. I don't have an optitrack system to be more specific, but I hope you can find such a setting. (Advanced network settings. Up axis: Z).

ChrisYang0830, 2019-04-12:
Hi, whoenig
I appreciate for your help, actually, I change the "Up Axis" to Y-up and
also Z-up, but both give me the same error I described above. I used the
Mocap package (Optitrack Track Ros Package) in another robot (not
crazyflie) to retrieve the position information which gives me the correct
one.
So I think the problem should be about the coordinate frame transformation.
Could you give me some suggestions? Do we need to change the coordinate
frame in ROS, or the only thing we can do is the Optitrack Track setting?
Thanks again for your help
Best
Chris

whoenig, 2019-04-12:
You can simply use a different ros package for tracking. The example uses ros_vrpn_client (see https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_demo/launch/hover_vrpn.launch#L46-L52). You can also remove those lines and replace them with your favorite motion capture tracking package. Crazyflie_ros simply uses the tf information.

ChrisYang0830, 2019-04-12:
Appreciate, I will try it, and I'll post my updates if got some.
Best
Chris

Manakoy, 2019-04-26:
Hi, ChrisYang0830 
Maybe we have encountered the same problem, and I have been worried about it for a long time. I think we can exchange it to solve it together.

ChrisYang0830, 2019-04-26:
> Hi, ChrisYang0830
> Maybe we have encountered the same problem, and I have been worried about it for a long time. I think we can solve it together.

Hello Manakoy
I think I find a way to solve this problem. First, you need to make it clear of the relationship between the Optitrack coordinate frame and the ROS default coordinate frame. If they are not coincident, you need to adjust the Optitrack coordinates frame by setting the UP axis to Z-up or Y-up.
Then you can use following functions below to output the coordinates in ROS.
transform.getOrigin().getX();
transform.getOrigin().getY(); 
transform.getOrigin().getZ();

ChrisYang0830, 2019-04-26:
Hello Manakoy
I think I find a way to solve this problem. First, you need to make it
clear of the relationship between the Optitrack coordinate frame and the
ROS default coordinate frame. If they are not coincident, you need to
adjust the Optitrack coordinates frame by setting the UP axis to Z-up or
Y-up.
Then you can use following functions below to output the coordinates in ROS.
transform.getOrigin().getX();
transform.getOrigin().getY();
transform.getOrigin().getZ();

Manakoy, 2019-04-27:
I've changed the up Axis to the z axis. Because of the up Axis of melodic is z axis. But this doesn't  have changed anything. Therefore, i modified  vrpn_client_ros.cpp.
The following sections have been modified:

        tracker->pose_msg_.header.stamp = ros::Time::now();      }      tracker->pose_msg_.pose.position.x = tracker_pose.pos[0];      tracker->pose_msg_.pose.position.y = tracker_pose.pos[1];      tracker->pose_msg_.pose.position.z = tracker_pose.pos[2];      tracker->pose_msg_.pose.orientation.x = tracker_pose.quat[0];      tracker->pose_msg_.pose.orientation.y = tracker_pose.quat[1];      tracker->pose_msg_.pose.orientation.z = tracker_pose.quat[2];      tracker->pose_msg_.pose.orientation.w = tracker_pose.quat[3];      pose_pub->publish(tracker->pose_msg_);    }

Change the previous 0,1,2 to 0,2,1.

And then I run:
$roslaunch vrpn_client_ros sample.launch
$roslaunch crazyflie_demo hover_vrpn.launch
I moved crazyflie by hand to verify that its information in rviz is correct. And then, I pressed the button on the handle to make it hover at 0.5m.But it failed, and crazyflie just flew around the ground.

Manakoy, 2019-04-27:
Hello, ChrisYang0830
Sorry, I don't seem to understand you very well. I don't quite understand how to call and where to call this function. I want to know which file you are correcting to correct the coordinates.

ChrisYang0830, 2019-04-27:
> Hello, ChrisYang0830 Sorry, I don't seem to understand you very well. I don't quite understand how to call and where to call this function. I want to know which file you are correcting to correct the coordinates.
> […](#)

Actually, you don't need to change anything in the vrpn_client_ros and also the hover_vrpn.launch if you just want to achieve the hovering function. The file to control the hover function is located in the crazyflie_controller/src/controller.cpp. The function 
targetDrone.pose.position.x
targetDrone.pose.position.y
targetDrone.pose.position.z
are used to import coordinates from the VRPN, however, these data is varying with yaw rotating in my Optitrack. So I replaced this function with
transform.getOrigin().getX();
transform.getOrigin().getY();
transform.getOrigin().getZ();
Such that, I can get the correct coordinates from the VRPN.
Sorry that I'm new to the Gitbuh, I don't know how to insert the link of code into the text.

Manakoy, 2019-04-28:
I changed part of controller.cpp as you said.. But it doesn't seem to work. Video with experiment in the attachment. 

                msg.linear.x = m_pidX.update(0, transform.getOrigin().getY());
                msg.linear.y = m_pidY.update(0.0, transform.getOrigin().getX());
                msg.linear.z = m_pidZ.update(0.0, transform.getOrigin().getZ());    

I uploaded my crazyflie_ros. Can you help me see where is the problem?  Many thanks！

ChrisYang0830, 2019-04-28:
> I changed part of controller.cpp as you said.. But it doesn't seem to work. Video with experiment in the attachment. msg.linear.x = m_pidX.update(0, transform.getOrigin().getY()); msg.linear.y = [...]

Hello, Manakoy
There are many reasons that influence the attitude of the crazyflie. You can try to change the parameters of the PID update function m_pidX.update. If you open the function update(float value, float targetValue) in the pid.hpp, you will find the first parameter is the current value of the crazyflie, and the second value is the targetValue. If you want to crazyflie to hover at the original point, the targetValue should be 0. However, in the controller.cpp, msg.linear.x = m_pidX.update(0, transform.getOrigin().getY()); the current value is put in the second. So probably you can try to change the position of these two values and see if the crazyflie is able to hover. If not, probably you need to adjust the PID value.