whoenig/crazyflie_ros #133: Hey

GalBrandwine, 2019-04-14:
Here what in initializing in my .launch file:
` <rosparam>
        genericLogTopics: ["log_pos" , "log_rpy" , "log_ranges" ]
                genericLogTopicFrequencies: [20 , 20 , 20 ]
                genericLogTopic_log_pos_Variables: ["stateEstimate.x", "stateEstimate.y", "stateEstimate.z"]
                genericLogTopic_log_rpy_Variables: ["stabilizer.roll" , "stabilizer.pitch" , "stabilizer.yaw"]
                genericLogTopic_log_ranges_Variables: ["range.front", "range.back", "range.up", "range.left",
                "range.right" , "range.zrange" ]
            </rosparam>]`

whoenig, 2019-04-15:
There is no real documentation right now, apart from the firmware source code. Perhaps you are interested in those variables: https://github.com/bitcraze/crazyflie-firmware/blob/master/src/deck/drivers/src/flowdeck_v1v2.c#L226-L227?

wydmynd, 2019-04-15:
what you want is - 
genericLogTopic_log_Variables: ["motion.deltaX" , "motion.deltaY" ]

GalBrandwine, 2019-04-16:
Thanks!
(haven't tryed yet, but still)

Is there more attributes to "motion" that i can log?

Gal