whoenig/crazyflie_ros #136: run hover_vrpn.launch have encountered some problems

Manakoy, 2019-04-28:
Hi，whoening.
I am using vrpn to try to hover crazyflie 2.0.  But crazyflie can't hover normally and flying disorderly. I think it is because there is a problem with the position data. So I have set ‘Up axis’ to ‘Z up’ in motive. And I also modified the controller.cpp

                msg.linear.x = m_pidX.update(0, targetDrone.pose.position.x);
                msg.linear.y = m_pidY.update(0.0, targetDrone.pose.position.y);
                msg.linear.z = m_pidZ.update(0.0, targetDrone.pose.position.z);

Changed to

                msg.linear.x = m_pidX.update(0, transform.getOrigin().getX());
                msg.linear.y = m_pidY.update(0.0, transform.getOrigin().getY());
                msg.linear.z = m_pidZ.update(0.0, transform.getOrigin().getZ());

But it also failed. 
[image]

whoenig, 2019-04-30:
There should be no need to change the controller. It does look like your position estimate is not correct. Did you verify (by walking around with the drone), that the coordinate system matches the ROS standard coordinate system?

Manakoy, 2019-04-30:
I tried it. I found that the position and attitude information of the aircraft did not seem to be correct. But I don't know where to correct.

Manakoy, 2019-04-30:
Thank you for your answer, but I still have some places that I don’t understand.

1.What is the standard coordinate system of ros? And in motive is xoz as the ground, Is it wrong to regard the z-axis as the y-axis?
2.How do I correct the coordinates? What file do I need to modify?

hanzheteng, 2019-04-30:
If you are working directly on top of `crazyflie_ros` rather than `crazyswarm`, you can try this package (https://github.com/ros-drivers/mocap_optitrack). The corresponding wiki page is here: http://wiki.ros.org/mocap_optitrack. It is designed to connect OptiTrack with ROS and works with the latest version of OptiTrack Motive.  If using this package, remember to set `Y up` by default, because this ROS package has already considered this coordinate.