whoenig/crazyflie_ros #137: "timeout " a new error

wangyubin11, 2019-05-05:
Hi,
  There is a new error,
 [ERROR] [1557057450.162429076]: Exception thrown while processing service call: timeout
[ERROR] [1557057450.162832149]: Service call failed: service [/add_crazyflie] responded with an error: timeout
 The feedback is differernt from the former issues,so thanks for your help.
 Looking forward  to your reply.
 Thanks.

whoenig, 2019-05-05:
What former issues are you referring to? The timeout error usually occurs if there is an issue with the URI, i.e. the Crazyradio can't connect to the Crazyflie.

wangyubin11, 2019-05-05:
roger that ，thanks!

wydmynd, 2019-05-13:
I am also having timeout issues while flying. the drone drops suddenly and crazyflie_server says error - timeout.this happens apparently due to RF interference since I get lots of "link quality low" messages.
could not find in crazyflie_server.cpp -- is there a parameter that can increase the number of seconds before a timeout is declared?

whoenig, 2019-05-13:
You can find that here: https://github.com/whoenig/crazyflie_cpp/blob/master/include/crazyflie_cpp/Crazyflie.h#L346 and https://github.com/whoenig/crazyflie_cpp/blob/master/include/crazyflie_cpp/Crazyflie.h#L381-L382. What messages do you send during the flight? Ideally, you only broadcast data which should not be able to trigger a timeout error.

wydmynd, 2019-05-13:
Thanks!
I'm supposedly sending just  /goto service calls , and logging some 10 parameters at the same time. 
flying high level mode using mellinger controller. 
the /goto calls are being sent at 1-2 hz though, because signal quality is so bad.  I don;t know how message buffering is implemented.

wydmynd, 2019-05-15:
it seems that /go_to can cause timeouts

[INFO] [1557911487.054907]: service [/cf8/go_to] responded with an error: timeout

whoenig, 2019-05-20:
Yes, if you send commands to a single CF, this can cause timeouts as it will wait to receive an acknowledgement.