whoenig/crazyflie_ros #139: Mellinger controller + advices on weight and balance

morsingher, 2019-05-16:
Good morning,

I'm working on my master thesis about aggressive maneuvers with the Crazyflie and I have a couple of questions:

1) I have read in some old issues that you were about to implement the Mellinger controller at high-level. Did this already happen?

2) I have to use a Qualisys motion capture system, with 12 mm markers, as the localization system. The main problem is that the total weight of the platform becomes around 40 g and the balance is quite hard to obtain if markers must be placed asymetrically. I have noticed that during take off and hovering the CF really struggles to provide enough thrust. I have modified the mass in the description of the CF, but there may be some parameters that I forgot to modify. Do you think it would work even with some added weight? Do you have any advice regarding to this aspect?

Thank you in advance for the help,

Marco.

whoenig, 2019-05-16:
1. Yes, the mellinger controller is implemented in the firmware (merged into the official firmware), see https://github.com/bitcraze/crazyflie-firmware/blob/master/src/modules/src/controller_mellinger.c.

2. You can try to use the active marker deck, see https://www.bitcraze.io/2019/04/the-active-marker-deck-for-mocap-systems/. You could also use the crazyswarm (https://crazyswarm.readthedocs.io/en/latest/), where I recently added support for single-marker tracking, i.e. you only put a single marker on your CF and the position is tracked frame-by-frame. The on-board EKF can recover the orientation as long as you move in the x/y plane.

In my experience the flight performance gets much worse with added weight and added structures. You should definitely try keeping it below 35g, or use more powerful motors.

morsingher, 2019-05-17:
Thank you for the quick reply. 

I knew about the Mellinger controller in the firmware but I wasn't sure that it could control both orientation and position. I mean, as far as I know the high-level controller in ROS takes care of the position, while the low-level controller in the Crazyflie controls the orientation. Therefore I had the feeling that a ROS node implementing the position control would be needed in any case. However, if the Crazyflie can control both it's even better, thank you for the information. 

As far as the weight and balance are concerned, I will probably buy the deck and some smaller markers, in order to stay around 28-29 g.

whoenig, 2019-05-17:
Mellinger is a trajectory tracking controller. You can set setpoints by either uploading a trajectory, or send fullStateCmd setpoints over the radio. I do not recommend running controllers on the client side - even if you implement your own controller, you should aim to have it run on-board for best results.

Good luck and let me know if you have more issues/questions!

closed 2019-05-17 by whoenig