whoenig/crazyflie_ros #141: Delay with the optitrack estimated position

mandelyoni, 2019-06-24:
Dear Dr. Whoenig,

We're working with an Optitrack and crazy flies (with your library crazyflie_ros).
While observing the optitrack's pose estimation, through ''rostopic echo /crazyflie8/vrpn_client_node/crazyflie8/pose'', we definitely identify a delay. This delay makes the position loop very unstable.
Any advice on this issue?

Thank you,  Yoni.

whoenig, 2019-06-25:
How is your optitrack computer connected to the machine that is running crazyflie_ros? I highly recommend using a wired connection and not WiFi.

Additionally, I recommend using NatNet instead of VRPN for (slightly) lower latency. For example, you can use https://github.com/whoenig/motioncapture_ros instead of vrpn_client_ros.

mandelyoni, 2019-06-25:
Dear Dr. Whoenig,

Thank you for your prompt answer. I will try it asap.

Another issue that we're having is with the latest version of the crazyflie. 
We've just purchased a crazyflie 2.1 and tried to operate it with crazyflie_ros. For some reason, the motors are moving slowly and the drone doesn't take off. Do we need to do a firmware update? To which version?

Thank you,   Yoni.

whoenig, 2019-06-25:
Any newer official firmware should be fine. Which "operation mode" are you using? I don't recommend using the external position controller in crazyflie_controller (neither for CF2.0 nor CF2.1) - this might need re-tuning for CF2.1. Instead, use one of the on-board position controllers (PID or mellinger).

mandelyoni, 2019-06-26:
Ok,  thank you very much.