whoenig/crazyflie_ros #143: LPS setup w/ rviz [crazyflie_server-12] process has died

1CoolDavid, 2019-07-03:
Hi, I've been trying to set up the crazyswarm system using lps motion capture. When I run rviz with just one drone (I'm only testing with one currently), I see this error:

```
[crazyflie_server-12] process has died [pid 30066, exit code -6, cmd /home/brandon/crazyswarm/ros_ws/devel/lib/crazyflie_driver/crazyflie_server __name:=crazyflie_server __log:=/home/brandon/.ros/log/3109deb4-9da9-11e9-8182-08119607ca64/crazyflie_server-12.log]. 
log file: /home/brandon/.ros/log/3109deb4-9da9-11e9-8182-08119607ca64/crazyflie_server-12*.log
```

I've also seen Errno 401 and various "connection refused" errors combined with this output. It changes its message even though no configuration on the drone or filesystem has been made. Even when the drone is physically plugged into my machine I recieve this log error. Once in a while it will work fine as if there is no reason for the error to exist at all. 

Any help would be appreciated. I'm using Ubuntu 16.04 with ros kinetic.

whoenig, 2019-07-03:
Your folder looks strange - are you using crazyflie_ros or the crazyswarm? How did you setup your catkin workspace?

1CoolDavid, 2019-07-03:
I've been using the crazyflie_ros workspace within crazyswarm. I have ros_ws as the main workspace for everything. Is that the incorrect format?

whoenig, 2019-07-03:
There will be issues with that setup. If you want to use crazyflie_ros, you should either use a different ROS workspace or use the crazyflie_ros version that is in crazyswarm/ros_ws/src/crazyflie_ros.

1CoolDavid, 2019-07-03:
Where should the new ROS workspace be located? I'm assuming lps-ros must be cloned into that workspace as well, correct?

whoenig, 2019-07-03:
There is no need for lps-ros anymore - the state estimation is now implemented on board the firmware.
You can put the ROS workspace anywhere on your disk (but not inside another ros workspace).

1CoolDavid, 2019-07-03:
So I don't need lps-ros, even though I'm using beacons for a crazyswarm? If so, what am I supposed to do with the crazyswarm project? crazyflie-ros doesn't have launch files to use.

whoenig, 2019-07-04:
For crazyswarm, use motion_capture_type None (see https://crazyswarm.readthedocs.io/en/latest/usage.html#none). For crazyflie_ros you would need to create your own example launch file at the moment. The launch files in lps-ros are, to my knowledge, outdated.

closed 2019-07-09 by 1CoolDavid