whoenig/crazyflie_ros #146: Crazyflie ros cmd position and external pose

harrafa, 2019-07-23:
Dear Dr. Hoenig,
I am using crazyflie__ros for waypoint following with Aruco markers. Going by one of your posts where you suggest using onboard controllers for position control instead of crazyflie_controller, I have written ros nodes that publish to topics cmd_position(target position) and external_pose(current position and orientation) subscribed by crazyflie_server. 
When I try to run the launch file the rotors spin but the crazyflie does not take off. Is this enough information for the onboard controller for waypoint following?

Thank you.

whoenig, 2019-07-24:
Yes, this information should suffice to achieve flight. Which firmware are you using? I assume cmd_position and external_position are different, i.e., the on-board controller has a position error to compensate for?

harrafa, 2019-07-24:
Thank you for your response Hoenig.
I have the latest firmware on the crazyflie, 2019.02 and am trying to correct for position error given current position(published to external_position) and setpoint(published to cmd_position)

I have attached an image of the launch file here. In a nutshell, I am publishing to the two topics from the two nodes talker and pose.
[image]

Do I need to setup the Kalman filter parameters like in position.py(attached here):

[image]

I will be glad to provide any other information you require.

Thank you.

harrafa, 2019-07-27:
Dear Dr. Hoenig, 
I edited the config.mk file in firmware to include ESTIMATOR=kalman. The crazyflie takesoff at high speed to crash. I have setup kalman resetEstimation to 0 going by the Position.py script above. 
Do I need to setup the update params service? Could you please help me figure what I am missing?
Thank you.

whoenig, 2019-07-27:
* You don't need to recompile the firmware. You can force the Kalman estimator by setting "stabilizer.estimator" (a CF parameter) to 2
* You should initialize and reset the kalman filter, similar to https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_demo/scripts/publish_external_position_vicon.py. Note this assumes that yaw is 0. I believe newer firmware versions also have a parameter to initialize yaw.

harrafa, 2019-07-28:
Thank you Dr.Hoenig.
Can you please direct me to where I can set stabilizer.estimator parameter. Thank you.