whoenig/crazyflie_ros #147: libusb error when trying to connect more than one CF over USB

wydmynd, 2019-08-20:
Hello, 

I know this is a weird use case, we are using the CF as a platform to log sensor data. 
using crazyflie_tools scan I can see 2 devices 
usb://0
usb://1

When connecting one CF over USB (usb://0) everything works as intended. 
I try to launch the second one  (usb://1) I get the following 

```
Adding usb://1 as crazyflie2 with trim(0.000000, 0.000000). Logging: 1, Parameters: 1, Use ROS time: 1
[ERROR] [1566304633.063972722]: Exception thrown while processing service call: LIBUSB_ERROR_BUSY
```

any suggestions greatly appreciated. thanks!

whoenig, 2019-08-22:
It is certainly supported in crazyflie_cpp, but I don't think I tried it with crazyflie_ros, yet. I won't be able to test this myself in the next few days, but if you want to debug it you can build a debug version, run the add_crazyflie node with gdb and figure out where the exception is thrown (see https://github.com/USC-ACTLab/crazyswarm/issues/68#issuecomment-398113489).

Supredan, 2019-10-10:
@wydmynd hi
I am doing exactly the same thing as you,And I only need one cf.
but when i use the launch file `customLogBlocks` in crazyfile_demo,i change the uri to usb0,here is my file:
```
<?xml version="1.0"?>

<launch>
  <arg name="uri" default="usb://0" />

  <include file="$(find crazyflie_driver)/launch/crazyflie_server.launch">
  </include>

  <group ns="crazyflie">

    <node pkg="crazyflie_driver" type="crazyflie_add" name="crazyflie_add" output="screen">
      <param name="uri" value="$(arg uri)" />
      <param name="tf_prefix" value="crazyflie" />
      <rosparam>
        genericLogTopics: ["log1", "log2"]
        genericLogTopicFrequencies: [10, 100]
        genericLogTopic_log1_Variables: ["stateEstimate.x"]
        genericLogTopic_log2_Variables: ["acc.x", "acc.y", "acc.z"]
      </rosparam>
    </node>

  </group>
</launch>
```
but when i run `$ roslaunch crazyflie_demo customLogBlocks.launch` here comes the error:
```
[ INFO] [1570717654.324237032]: wait_for_service /add_crazyflie
[ INFO] [1570717654.324715677]: found /add_crazyflie
[ INFO] [1570717654.327902891]: Adding usb://0 as crazyflie with trim(0.000000, 0.000000). Logging: 1, Parameters: 1, Use ROS time: 1
[ INFO] [1570717654.391751804]: CF Console: ---
[ INFO] [1570717654.392052000]: CF Console: SYS: Crazyflie 2.1 is up and running!
[ INFO] [1570717654.392283410]: CF Console: SYS: Production release 2019.09
[ INFO] [1570717654.392858509]: CF Console: SYS: I am 0x203937434848500E0036005C and I have 1024KB of flash!
[ INFO] [1570717654.393696935]: CF Console: CFGBLK: v1, verification [OK]
[ INFO] [1570717654.393793855]: CF Console: DECK_CORE: 1 deck(s) found
[ INFO] [1570717654.393967462]: CF Console: DECK_CORE: Calling INIT on driver bcDWM1000 for deck 0
[ INFO] [1570717654.394139554]: CF Console: IMU: BMI088 Gyro I2C connection [OK].
[ INFO] [1570717654.394312457]: CF Console: IMU: BMI088 Accel I2C connection [OK]
[ INFO] [1570717654.394777818]: CF Console: IMU: BMP388 I2C connection [OK]
[ INFO] [1570717654.394969986]: CF Console: ESTIMATOR: Using Kalman (2) estimator
[ INFO] [1570717654.395777239]: CF Console: CONTROLLER: Using PID (1) controller
[ INFO] [1570717654.395949380]: CF Console: MTR-DRV: Using brushed motor driver
[ INFO] [1570717654.396697302]: CF Console: EEPROM: I2C connection [OK].
[ INFO] [1570717654.396789454]: CF Console: DECK_CORE: Deck 0 test [OK].
[ INFO] [1570717654.396961023]: CF Console: STAB: Wait for sensor calibration...
[ INFO] [1570717654.397053796]: CF Console: SYS: Free heap: 8664 bytes
[ INFO] [1570717654.397153430]: Requesting parameters...
[ WARN] [1570717654.404035469]: Don't know ack: Port: 13 Channel: 1 Len: 6
[ WARN] [1570717654.404696542]: Don't know ack: Port: 13 Channel: 1 Len: 6
[ WARN] [1570717654.404816655]: Don't know ack: Port: 13 Channel: 1 Len: 6
[ WARN] [1570717654.404931767]: Don't know ack: Port: 13 Channel: 1 Len: 6
[ WARN] [1570717654.405056941]: Don't know ack: Port: 13 Channel: 1 Len: 6
[ WARN] [1570717654.405851283]: Don't know ack: Port: 13 Channel: 1 Len: 6
[ WARN] [1570717654.406695438]: Don't know ack: Port: 13 Channel: 1 Len: 6
[ WARN] [1570717654.406838504]: Don't know ack: Port: 13 Channel: 1 Len: 6
[ WARN] [1570717654.406971204]: Don't know ack: Port: 13 Channel: 1 Len: 6
[ WARN] [1570717654.407105957]: Don't know ack: Port: 13 Channel: 1 Len: 6
[ WARN] [1570717654.407237177]: Don't know ack: Port: 13 Channel: 1 Len: 6
[ WARN] [1570717654.407379352]: Don't know ack: Port: 13 Channel: 1 Len: 6
[ WARN] [1570717654.407694275]: Don't know ack: Port: 13 Channel: 1 Len: 6
[ WARN] [1570717654.407804477]: Don't know ack: Port: 13 Channel: 1 Len: 6
[ WARN] [1570717654.407916473]: Don't know ack: Port: 13 Channel: 1 Len: 6
[ WARN] [1570717654.408024524]: Don't know ack: Port: 13 Channel: 1 Len: 6
[ WARN] [1570717654.408136422]: Don't know ack: Port: 13 Channel: 1 Len: 6
[ WARN] [1570717654.408244960]: Don't know ack: Port: 13 Channel: 1 Len: 6
[ WARN] [1570717654.408386931]: Don't know ack: Port: 13 Channel: 1 Len: 6
[ WARN] [1570717654.408695014]: Don't know ack: Port: 13 Channel: 1 Len: 6
[ WARN] [1570717654.408816432]: Don't know ack: Port: 13 Channel: 1 Len: 6
[ WARN] [1570717654.408937373]: Don't know ack: Port: 13 Channel: 1 Len: 6
[ WARN] [1570717654.409063141]: Don't know ack: Port: 13 Channel: 1 Len: 6
[ WARN] [1570717654.409184569]: Don't know ack: Port: 13 Channel: 1 Len: 6
[ WARN] [1570717654.409308813]: Don't know ack: Port: 13 Channel: 1 Len: 6
[ INFO] [1570717654.409467006]: Params: 164
[ INFO] [1570717654.456813519]: CF Console: SYS: Assert failed src/modules/src/crtp.c:186
terminate called after throwing an instance of 'std::runtime_error'
  what():  LIBUSB_ERROR_IO
[crazyflie/crazyflie_add-3] process has finished cleanly
log file: /home/supdan/.ros/log/1971ba9c-eb6a-11e9-af6a-04d9f5821fc4/crazyflie-crazyflie_add-3*.log
[crazyflie_server-2] process has died [pid 14377, exit code -6, cmd /home/supdan/catkin_ws/devel/lib/crazyflie_driver/crazyflie_server __name:=crazyflie_server __log:=/home/supdan/.ros/log/1971ba9c-eb6a-11e9-af6a-04d9f5821fc4/crazyflie_server-2.log].
log file: /home/supdan/.ros/log/1971ba9c-eb6a-11e9-af6a-04d9f5821fc4/crazyflie_server-2*.log
```
do you have any ideas?
if not,can you describe your steps please?
thanks !!!

Supredan, 2019-10-13:
After i update the firmware to the latest version, i  have solved this problem, thx any way!!!