whoenig/crazyflie_ros #149: The directory of crazyflie_cpp and crazyflie_tools is empty

Allenhanbo, 2019-10-09:
Hi, after I git clone https://github.com/whoenig/crazyflie_ros.git to my workspace, I found that the directory of crazyflie_cpp and crazyflie_tools is empty. So I download these two directory separately and put together with other directory in crazyflie_ros. Then I catkin_make the workspace and the erroe occured like this:

[ 94%] Built target crazyflie_driver_generate_messages_lisp
[ 94%] Built target crazyflie_driver_generate_messages_nodejs
[ 95%] Building CXX object crazyflie_ros/crazyflie_driver/CMakeFiles/crazyflie_add.dir/src/crazyflie_add.cpp.o
[ 96%] Built target vicon_bridge
[ 96%] Building CXX object crazyflie_ros/crazyflie_driver/CMakeFiles/crazyflie_server.dir/src/crazyflie_server.cpp.o
[ 97%] Built target tf_distort
[ 97%] Built target crazyflie_driver_generate_messages
[ 98%] Built target calibrate
[ 98%] Linking CXX executable /home/ardrone/crazyflie_ws/devel/lib/crazyflie_controller/crazyflie_controller
[ 98%] Built target crazyflie_controller
[ 98%] Linking CXX executable /home/ardrone/crazyflie_ws/devel/lib/crazyflie_driver/crazyflie_add
[ 98%] Built target crazyflie_add
[100%] Linking CXX executable /home/ardrone/crazyflie_ws/devel/lib/crazyflie_driver/crazyflie_server
CMakeFiles/crazyflie_server.dir/src/crazyflie_server.cpp.o: In function `void Crazyflie::setParam<unsigned char>(unsigned short, unsigned char const&)':
crazyflie_server.cpp:(.text._ZN9Crazyflie8setParamIhEEvtRKT_[_ZN9Crazyflie8setParamIhEEvtRKT_]+0x41): undefined reference to `Crazyflie::setParam(unsigned short, Crazyflie::ParamValue const&)'
CMakeFiles/crazyflie_server.dir/src/crazyflie_server.cpp.o: In function `void Crazyflie::setParam<signed char>(unsigned short, signed char const&)':
crazyflie_server.cpp:(.text._ZN9Crazyflie8setParamIaEEvtRKT_[_ZN9Crazyflie8setParamIaEEvtRKT_]+0x41): undefined reference to `Crazyflie::setParam(unsigned short, Crazyflie::ParamValue const&)'
CMakeFiles/crazyflie_server.dir/src/crazyflie_server.cpp.o: In function `void Crazyflie::setParam<unsigned short>(unsigned short, unsigned short const&)':
crazyflie_server.cpp:(.text._ZN9Crazyflie8setParamItEEvtRKT_[_ZN9Crazyflie8setParamItEEvtRKT_]+0x42): undefined reference to `Crazyflie::setParam(unsigned short, Crazyflie::ParamValue const&)'
CMakeFiles/crazyflie_server.dir/src/crazyflie_server.cpp.o: In function `void Crazyflie::setParam<short>(unsigned short, short const&)':
crazyflie_server.cpp:(.text._ZN9Crazyflie8setParamIsEEvtRKT_[_ZN9Crazyflie8setParamIsEEvtRKT_]+0x42): undefined reference to `Crazyflie::setParam(unsigned short, Crazyflie::ParamValue const&)'
CMakeFiles/crazyflie_server.dir/src/crazyflie_server.cpp.o: In function `void Crazyflie::setParam<unsigned int>(unsigned short, unsigned int const&)':
crazyflie_server.cpp:(.text._ZN9Crazyflie8setParamIjEEvtRKT_[_ZN9Crazyflie8setParamIjEEvtRKT_]+0x40): undefined reference to `Crazyflie::setParam(unsigned short, Crazyflie::ParamValue const&)'
CMakeFiles/crazyflie_server.dir/src/crazyflie_server.cpp.o:crazyflie_server.cpp:(.text._ZN9Crazyflie8setParamIiEEvtRKT_[_ZN9Crazyflie8setParamIiEEvtRKT_]+0x40): more undefined references to `Crazyflie::setParam(unsigned short, Crazyflie::ParamValue const&)' follow
collect2: error: ld returned 1 exit status
crazyflie_ros/crazyflie_driver/CMakeFiles/crazyflie_server.dir/build.make:120: recipe for target '/home/ardrone/crazyflie_ws/devel/lib/crazyflie_driver/crazyflie_server' failed
make[2]: *** [/home/ardrone/crazyflie_ws/devel/lib/crazyflie_driver/crazyflie_server] Error 1
CMakeFiles/Makefile2:2729: recipe for target 'crazyflie_ros/crazyflie_driver/CMakeFiles/crazyflie_server.dir/all' failed
make[1]: *** [crazyflie_ros/crazyflie_driver/CMakeFiles/crazyflie_server.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

whoenig, 2019-10-09:
Instead of downloading separately, please follow the instructions at https://github.com/whoenig/crazyflie_ros#installation (which will pull the correct versions of those submodules).

Allenhanbo, 2019-10-09:
Thanks for replying. I follow the instructions to git clone again. But the directory of crazyflie_cpp and crazyflie_tool is still empty.

Allenhanbo, 2019-10-10:
Thanks, I've already catkin_make successfully. Then I tried to hover my drone using: roslaunch crazyflie_demo hover_vicon.launch uri:=radio://0/80/2M frame:=/vicon/crazyflie/crazyflie x:=0 y:=0 z:=1
The erroe occured:

ardrone@ubuntu:~$ roslaunch crazyflie_demo hover_vicon.launch uri:=radio://0/80/2M frame:=/vicon/crazyflie/crazyflie x:=0 y:=0 z:=1
... logging to /home/ardrone/.ros/log/163defac-eaf3-11e9-bac4-000c29a0daf3/roslaunch-ubuntu-13254.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
xacro.py is deprecated; please use xacro instead
started roslaunch server http://ubuntu:39279/

SUMMARY
========

PARAMETERS
 * /crazyflie/controller/PIDs/X/integratorMax: 0.1
 * /crazyflie/controller/PIDs/X/integratorMin: -0.1
 * /crazyflie/controller/PIDs/X/kd: 20.0
 * /crazyflie/controller/PIDs/X/ki: 2.0
 * /crazyflie/controller/PIDs/X/kp: 40.0
 * /crazyflie/controller/PIDs/X/maxOutput: 10.0
 * /crazyflie/controller/PIDs/X/minOutput: -10.0
 * /crazyflie/controller/PIDs/Y/integratorMax: 0.1
 * /crazyflie/controller/PIDs/Y/integratorMin: -0.1
 * /crazyflie/controller/PIDs/Y/kd: -20.0
 * /crazyflie/controller/PIDs/Y/ki: -2.0
 * /crazyflie/controller/PIDs/Y/kp: -40.0
 * /crazyflie/controller/PIDs/Y/maxOutput: 10.0
 * /crazyflie/controller/PIDs/Y/minOutput: -10.0
 * /crazyflie/controller/PIDs/Yaw/integratorMax: 0.0
 * /crazyflie/controller/PIDs/Yaw/integratorMin: 0.0
 * /crazyflie/controller/PIDs/Yaw/kd: -20.0
 * /crazyflie/controller/PIDs/Yaw/ki: 0.0
 * /crazyflie/controller/PIDs/Yaw/kp: -200.0
 * /crazyflie/controller/PIDs/Yaw/maxOutput: 200.0
 * /crazyflie/controller/PIDs/Yaw/minOutput: -200.0
 * /crazyflie/controller/PIDs/Z/integratorMax: 1000.0
 * /crazyflie/controller/PIDs/Z/integratorMin: -1000.0
 * /crazyflie/controller/PIDs/Z/kd: 6000.0
 * /crazyflie/controller/PIDs/Z/ki: 3500.0
 * /crazyflie/controller/PIDs/Z/kp: 5000.0
 * /crazyflie/controller/PIDs/Z/maxOutput: 60000.0
 * /crazyflie/controller/PIDs/Z/minOutput: 10000.0
 * /crazyflie/controller/frame: /vicon/crazyflie/...
 * /crazyflie/controller/worldFrame: world
 * /crazyflie/crazyflie_add/enable_logging: False
 * /crazyflie/crazyflie_add/enable_logging_battery: True
 * /crazyflie/crazyflie_add/enable_logging_imu: True
 * /crazyflie/crazyflie_add/enable_logging_magnetic_field: True
 * /crazyflie/crazyflie_add/enable_logging_packets: True
 * /crazyflie/crazyflie_add/enable_logging_pose: False
 * /crazyflie/crazyflie_add/enable_logging_pressure: True
 * /crazyflie/crazyflie_add/enable_logging_temperature: True
 * /crazyflie/crazyflie_add/pitch_trim: 0
 * /crazyflie/crazyflie_add/roll_trim: 0
 * /crazyflie/crazyflie_add/tf_prefix: crazyflie
 * /crazyflie/crazyflie_add/uri: radio://0/80/2M
 * /crazyflie/crazyflie_add/use_ros_time: True
 * /crazyflie/joy/dev: /dev/input/js0
 * /crazyflie/joystick_controller/use_crazyflie_controller: True
 * /crazyflie/pose/name: goal
 * /crazyflie/pose/rate: 30
 * /crazyflie/pose/x: 0
 * /crazyflie/pose/y: 0
 * /crazyflie/pose/z: 1
 * /robot_description: <?xml version="1....
[... 54 lines cut ...]
[ INFO] [1570666540.716052618]: ... done, advertised as " vicon/crazyflie1/crazyflie1" 
terminate called after throwing an instance of 'std::runtime_error'
  what():  timeout
[crazyflie_server-2] process has died [pid 13297, exit code -6, cmd /home/ardrone/crazyflie_ws/devel/lib/crazyflie_driver/crazyflie_server __name:=crazyflie_server __log:=/home/ardrone/.ros/log/163defac-eaf3-11e9-bac4-000c29a0daf3/crazyflie_server-2.log].
log file: /home/ardrone/.ros/log/163defac-eaf3-11e9-bac4-000c29a0daf3/crazyflie_server-2*.log
^C[rviz-10] killing on exit
[vicon-9] killing on exit
[crazyflie/baselink_broadcaster-8] killing on exit
[crazyflie/joystick_controller-5] killing on exit
[crazyflie/joy-4] killing on exit
[crazyflie/controller-6] killing on exit
[crazyflie/pose-7] killing on exit
Traceback (most recent call last):
  File "/home/ardrone/crazyflie_ws/src/crazyflie_ros/crazyflie_demo/scripts/controller.py", line 62, in <module>
    controller = Controller(use_controller, joy_topic)
  File "/home/ardrone/crazyflie_ws/src/crazyflie_ros/crazyflie_demo/scripts/controller.py", line 10, in __init__
    rospy.wait_for_service('update_params')
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 159, in wait_for_service
    raise ROSInterruptException("rospy shutdown")
rospy.exceptions.ROSInterruptException: rospy shutdown
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

closed 2019-10-10 by Allenhanbo