whoenig/crazyflie_ros #151: Singly Crazyflie flying issue using vicon

Allenhanbo, 2019-10-14:
Rencently, I used vicon and crazyflie to try single drone hover using roslaunch crazyflie_demo hover_vicon.launch uri:=radio://0/80/2M/E7E7E7E7E7 frame:=vicon/cf1/cf1 x:=0 y:=0 z:=0.5.
Everything goes well, there's no error occured. I can see my drone in Rviz window. When I press my green button to fly, the drone did not fly upward, but fly undirectional. I use my flow deck to check the controller, theres no problem, my drone can hover using flow deck. I don't know which part cause this issue. Could you give me some advice? Thanks a lot! 

hc@hc-Lenovo:~$ roslaunch crazyflie_demo hover_vicon.launch uri:=radio://0/80/2M/E7E7E7E7E7 frame:=vicon/cf1/cf1 x:=0 y:=0 z:=0.3
... logging to /home/hc/.ros/log/7b5f521a-eebb-11e9-99be-680715cf7bd2/roslaunch-hc-Lenovo-20131.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
xacro.py is deprecated; please use xacro instead
started roslaunch server http://hc-Lenovo:35801/

SUMMARY
========

PARAMETERS
 * /crazyflie/controller/PIDs/X/integratorMax: 0.1
 * /crazyflie/controller/PIDs/X/integratorMin: -0.1
 * /crazyflie/controller/PIDs/X/kd: 20.0
 * /crazyflie/controller/PIDs/X/ki: 2.0
 * /crazyflie/controller/PIDs/X/kp: 40.0
 * /crazyflie/controller/PIDs/X/maxOutput: 10.0
 * /crazyflie/controller/PIDs/X/minOutput: -10.0
 * /crazyflie/controller/PIDs/Y/integratorMax: 0.1
 * /crazyflie/controller/PIDs/Y/integratorMin: -0.1
 * /crazyflie/controller/PIDs/Y/kd: -20.0
 * /crazyflie/controller/PIDs/Y/ki: -2.0
 * /crazyflie/controller/PIDs/Y/kp: -40.0
 * /crazyflie/controller/PIDs/Y/maxOutput: 10.0
 * /crazyflie/controller/PIDs/Y/minOutput: -10.0
 * /crazyflie/controller/PIDs/Yaw/integratorMax: 0.0
 * /crazyflie/controller/PIDs/Yaw/integratorMin: 0.0
 * /crazyflie/controller/PIDs/Yaw/kd: -20.0
 * /crazyflie/controller/PIDs/Yaw/ki: 0.0
 * /crazyflie/controller/PIDs/Yaw/kp: -200.0
 * /crazyflie/controller/PIDs/Yaw/maxOutput: 200.0
 * /crazyflie/controller/PIDs/Yaw/minOutput: -200.0
 * /crazyflie/controller/PIDs/Z/integratorMax: 1000.0
 * /crazyflie/controller/PIDs/Z/integratorMin: -1000.0
 * /crazyflie/controller/PIDs/Z/kd: 6000.0
 * /crazyflie/controller/PIDs/Z/ki: 3500.0
 * /crazyflie/controller/PIDs/Z/kp: 5000.0
 * /crazyflie/controller/PIDs/Z/maxOutput: 60000.0
 * /crazyflie/controller/PIDs/Z/minOutput: 10000.0
 * /crazyflie/controller/frame: vicon/cf1/cf1
 * /crazyflie/controller/worldFrame: world
 * /crazyflie/crazyflie_add/enable_logging: False
 * /crazyflie/crazyflie_add/enable_logging_battery: True
 * /crazyflie/crazyflie_add/enable_logging_imu: True
 * /crazyflie/crazyflie_add/enable_logging_magnetic_field: True
 * /crazyflie/crazyflie_add/enable_logging_packets: True
 * /crazyflie/crazyflie_add/enable_logging_pose: False
 * /crazyflie/crazyflie_add/enable_logging_pressure: True
 * /crazyflie/crazyflie_add/enable_logging_temperature: True
 * /crazyflie/crazyflie_add/pitch_trim: 0
 * /crazyflie/crazyflie_add/roll_trim: 0
 * /crazyflie/crazyflie_add/tf_prefix: crazyflie
 * /crazyflie/crazyflie_add/uri: radio://0/80/2M/E...
 * /crazyflie/crazyflie_add/use_ros_time: True
 * /crazyflie/joy/dev: /dev/input/js0
 * /crazyflie/joystick_controller/use_crazyflie_controller: True
 * /crazyflie/pose/name: goal
 * /crazyflie/pose/rate: 30
 * /crazyflie/pose/x: 0
 * /crazyflie/pose/y: 0
 * /crazyflie/pose/z: 0.3
 * /robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /vicon/datastream_hostport: 192.168.0.100
 * /vicon/stream_mode: ClientPull
 * /vicon/tf_ref_frame_id: /world

NODES
  /crazyflie/
    baselink_broadcaster (tf/static_transform_publisher)
    controller (crazyflie_controller/crazyflie_controller)
    crazyflie_add (crazyflie_driver/crazyflie_add)
    joy (joy/joy_node)
    joystick_controller (crazyflie_demo/controller.py)
    pose (crazyflie_demo/publish_pose.py)
  /
    crazyflie_server (crazyflie_driver/crazyflie_server)
    rviz (rviz/rviz)
    vicon (vicon_bridge/vicon_bridge)

auto-starting new master
process[master]: started with pid [20167]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 7b5f521a-eebb-11e9-99be-680715cf7bd2
process[rosout-1]: started with pid [20189]
started core service [/rosout]
process[crazyflie_server-2]: started with pid [20206]
process[crazyflie/crazyflie_add-3]: started with pid [20207]
process[crazyflie/joy-4]: started with pid [20214]
process[crazyflie/joystick_controller-5]: started with pid [20234]
process[crazyflie/controller-6]: started with pid [20248]
process[crazyflie/pose-7]: started with pid [20270]
[ INFO] [1571082461.837633462]: wait_for_service /add_crazyflie
[ INFO] [1571082461.839857230]: found /add_crazyflie
process[crazyflie/baselink_broadcaster-8]: started with pid [20283]
process[vicon-9]: started with pid [20301]
[ INFO] [1571082461.862984331]: Adding radio://0/80/2M/E7E7E7E7E7 as crazyflie with trim(0.000000, 0.000000). Logging: 1, Parameters: 0, Use ROS time: 1
process[rviz-10]: started with pid [20319]
[ERROR] [1571082461.938217273]: Couldn't open joystick force feedback!
[ INFO] [1571082461.938333856]: Opened joystick: /dev/input/js0. deadzone_: 0.050000.
[ INFO] [1571082461.940935946]: Connecting to Vicon DataStream SDK at 192.168.0.100 ...
[ INFO] [1571082461.946260196]: .
[crazyflie/crazyflie_add-3] process has finished cleanly
log file: /home/hc/.ros/log/7b5f521a-eebb-11e9-99be-680715cf7bd2/crazyflie-crazyflie_add-3*.log
[ INFO] [1571082462.016232568]: Requesting parameters...
[ WARN] [1571082462.018749321]: Don't know ack: Port: 13 Channel: 1 Len: 6
[ WARN] [1571082462.020586815]: Don't know ack: Port: 13 Channel: 1 Len: 6
[ INFO] [1571082462.021568961]: Found variables in cache.
[ INFO] [1571082462.335173449]: Requesting memories...
[ INFO] [1571082462.337309453]: Memories: 9
[ INFO] [1571082462.349573091]: Ready...
[ INFO] [1571082462.349612209]: Elapsed: 0.336350 s
[INFO] [1571082462.439883]: found update_params service
[INFO] [1571082462.440281]: waiting for emergency service
[INFO] [1571082462.441801]: found emergency service
[INFO] [1571082462.442389]: waiting for land service
[INFO] [1571082462.443890]: found land service
[INFO] [1571082462.444289]: waiting for takeoff service
[INFO] [1571082462.445825]: found takeoff service
[ INFO] [1571082462.946529048]: ... connected!
[ INFO] [1571082462.946634219]: Setting Stream Mode to ClientPull: Success
[ INFO] [1571082462.946698540]: Axis Mapping: X-Forward Y-Left Z-Up
[ INFO] [1571082462.946756434]: Version: 1.3.0
[ INFO] [1571082462.946791225]: setting up grab_vicon_pose service server ... 
[ INFO] [1571082462.949037669]: setting up segment calibration service server ... 
[ WARN] [1571082462.953186643]: grab frame returned false
[ WARN] [1571082462.970832948]: cf1 occluded, not publishing... 
[ INFO] [1571082462.979719208]: creating new object cf1/cf1 ...
[ WARN] [1571082462.983611292]: unable to load zero pose for cf1/cf1
[ INFO] [1571082462.983681665]: ... done, advertised as " vicon/cf1/cf1" 
[ INFO] [1571082464.774065169]: Takeoff requested!

Allenhanbo, 2019-10-14:
By the way, I can see the red arrow in Rviz, but it did not point to Z-axis, it point to x-y plane. Is it the normal case?

whoenig, 2019-10-15:
That example uses an off-board position controller that I don't recommend for new experiments. Instead, you can use the on-board EKF and controller. The documentation in crazyflie_ros lacks behind on this, but you can check external_pose_vicon.launch and https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_demo/scripts/test_high_level.py. The latter would also work just with the flow deck.

Allenhanbo, 2019-10-15:
Hi whoenig, thanks for your reply. I still have two queations, if you could help me explain is. First, for your meaning, if I want to achieve the crazyflie hovering using vicon, I should add the  https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_demo/scripts/test_high_level.py to external_pose_vicon.launch and abandon the old launch file hover_vicon.launch?

Second, if I just want to use flow deck without vicon, how could I change the test_high_level.py? Is it to add the uri address of crazyflie? 

Thanks a lot.

whoenig, 2019-10-15:
1. No need to change `external_pose_vicon.launch`. You can simply run it and in a second terminal window execute `test_high_level.py`.
2. You can use one of the telop launch files, or create a custom launch file that simply connects to your CF (but does not launch any other nodes). Then, as before, execute `test_high_level.py` in a separate window.

Allenhanbo, 2019-10-15:
I really appreciate for your explain, it's a big help for me. I used the external_position_vicon.launch and test_hign_level.py to achieve autonomous flying. 
For my understanding, this two file just let the drone to fly in a certain trajectory using .csv file saved before flying. If I want my drone to fly in an absolutely autonomous flying in my vicon volume space, like flying to certain point, how can I do that? Do you have some advices? Thanks!

Allenhanbo, 2019-10-15:
I find there's a launch file named multi_waypoint_vicon.launch. This file seems like related to waypoint flying, but the controller in this launch file is also off-board controller.

whoenig, 2019-10-16:
You can use the goTo function (https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_demo/scripts/crazyflie.py#L45-L47), or plan a trajectory that executes your behavior (helper scripts are in https://github.com/whoenig/uav_trajectories).