whoenig/crazyflie_ros #152: On-board controller and EKF source code

Allenhanbo, 2019-10-18:
Hi Whoening, I was trying to use test_high_level.py and vicon to fly drone. You told me that this script is using the onboard controller. How can I access to the source code to see how the controller and EKF is designed? Thanks.

JIMMMY0, 2019-10-21:
Check the crazyflie firmware [here](https://github.com/bitcraze/crazyflie-firmware/tree/master/src/modules/src).
You can find them in the crazyswarm repository also.

Allenhanbo, 2019-10-21:
Hi Jimmyo, thanks for your reply. I have found the source code from the link you send to me. For the  mellinger controller used in line 15 of test_high_level.py(https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_demo/scripts/test_high_level.py), I can find the source code in there https://github.com/bitcraze/crazyflie-firmware/blob/master/src/modules/interface/controller_mellinger.h

For the EKF used in test_high_level.py, could you tell me which file should I read through in the link you sent to me?

Thanks a lot.