whoenig/crazyflie_ros #164: Enable multiple dongles

ZPHuang-ACM, 2020-01-15:
Hi Whoenig, 
I am doing research in cooperative control of multi-agent systems. 

I noticed that the control performances degraded significantly when I applied my control algorithms to a group of three CFs with one dongle. I would like to enable multiple dongles (possibly one for each CF) to increase communication to verify if it was a communication issue or a deficiency of my algorithm. 

I wonder if it is trivial to enable multiple dongles in crazflie_ros? If so, could you please give me some guidelines? 

Any help will be appreciated.

whoenig, 2020-01-16:
Yes, just plug in multiple radios and then use `radio://1/...` `radio://2/...` etc for the other dongles. Make sure that you give each CF a unique address though. Also, use a different channel for each radio (and CF) to avoid packet collisions.

ZPHuang-ACM, 2020-01-16:
Hi Whoenig, 
It worked, thank you for your help.

closed 2020-01-16 by whoenig