whoenig/crazyflie_ros #165: Issues in controlling multiple Crazyflies using multiple USB dongles

ZPHuang-ACM, 2020-01-30:
Hi Whoring,

I was able to control three Crazyflies using three USB dongles (specified for using different channels). However, the control performance did not improve too much from just using one dongle (had lots of waggling even using three dongles). 

Bt the way, I am using a USB hub for connecting three dongles. 

Any thoughts will be appreciated!

whoenig, 2020-02-04:
Using a USB Hub should be fine. Which controller are you using?

ZPHuang-ACM, 2020-02-08:
Hi Whoenig,
I just used the PID controller provided by crazyflie_ros. It worked fine when I was controlling one Crazyflie. However, one of the Crazyflie was not stable when controlling three Crazyflies. 

I wonder if it was the issue of airflow generated from other Crazyflie or just the inherent instability of that specific Crazyflie?

whoenig, 2020-02-10:
In general, I don't recommend using the provided PID controller anymore. There are better on-board controllers available.

For the airflow: This can be indeed a problem. As a rule of thumb, you need a vertical safety distance of 0.6m and a horizontal one of 0.3m.

ZPHuang-ACM, 2020-02-12:
Hi Whoenig,
Thank you for your responses.
For the onboard controller, is it the one you used in Crazyswam? If not, could you please send me some relevant papers?

whoenig, 2020-02-12:
There are a three controllers that come with the firmware. You can find a good overview here: https://www.bitcraze.io/2020/02/out-of-control/. The one that the Crazyswarm uses by default is the Mellinger controller.

ZPHuang-ACM, 2020-02-12:
Thank you. This link is very helpful.

closed 2020-02-12 by ZPHuang-ACM