whoenig/crazyflie_ros #166: Crazyflie 2.1 Motors don’t start

OmarShadeed, 2020-02-06:
Hi Whoeing, 
I run the demo_localization.launch file and I get connected to the CF and I see values being published to the /crazyflie/cmd_vel topic, however, the motors don’t work. I have two Linux PCs on one of them the CF works and on the other it doesn’t. I also tried two different CFs and I still get the same problem. I checked the imu topic to make sure I’m connected to the CF and I could get reading from the imu successfully but still can’t start the motors. 
Thanks,

whoenig, 2020-02-07:
Do you manually send data to the cmd_vel topic? I remember that the firmware used to have a thrust lock, where the motors would only spin if you initially send (0,0,0,0) a couple of times to the cmd_vel topic. Otherwise there are various error states in which the CF won't spin the motors, but I guess you covered that possibility by testing multiple CFs.

codygou, 2020-12-11:
> Hi Whoeing,
> I run the demo_localization.launch file and I get connected to the CF and I see values being published to the /crazyflie/cmd_vel topic, however, the motors don’t work. I have two Linux [...]

Hi, I met the same issue, and I check the callback(), the thrust is always 0, should be mapped. but how to handle?

whoenig, 2020-12-21:
Are you trying to fly with a joystick? What does your launch file look like (`demo_localization.launch` does not seem to be part of this package.)