whoenig/crazyflie_ros #168: catkin_make: Invoking "make -j12 -l12" failed

PranjaLBiswas27, 2020-02-16:
Hi,
Can someone help me with this error? What exactly is going wrong during **catkin_make**. I am getting **Invoking "make -j12 -l12" failed** error and I dont see and relevant threads on google. I have already done ´git submodue init´ and ´git submodule update´. Please let me know.
Kind Regards
Pranjal

> root@devpcX:/home/robocup/crazyflie/ros/crazyflie_catkin_ws# ls
build  devel  src
root@devpcX:/home/robocup/crazyflie/ros/crazyflie_catkin_ws# catkin_make
Base path: /home/robocup/crazyflie/ros/crazyflie_catkin_ws
Source space: /home/robocup/crazyflie/ros/crazyflie_catkin_ws/src
Build space: /home/robocup/crazyflie/ros/crazyflie_catkin_ws/build
Devel space: /home/robocup/crazyflie/ros/crazyflie_catkin_ws/devel
Install space: /home/robocup/crazyflie/ros/crazyflie_catkin_ws/install
####
> #### Running command: "make cmake_check_build_system" in "/home/robocup/crazyflie/ros/crazyflie_catkin_ws/build"
####
####
> #### Running command: "make -j12 -l12" in "/home/robocup/crazyflie/ros/crazyflie_catkin_ws/build"
####
> [  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_lisp
[  1%] Building CXX object crazyflie_ros/crazyflie_cpp/CMakeFiles/crazyflie_cpp.dir/src/USBDevice.cpp.o
[  2%] Building CXX object crazyflie_ros/crazyflie_cpp/CMakeFiles/crazyflie_cpp.dir/src/Crazyradio.cpp.o
[  2%] Linking CXX executable /home/robocup/crazyflie/ros/crazyflie_catkin_ws/devel/lib/crazyflie_demo/quadrotor_teleop
[  2%] Building CXX object crazyflie_ros/crazyflie_cpp/CMakeFiles/crazyflie_cpp.dir/src/CrazyflieUSB.cpp.o
[  2%] Linking CXX executable /home/robocup/crazyflie/ros/crazyflie_catkin_ws/devel/lib/crazyflie_controller/crazyflie_controller
[  2%] Built target _crazyflie_driver_generate_messages_check_deps_Takeoff
[  2%] Built target _crazyflie_driver_generate_messages_check_deps_Land
[  2%] Built target _crazyflie_driver_generate_messages_check_deps_Position
[  2%] Built target _bitcraze_lps_estimator_generate_messages_check_deps_RangeArray
[  2%] Built target _crazyflie_driver_generate_messages_check_deps_crtpPacket
[  2%] Built target _crazyflie_driver_generate_messages_check_deps_sendPacket
[  2%] Built target _crazyflie_driver_generate_messages_check_deps_Hover
[  2%] Built target _crazyflie_driver_generate_messages_check_deps_LogBlock
[  2%] Built target _crazyflie_driver_generate_messages_check_deps_UploadTrajectory
[  2%] Built target _crazyflie_driver_generate_messages_check_deps_RemoveCrazyflie
[  2%] Built target _crazyflie_driver_generate_messages_check_deps_FullState
[  2%] Built target _crazyflie_driver_generate_messages_check_deps_VelocityWorld
[  2%] Built target _crazyflie_driver_generate_messages_check_deps_SetGroupMask
[  2%] Built target _crazyflie_driver_generate_messages_check_deps_TrajectoryPolynomialPiece
[  2%] Built target _crazyflie_driver_generate_messages_check_deps_UpdateParams
[  2%] Built target _crazyflie_driver_generate_messages_check_deps_GoTo
[  2%] Built target geometry_msgs_generate_messages_py
[  2%] Built target _crazyflie_driver_generate_messages_check_deps_StartTrajectory
[  2%] Built target _crazyflie_driver_generate_messages_check_deps_Stop
[  2%] Built target _crazyflie_driver_generate_messages_check_deps_GenericLogData
[  2%] Built target _crazyflie_driver_generate_messages_check_deps_AddCrazyflie
[  2%] Built target geometry_msgs_generate_messages_lisp
[  2%] Built target geometry_msgs_generate_messages_eus
[  2%] Built target geometry_msgs_generate_messages_nodejs
[  3%] Built target bitcraze_lps_estimator_generate_messages_py
[  5%] Built target bitcraze_lps_estimator_generate_messages_eus
[  5%] Built target bitcraze_lps_estimator_generate_messages_cpp
[  6%] Built target bitcraze_lps_estimator_generate_messages_nodejs
[  6%] Built target bitcraze_lps_estimator_generate_messages_lisp
[ 19%] Built target crazyflie_driver_generate_messages_cpp
[ 33%] Built target crazyflie_driver_generate_messages_py
[ 46%] Built target crazyflie_driver_generate_messages_lisp
[ 59%] Built target crazyflie_driver_generate_messages_nodejs
[ 73%] Built target crazyflie_driver_generate_messages_eus
[ 73%] Built target bitcraze_lps_estimator_generate_messages
Scanning dependencies of target crazyflie_driver_generate_messages
CMakeFiles/quadrotor_teleop.dir/src/quadrotor_teleop.cpp.o: In function `main':
quadrotor_teleop.cpp:(.text+0x61): undefined reference to `ros::init(int&, char**, std::string const&, unsigned int)'
CMakeFiles/quadrotor_teleop.dir/src/quadrotor_teleop.cpp.o: In function `Teleop::Teleop()':
quadrotor_teleop.cpp:(.text._ZN6TeleopC2Ev[_ZN6TeleopC5Ev]+0x4f): undefined reference to `ros::NodeHandle::NodeHandle(std::string const&, std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > const&)'
[... 78 lines cut ...]
/home/robocup/crazyflie/ros/crazyflie_catkin_ws/src/crazyflie_ros/crazyflie_cpp/src/Crazyradio.cpp: In member function ‘virtual void Crazyradio::sendPacket(const uint8_t*, uint32_t, ITransport::Ack&)’:
/home/robocup/crazyflie/ros/crazyflie_catkin_ws/src/crazyflie_ros/crazyflie_cpp/src/Crazyradio.cpp:167:58: error: ‘libusb_error_name’ was not declared in this scope
/home/robocup/crazyflie/ros/crazyflie_catkin_ws/src/crazyflie_ros/crazyflie_cpp/src/Crazyradio.cpp:187:58: error: ‘libusb_error_name’ was not declared in this scope
/home/robocup/crazyflie/ros/crazyflie_catkin_ws/src/crazyflie_ros/crazyflie_cpp/src/Crazyradio.cpp: In member function ‘virtual void Crazyradio::sendPacketNoAck(const uint8_t*, uint32_t)’:
/home/robocup/crazyflie/ros/crazyflie_catkin_ws/src/crazyflie_ros/crazyflie_cpp/src/Crazyradio.cpp:224:58: error: ‘libusb_error_name’ was not declared in this scope
/home/robocup/crazyflie/ros/crazyflie_catkin_ws/src/crazyflie_ros/crazyflie_cpp/src/Crazyradio.cpp: In member function ‘virtual void Crazyradio::send2PacketsNoAck(const uint8_t*, uint32_t)’:
/home/robocup/crazyflie/ros/crazyflie_catkin_ws/src/crazyflie_ros/crazyflie_cpp/src/Crazyradio.cpp:260:58: error: ‘libusb_error_name’ was not declared in this scope
crazyflie_ros/crazyflie_cpp/CMakeFiles/crazyflie_cpp.dir/build.make:62: recipe for target 'crazyflie_ros/crazyflie_cpp/CMakeFiles/crazyflie_cpp.dir/src/USBDevice.cpp.o' failed
make[2]: *** [crazyflie_ros/crazyflie_cpp/CMakeFiles/crazyflie_cpp.dir/src/USBDevice.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
crazyflie_ros/crazyflie_cpp/CMakeFiles/crazyflie_cpp.dir/build.make:86: recipe for target 'crazyflie_ros/crazyflie_cpp/CMakeFiles/crazyflie_cpp.dir/src/Crazyradio.cpp.o' failed
make[2]: *** [crazyflie_ros/crazyflie_cpp/CMakeFiles/crazyflie_cpp.dir/src/Crazyradio.cpp.o] Error 1
/home/robocup/crazyflie/ros/crazyflie_catkin_ws/src/crazyflie_ros/crazyflie_cpp/src/CrazyflieUSB.cpp: In member function ‘virtual void CrazyflieUSB::sendPacket(const uint8_t*, uint32_t, ITransport::Ack&)’:
/home/robocup/crazyflie/ros/crazyflie_catkin_ws/src/crazyflie_ros/crazyflie_cpp/src/CrazyflieUSB.cpp:61:58: error: ‘libusb_error_name’ was not declared in this scope
/home/robocup/crazyflie/ros/crazyflie_catkin_ws/src/crazyflie_ros/crazyflie_cpp/src/CrazyflieUSB.cpp:80:58: error: ‘libusb_error_name’ was not declared in this scope
/home/robocup/crazyflie/ros/crazyflie_catkin_ws/src/crazyflie_ros/crazyflie_cpp/src/CrazyflieUSB.cpp: In member function ‘virtual void CrazyflieUSB::sendPacketNoAck(const uint8_t*, uint32_t)’:
/home/robocup/crazyflie/ros/crazyflie_catkin_ws/src/crazyflie_ros/crazyflie_cpp/src/CrazyflieUSB.cpp:115:58: error: ‘libusb_error_name’ was not declared in this scope
crazyflie_ros/crazyflie_cpp/CMakeFiles/crazyflie_cpp.dir/build.make:110: recipe for target 'crazyflie_ros/crazyflie_cpp/CMakeFiles/crazyflie_cpp.dir/src/CrazyflieUSB.cpp.o' failed
make[2]: *** [crazyflie_ros/crazyflie_cpp/CMakeFiles/crazyflie_cpp.dir/src/CrazyflieUSB.cpp.o] Error 1
CMakeFiles/Makefile2:462: recipe for target 'crazyflie_ros/crazyflie_cpp/CMakeFiles/crazyflie_cpp.dir/all' failed
make[1]: *** [crazyflie_ros/crazyflie_cpp/CMakeFiles/crazyflie_cpp.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
**Invoking "make -j12 -l12" failed**
root@devpcX:/home/robocup/crazyflie/ros/crazyflie_catkin_ws#

whoenig, 2020-02-16:
Looks like the linking step fails and catkin doesn't link against the ROS libraries. What OS and ROS versions are you using?

PranjaLBiswas27, 2020-02-16:
I am currently using Ubuntu 16.04 and ROS Kinetic.

whoenig, 2020-02-16:
For the libusb errors you can try `sudo apt-get install libusb-dev`. For the errors in the controller you might need to add roscpp to the package.xml (this seems a bug, but strange that nobody else ran into it so far).

PranjaLBiswas27, 2020-02-16:
I ran the following 
> sudo apt-get install libusb-dev

but it is already there, nothing to install and upgrade.

Also, I think roscpp already exists here, no?https://github.com/whoenig/crazyflie_ros/blob/c42af1b0bd49ca72e3b01803de62a27974e18771/crazyflie_demo/package.xml#L18

Yes, it is odd, I have not found any threads related to this as well. What will you suggest me to check?

whoenig, 2020-02-16:
libusb: Can you make sure it is version 1.0 and not 0.1? For me (newer Ubuntu), there is a package libusb-1.0-0-dev. 

roscpp: It seems to be missing in https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_controller/package.xml and https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_controller/CMakeLists.txt, which is one of the projects that fails for you. 

Until a while ago we had travis CI working with xenial, and there seems to be no special packages needed apart from a ROS base (see https://github.com/whoenig/crazyflie_ros/blob/mast/.travis.yml). So this is strange indeed. Do you have trouble with any other ROS packages on your machine?

PranjaLBiswas27, 2020-02-16:
**libusb**: I see the following installed packages on executing dpkh -l. I think lbusb 1.0 is available.
> ii  libusb-0.1-4:amd64     2:0.1.12-28      amd64            userspace USB programming library
ii  libusb-1.0-0:amd64     2:1.0.20-1       amd64            userspace USB programming library
ii  libusb-1.0-0-dev:amd64 2:1.0.20-1       amd64            userspace USB programming library development file
ii  libusb-1.0-doc         2:1.0.20-1       all              documentation for userspace USB programming
ii  libusb-dev             2:0.1.12-28      amd64            userspace USB programming library development file
ii  libusbmuxd4:amd64      1.0.10-2ubuntu0. amd64            USB multiplexor daemon for iPhone and iPod Touch d

**roscpp**: I added it here https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_controller/package.xml as follows
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
<?xml version="1.0"?>
<package>
  <name>crazyflie_controller</name>
  <version>0.0.1</version>
  <description>Controller to fly a Crazyflie in a space with location feedback (such as motion capture).</description>

  <maintainer email="whoenig@usc.edu">Wolfgang Hoenig</maintainer>

  <license>MIT</license>

  <url type="website">http://ros.org/wiki/crazyflie</url>
  <url type="bugtracker">https://github.com/whoenig/crazyflie_ros/issues</url>

  <author email="whoenig@usc.edu">Wolfgang Hoenig</author>

  <buildtool_depend>catkin</buildtool_depend>
  <build_depend>std_msgs</build_depend>
  <build_depend>tf</build_depend>
  <build_depend>**roscpp**</build_depend>  

  <run_depend>std_msgs</run_depend>
  <run_depend>tf</run_depend>
  <run_depend>**roscpp**</run_depend>
</package>
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%

Next, where and how exactly should I add roscpp in https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_controller/CMakeLists.txt. Sorry if this is too trivial, I do not have much experience with cmake.

Also,  https://github.com/whoenig/crazyflie_ros/blob/mast/.travis.yml is not accessible, gives Error 404 (page not found).

whoenig, 2020-02-17:
* Add "roscpp" after https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_controller/CMakeLists.txt#L6 and "CATKIN_DEPENDS roscpp" at https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_controller/CMakeLists.txt#L81.

* Sorry about the link - it should have been: https://github.com/whoenig/crazyflie_ros/blob/master/.travis.yml

* libusb: it is possible that cmake finds the old libusb (v0.1). If you do not need it for any other project, I suggest uninstalling that version. Otherwise we'll have to find a way to tell cmake to only use the new version (1.0).

PranjaLBiswas27, 2020-02-17:
**- I removed libusb 0.1-4 and made the following edits:**
1.  Added  "roscpp" after https://github.com/whoenig/crazyflie_ros/blob/c42af1b0bd49ca72e3b01803de62a27974e18771/crazyflie_controller/CMakeLists.txt#L6.
2. Replaced  "#  CATKIN_DEPENDS other_catkin_pkg" with "CATKIN_DEPENDS roscpp" at https://github.com/whoenig/crazyflie_ros/blob/c42af1b0bd49ca72e3b01803de62a27974e18771/crazyflie_controller/CMakeLists.txt#L81
3. Added <build_depend>roscpp</build_depend> after https://github.com/whoenig/crazyflie_ros/blob/c42af1b0bd49ca72e3b01803de62a27974e18771/crazyflie_controller/package.xml#L18
4. Added <run_depend>roscpp</run_depend> after https://github.com/whoenig/crazyflie_ros/blob/c42af1b0bd49ca72e3b01803de62a27974e18771/crazyflie_controller/package.xml#L21

**Executing catkin_make now gives the same error with the following error stack:**

>robocup@devpcX:~/crazyflie/ros/crazyflie_catkin_ws$ catkin_make
Base path: /home/robocup/crazyflie/ros/crazyflie_catkin_ws
Source space: /home/robocup/crazyflie/ros/crazyflie_catkin_ws/src
Build space: /home/robocup/crazyflie/ros/crazyflie_catkin_ws/build
Devel space: /home/robocup/crazyflie/ros/crazyflie_catkin_ws/devel
Install space: /home/robocup/crazyflie/ros/crazyflie_catkin_ws/install
Creating symlink "/home/robocup/crazyflie/ros/crazyflie_catkin_ws/src/CMakeLists.txt" pointing to "/opt/ros/kinetic/share/catkin/cmake/toplevel.cmake"
####
>#### Running command: "cmake /home/robocup/crazyflie/ros/crazyflie_catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/robocup/crazyflie/ros/crazyflie_catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/robocup/crazyflie/ros/crazyflie_catkin_ws/install -G Unix Makefiles" in "/home/robocup/crazyflie/ros/crazyflie_catkin_ws/build"
####
>-- The C compiler identification is GNU 4.7.4
-- The CXX compiler identification is GNU 4.7.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/robocup/crazyflie/ros/crazyflie_catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/kinetic
-- This workspace overlays: /opt/ros/kinetic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.12", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/robocup/crazyflie/ros/crazyflie_catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.12") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.20
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 8 packages in topological order:
-- ~~  - crazyflie (metapackage)
-- ~~  - crazyflie_cpp
-- ~~  - crazyflie_description
-- ~~  - crazyflie_tools
-- ~~  - bitcraze_lps_estimator
-- ~~  - crazyflie_demo
-- ~~  - crazyflie_controller
-- ~~  - crazyflie_driver
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin metapackage: 'crazyflie'
[... 319 lines cut ...]
controller.cpp:(.text._ZNK3ros9Publisher7publishIN13geometry_msgs6Twist_ISaIvEEEEEvRKT_[_ZNK3ros9Publisher7publishIN13geometry_msgs6Twist_ISaIvEEEEEvRKT_]+0x1b0): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
controller.cpp:(.text._ZNK3ros9Publisher7publishIN13geometry_msgs6Twist_ISaIvEEEEEvRKT_[_ZNK3ros9Publisher7publishIN13geometry_msgs6Twist_ISaIvEEEEEvRKT_]+0x2af): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
controller.cpp:(.text._ZNK3ros9Publisher7publishIN13geometry_msgs6Twist_ISaIvEEEEEvRKT_[_ZNK3ros9Publisher7publishIN13geometry_msgs6Twist_ISaIvEEEEEvRKT_]+0x3d6): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
controller.cpp:(.text._ZNK3ros9Publisher7publishIN13geometry_msgs6Twist_ISaIvEEEEEvRKT_[_ZNK3ros9Publisher7publishIN13geometry_msgs6Twist_ISaIvEEEEEvRKT_]+0x4ef): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
CMakeFiles/crazyflie_controller.dir/src/controller.cpp.o:controller.cpp:(.text._ZNK3ros9Publisher7publishIN13geometry_msgs6Twist_ISaIvEEEEEvRKT_[_ZNK3ros9Publisher7publishIN13geometry_msgs6Twist_ISaIvEEEEEvRKT_]+0x60d): more undefined references to `ros::console::initializeLogLocation(ros::console::LogLocation*, [... cut ...]
make[2]: *** [/home/robocup/crazyflie/ros/crazyflie_catkin_ws/devel/lib/crazyflie_controller/crazyflie_controller] Error 1
CMakeFiles/Makefile2:2737: recipe for target 'crazyflie_ros/crazyflie_controller/CMakeFiles/crazyflie_controller.dir/all' failed
make[1]: *** [crazyflie_ros/crazyflie_controller/CMakeFiles/crazyflie_controller.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j12 -l12" failed
robocup@devpcX:~/crazyflie/ros/crazyflie_catkin_ws$

PranjaLBiswas27, 2020-02-17:
**- I removed libusb 0.1-4 and made the following edits:**
1.  Added  "roscpp" after https://github.com/whoenig/crazyflie_ros/blob/c42af1b0bd49ca72e3b01803de62a27974e18771/crazyflie_controller/CMakeLists.txt#L6
2. Replaced  "#  CATKIN_DEPENDS other_catkin_pkg" with "CATKIN_DEPENDS roscpp" at https://github.com/whoenig/crazyflie_ros/blob/c42af1b0bd49ca72e3b01803de62a27974e18771/crazyflie_controller/CMakeLists.txt#L81
3. Added <build_depend>roscpp</build_depend> after https://github.com/whoenig/crazyflie_ros/blob/c42af1b0bd49ca72e3b01803de62a27974e18771/crazyflie_controller/package.xml#L18
4. Added <run_depend>roscpp</run_depend> after https://github.com/whoenig/crazyflie_ros/blob/c42af1b0bd49ca72e3b01803de62a27974e18771/crazyflie_controller/package.xml#L21

**Executing catkin_make now gives the same error with the following error stack:**

>robocup@devpcX:~/crazyflie/ros/crazyflie_catkin_ws$ catkin_make
Base path: /home/robocup/crazyflie/ros/crazyflie_catkin_ws
Source space: /home/robocup/crazyflie/ros/crazyflie_catkin_ws/src
Build space: /home/robocup/crazyflie/ros/crazyflie_catkin_ws/build
Devel space: /home/robocup/crazyflie/ros/crazyflie_catkin_ws/devel
Install space: /home/robocup/crazyflie/ros/crazyflie_catkin_ws/install
Creating symlink "/home/robocup/crazyflie/ros/crazyflie_catkin_ws/src/CMakeLists.txt" pointing to "/opt/ros/kinetic/share/catkin/cmake/toplevel.cmake"
####
> #### Running command: "cmake /home/robocup/crazyflie/ros/crazyflie_catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/robocup/crazyflie/ros/crazyflie_catkin_ws/devel [...]
####
>-- The C compiler identification is GNU 4.7.4
-- The CXX compiler identification is GNU 4.7.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/robocup/crazyflie/ros/crazyflie_catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/kinetic
-- This workspace overlays: /opt/ros/kinetic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.12", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/robocup/crazyflie/ros/crazyflie_catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.12") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.20
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 8 packages in topological order:
-- ~~  - crazyflie (metapackage)
-- ~~  - crazyflie_cpp
-- ~~  - crazyflie_description
-- ~~  - crazyflie_tools
-- ~~  - bitcraze_lps_estimator
-- ~~  - crazyflie_demo
-- ~~  - crazyflie_controller
-- ~~  - crazyflie_driver
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin metapackage: 'crazyflie'
[... 321 lines cut ...]
controller.cpp:(.text._ZNK3ros9Publisher7publishIN13geometry_msgs6Twist_ISaIvEEEEEvRKT_[_ZNK3ros9Publisher7publishIN13geometry_msgs6Twist_ISaIvEEEEEvRKT_]+0x1b0): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
controller.cpp:(.text._ZNK3ros9Publisher7publishIN13geometry_msgs6Twist_ISaIvEEEEEvRKT_[_ZNK3ros9Publisher7publishIN13geometry_msgs6Twist_ISaIvEEEEEvRKT_]+0x2af): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
controller.cpp:(.text._ZNK3ros9Publisher7publishIN13geometry_msgs6Twist_ISaIvEEEEEvRKT_[_ZNK3ros9Publisher7publishIN13geometry_msgs6Twist_ISaIvEEEEEvRKT_]+0x3d6): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
controller.cpp:(.text._ZNK3ros9Publisher7publishIN13geometry_msgs6Twist_ISaIvEEEEEvRKT_[_ZNK3ros9Publisher7publishIN13geometry_msgs6Twist_ISaIvEEEEEvRKT_]+0x4ef): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
CMakeFiles/crazyflie_controller.dir/src/controller.cpp.o:controller.cpp:(.text._ZNK3ros9Publisher7publishIN13geometry_msgs6Twist_ISaIvEEEEEvRKT_[_ZNK3ros9Publisher7publishIN13geometry_msgs6Twist_ISaIvEEEEEvRKT_]+0x60d): more undefined references to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)' follow
[... cut ...]
make[2]: *** [/home/robocup/crazyflie/ros/crazyflie_catkin_ws/devel/lib/crazyflie_controller/crazyflie_controller] Error 1
CMakeFiles/Makefile2:2737: recipe for target 'crazyflie_ros/crazyflie_controller/CMakeFiles/crazyflie_controller.dir/all' failed
make[1]: *** [crazyflie_ros/crazyflie_controller/CMakeFiles/crazyflie_controller.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j12 -l12" failed
robocup@devpcX:~/crazyflie/ros/crazyflie_catkin_ws$

PranjaLBiswas27, 2020-02-17:
By any chance does the gcc version has any role to play with repect to the linking error? (https://github.com/ros-drivers/velodyne/issues/166#issuecomment-391206432)

I have gcc 4.7.4, while the default in Ubuntu 16.04 is gcc 5.4.4.

PranjaLBiswas27, 2020-02-17:
After updating to gcc 5.5.0, now I get the dollowing error

> [ 77%] Built target quadrotor_teleop
[ 77%] Linking CXX executable /home/robocup/crazyflie/ros/crazyflie_catkin_ws/devel/lib/crazyflie_controller/crazyflie_controller
[ 77%] Built target crazyflie_controller
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j12 -l12" failed
robocup@devpcX:~/crazyflie/ros/crazyflie_catkin_ws$

whoenig, 2020-02-17:
Great finding! Unfortunately, the last build log seems somewhat incomplete. Make sure you clean first (`catkin_make clean`). If you still don't get a good error message, try `catkin_make -j1`. If that still doesn't help: `catkin_make VERBOSE=1` and post the output here.

PranjaLBiswas27, 2020-02-18:
**I entered only a section of the last build log actually. On executing catkin_make with gcc 5.5 the full build log is as follows:**

>robocup@devpcX:~/crazyflie/ros/crazyflie_original_catkin_ws$ catkin_make
Base path: /home/robocup/crazyflie/ros/crazyflie_original_catkin_ws
Source space: /home/robocup/crazyflie/ros/crazyflie_original_catkin_ws/src
Build space: /home/robocup/crazyflie/ros/crazyflie_original_catkin_ws/build
Devel space: /home/robocup/crazyflie/ros/crazyflie_original_catkin_ws/devel
Install space: /home/robocup/crazyflie/ros/crazyflie_original_catkin_ws/install
####
>#### Running command: "make cmake_check_build_system" in "/home/robocup/crazyflie/ros/crazyflie_original_catkin_ws/build"
####
####
>#### Running command: "make -j12 -l12" in "/home/robocup/crazyflie/ros/crazyflie_original_catkin_ws/build"
####
>[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_cpp
[  1%] Built target std_msgs_generate_messages_eus
[  1%] Building CXX object crazyflie_ros/crazyflie_cpp/CMakeFiles/crazyflie_cpp.dir/src/USBDevice.cpp.o
[  1%] Built target std_msgs_generate_messages_lisp
[  1%] Built target geometry_msgs_generate_messages_cpp
[  2%] Building CXX object crazyflie_ros/crazyflie_demo/CMakeFiles/quadrotor_teleop.dir/src/quadrotor_teleop.cpp.o
[  4%] Building CXX object crazyflie_ros/crazyflie_controller/CMakeFiles/crazyflie_controller.dir/src/controller.cpp.o
[  4%] Building CXX object crazyflie_ros/crazyflie_cpp/CMakeFiles/crazyflie_cpp.dir/src/Crazyradio.cpp.o
[  4%] Building CXX object crazyflie_ros/crazyflie_cpp/CMakeFiles/crazyflie_cpp.dir/src/CrazyflieUSB.cpp.o
[  4%] Built target _crazyflie_driver_generate_messages_check_deps_TrajectoryPolynomialPiece
[  4%] Built target _bitcraze_lps_estimator_generate_messages_check_deps_RangeArray
[  4%] Built target _crazyflie_driver_generate_messages_check_deps_SetGroupMask
[  5%] Building CXX object crazyflie_ros/crazyflie_cpp/CMakeFiles/crazyflie_cpp.dir/src/Crazyflie.cpp.o
[  6%] Building CXX object crazyflie_ros/crazyflie_cpp/CMakeFiles/crazyflie_cpp.dir/src/crtp.cpp.o
[  6%] Built target _crazyflie_driver_generate_messages_check_deps_LogBlock
[  6%] Built target _crazyflie_driver_generate_messages_check_deps_VelocityWorld
[  6%] Built target _crazyflie_driver_generate_messages_check_deps_Hover
[  6%] Building CXX object crazyflie_ros/crazyflie_cpp/CMakeFiles/crazyflie_cpp.dir/src/ITransport.cpp.o
[  6%] Built target _crazyflie_driver_generate_messages_check_deps_RemoveCrazyflie
[  6%] Built target _crazyflie_driver_generate_messages_check_deps_FullState
[  6%] Built target _crazyflie_driver_generate_messages_check_deps_UploadTrajectory
[  6%] Built target _crazyflie_driver_generate_messages_check_deps_UpdateParams
[  6%] Built target _crazyflie_driver_generate_messages_check_deps_Land
[  6%] Built target _crazyflie_driver_generate_messages_check_deps_crtpPacket
/home/robocup/crazyflie/ros/crazyflie_original_catkin_ws/src/crazyflie_ros/crazyflie_cpp/src/USBDevice.cpp: In constructor ‘USBDevice::USBDevice(uint16_t, uint16_t)’:
/home/robocup/crazyflie/ros/crazyflie_original_catkin_ws/src/crazyflie_ros/crazyflie_cpp/src/USBDevice.cpp:19:58: error: ‘libusb_error_name’ was not declared in this scope
         throw std::runtime_error(libusb_error_name(result));
                                                          ^
/home/robocup/crazyflie/ros/crazyflie_original_catkin_ws/src/crazyflie_ros/crazyflie_cpp/src/USBDevice.cpp: In static member function ‘static uint32_t USBDevice::numDevices(uint16_t, uint16_t)’:
/home/robocup/crazyflie/ros/crazyflie_original_catkin_ws/src/crazyflie_ros/crazyflie_cpp/src/USBDevice.cpp:44:58: error: ‘libusb_error_name’ was not declared in this scope
         throw std::runtime_error(libusb_error_name(result));
                                                          ^
/home/robocup/crazyflie/ros/crazyflie_original_catkin_ws/src/crazyflie_ros/crazyflie_cpp/src/USBDevice.cpp:62:59: error: ‘libusb_error_name’ was not declared in this scope
             throw std::runtime_error(libusb_error_name(err));
                                                           ^
/home/robocup/crazyflie/ros/crazyflie_original_catkin_ws/src/crazyflie_ros/crazyflie_cpp/src/USBDevice.cpp: In member function ‘void USBDevice::open(uint32_t)’:
/home/robocup/crazyflie/ros/crazyflie_original_catkin_ws/src/crazyflie_ros/crazyflie_cpp/src/USBDevice.cpp:96:59: error: ‘libusb_error_name’ was not declared in this scope
             throw std::runtime_error(libusb_error_name(err));
                                                           ^
/home/robocup/crazyflie/ros/crazyflie_original_catkin_ws/src/crazyflie_ros/crazyflie_cpp/src/USBDevice.cpp:111:59: error: ‘libusb_error_name’ was not declared in this scope
             throw std::runtime_error(libusb_error_name(err));
                                                           ^
/home/robocup/crazyflie/ros/crazyflie_original_catkin_ws/src/crazyflie_ros/crazyflie_cpp/src/USBDevice.cpp:116:59: error: ‘libusb_error_name’ was not declared in this scope
             throw std::runtime_error(libusb_error_name(err));
                                                           ^
/home/robocup/crazyflie/ros/crazyflie_original_catkin_ws/src/crazyflie_ros/crazyflie_cpp/src/USBDevice.cpp:129:59: error: ‘libusb_error_name’ was not declared in this scope
[... cut ...]
[ 77%] Linking CXX executable /home/robocup/crazyflie/ros/crazyflie_original_catkin_ws/devel/lib/crazyflie_demo/quadrotor_teleop
[ 77%] Built target quadrotor_teleop
[ 77%] Linking CXX executable /home/robocup/crazyflie/ros/crazyflie_original_catkin_ws/devel/lib/crazyflie_controller/crazyflie_controller
[ 77%] Built target crazyflie_controller
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j12 -l12" failed
robocup@devpcX:~/crazyflie/ros/crazyflie_original_catkin_ws$

whoenig, 2020-02-19:
It is still pulling the wrong libusb version I fear. Make sure that there is no libusb.h in your include path from the older 0.1 version.

PranjaLBiswas27, 2020-02-19:
For some reason I was not able to get the default version of gcc (gcc (Ubuntu 5.4.0-6ubuntu1~16.04.12) 5.4.0 20160609) working that is shipped with Ubuntu 16.04 . So I resintalled Ubuntu 16.04 and now I can catkin_make without the errors. But, what I would like to confirm is this repository only compatible with gcc-5.4? Cause I require gcc 4.7 for other projects.

whoenig, 2020-02-21:
I don't think the project uses any features that are not available in gcc 4.7. Your error message seem more related to wrong include paths. It is also possible to have multiple compilers installed and force CMake/Catkin_make to use a specific compiler version if needed.

PranjaLBiswas27, 2020-03-13:
Hi, I have stumbled upon this issue again. I have not yet made these edits (https://github.com/whoenig/crazyflie_ros/issues/168#issuecomment-586945888), since I was able to run it on fresh install of Ubuntu 16.04. So I suppose the issue is something else. At the moment I don´t have libusb-0.1 but libusb-1.0 only, and also I only have default gcc and g++ i.e 5.4 installed currently.

I don´t see what is causing the issue. Could you help me with this?