whoenig/crazyflie_ros #170: Execute test_high_level.py in launch file

Allenhanbo, 2020-02-19:
Hi all, I trying to execute test_high_level.py and rosbag command in launch file together after I roslaunch external_position_vicon.launch. I write a test_high_level.launch file like this:

<?xml version="1.0"?>
<launch>

  <node name="move" pkg="crazyflie_demo" type="test_high_level.py" />

</launch>

But when I launch this file after launching external_position_vicon.launch: the error occurs:

process[move-1]: started with pid [2688]
Traceback (most recent call last):
  File "/home/hc/cf_ws/src/crazyflie_ros/crazyflie_demo/scripts/test_high_level.py", line 29, in <module>
    traj1.loadcsv("takeoff.csv")
  File "/home/hc/cf_ws/src/crazyflie_ros/crazyflie_demo/scripts/uav_trajectory.py", line 93, in loadcsv
    data = np.loadtxt(filename, delimiter=",", skiprows=1, usecols=range(33))
  File "/usr/local/lib/python2.7/dist-packages/numpy/lib/npyio.py", line 962, in loadtxt
    fh = np.lib._datasource.open(fname, 'rt', encoding=encoding)
  File "/usr/local/lib/python2.7/dist-packages/numpy/lib/_datasource.py", line 266, in open
    return ds.open(path, mode, encoding=encoding, newline=newline)
  File "/usr/local/lib/python2.7/dist-packages/numpy/lib/_datasource.py", line 624, in open
    raise IOError("%s not found." % path)
IOError: takeoff.csv not found.

If I run python test_high_level.py. everything goes well.

whoenig, 2020-02-19:
If you run a launch file, the current working directory is different. You can, e.g., change the file path to an absolute path to avoid this error.

Allenhanbo, 2020-02-19:
Hi Whoenig, thanks for your help. I succeed by using the absolute path to execute test_high_level.py in launch file. For my next purpose, I would like to add a rosbag node in this launch file. Like this:

<?xml version="1.0"?>
<launch>

  <node name="move" pkg="crazyflie_demo" type="test_high_level.py" />

  <node pkg="rosbag" type="record" name="rosbag_record_test"
          args="record -O figure8.bag /home/hc/cf_ws/src/crazyflie_ros/crazyflie_demo/launch/figure8.bag /vicon/cf2/cf2"/>    

</launch>

But it can not work and the error is:
[rosbag/rosbag_record_test-2] process has died [pid 24282, exit code -6, cmd /opt/ros/kinetic/lib/rosbag/record record -O figure8.bag /home/hc/cf_ws/src/crazyflie_ros/crazyflie_demo/launch/figure8.bag /vicon/cf2/cf2 __name:=rosbag_record_test __log:=/home/hc/.ros/log/d733985a-5331-11ea-9c06-680715cf7bd2/rosbag-rosbag_record_test-2.log].
log file: /home/hc/.ros/log/d733985a-5331-11ea-9c06-680715cf7bd2/rosbag-rosbag_record_test-2*.log

When I only execute a rosbag node in launch file, it can work well:
<?xml version="1.0"?>
<launch>

  <node pkg="rosbag" type="record" name="rosbag_record_diag"
        args="record -O /home/hc/cf_ws/src/crazyflie_ros/crazyflie_demo/launch/figure8.bag /vicon/cf2/cf2"/>    

</launch>

Allenhanbo, 2020-02-19:
Hi Whoenig, thanks for your help. I succeed by using the absolute path to execute test_high_level.py in launch file. For my next purpose, I would like to add a rosbag node in this launch file. Like this:

But it can not work and the error is:
[rosbag/rosbag_record_test-2] process has died [pid 24282, exit code -6, cmd /opt/ros/kinetic/lib/rosbag/record record -O figure8.bag /home/hc/cf_ws/src/crazyflie_ros/crazyflie_demo/launch/figure8.bag /vicon/cf2/cf2 __name:=rosbag_record_test __log:=/home/hc/.ros/log/d733985a-5331-11ea-9c06-680715cf7bd2/rosbag-rosbag_record_test-2.log].
log file: /home/hc/.ros/log/d733985a-5331-11ea-9c06-680715cf7bd2/rosbag-rosbag_record_test-2*.log

When I only execute a rosbag node in launch file, it can work well:
<?xml version="1.0"?>
<launch>

  <node pkg="rosbag" type="record" name="rosbag_record_diag"
        args="record -O /home/hc/cf_ws/src/crazyflie_ros/crazyflie_demo/launch/figure8.bag /vicon/cf2/cf2"/>    

</launch>

Allenhanbo, 2020-02-19:
Hi, whoenig, thanks for your help. I succeed launch file by changing file path to an absolute path. For my next purpose, I want add a rosbag node into launch file with test_high_level.py node together. The launch file likes this:
<launch>

  <node name="move" pkg="crazyflie_demo" type="test_high_level.py" />

  <node pkg="rosbag" type="record" name="rosbag_record_test"
          args="record -O figure8.bag /home/hc/cf_ws/src/crazyflie_ros/crazyflie_demo/launch/figure8.bag /vicon/cf2/cf2"/>    

</launch>

But the error occur:
terminate called after throwing an instance of 'ros::InvalidNameException'
  what():  Character [.] at element [62] is not valid in Graph Resource Name [/home/hc/cf_ws/src/crazyflie_ros/crazyflie_demo/launch/figure8.bag].  Valid characters are a-z, A-Z, 0-9, / and _.
[rosbag/rosbag_record_test-2] process has died [pid 24282, exit code -6, cmd /opt/ros/kinetic/lib/rosbag/record record -O figure8.bag /home/hc/cf_ws/src/crazyflie_ros/crazyflie_demo/launch/figure8.bag /vicon/cf2/cf2 __name:=rosbag_record_test __log:=/home/hc/.ros/log/d733985a-5331-11ea-9c06-680715cf7bd2/rosbag-rosbag_record_test-2.log].
log file: /home/hc/.ros/log/d733985a-5331-11ea-9c06-680715cf7bd2/rosbag-rosbag_record_test-2*.log

If I only execute rosbag node in launch, it works well:
"<launch>
<node pkg="rosbag" type="record" name="rosbag_record_diag"
        args="record -O /home/hc/cf_ws/src/crazyflie_ros/crazyflie_demo/launch/figure8.bag /vicon/cf2/cf2"/>    
</launch> "

Allenhanbo, 2020-02-19:
Hi, whoenig, thanks for your help. I succeed launch file by changing file path to an absolute path. For my next purpose, I want add a rosbag node into launch file with test_high_level.py node together. The launch file likes this:

launch

  <node name="move" pkg="crazyflie_demo" type="test_high_level.py" 

  <node pkg="rosbag" type="record" name="rosbag_record_test"
          args="record -O figure8.bag /home/hc/cf_ws/src/crazyflie_ros/crazyflie_demo/launch/figure8.bag [/vicon/cf2/cf2"

<launch

But the error occur:
terminate called after throwing an instance of 'ros::InvalidNameException'
  what():  Character [.] at element [62] is not valid in Graph Resource Name [/home/hc/cf_ws/src/crazyflie_ros/crazyflie_demo/launch/figure8.bag].  Valid characters are a-z, A-Z, 0-9, / and _.
[rosbag/rosbag_record_test-2] process has died [pid 24282, exit code -6, cmd /opt/ros/kinetic/lib/rosbag/record record -O figure8.bag /home/hc/cf_ws/src/crazyflie_ros/crazyflie_demo/launch/figure8.bag /vicon/cf2/cf2 __name:=rosbag_record_test __log:=/home/hc/.ros/log/d733985a-5331-11ea-9c06-680715cf7bd2/rosbag-rosbag_record_test-2.log].
log file: /home/hc/.ros/log/d733985a-5331-11ea-9c06-680715cf7bd2/rosbag-rosbag_record_test-2*.log

If I only execute rosbag node in launch, it works well:
<launch
<node pkg="rosbag" type="record" name="rosbag_record_diag"
        args="record -O /home/hc/cf_ws/src/crazyflie_ros/crazyflie_demo/launch/figure8.bag /vicon/cf2/cf2"/
</launch

whoenig, 2020-02-21:
Looks to me like you had a typo (additional [) in your first launch file?