whoenig/crazyflie_ros #171: Publishing the commanded angle rates by ROS

OmarShadeed, 2020-02-19:
Hi everyone, 
I’m trying to publish the commanded angle rates from the onboard attitude controller. I see it already logged in the controller_pid.c file, however, I don’t know how to publish it in the  crazyflie_server.cpp file 
I really appreciate your help! 
Thanks,

whoenig, 2020-02-21:
I am not entirely sure what you want to do. Do you want to log the angle rates from the onboard attitude controller while using the onboard position controller? Or do you want to log the commands that you send from the PC when you fly using a joystick or external position controller?

OmarShadeed, 2020-02-21:
I am using an off board position controller and the commands i send the CF are thrust, pitch (deg), roll (deg), and yaw rate (deg/sec). I know that the on board attitude controller has two layers; angles controller and angles rates controller. What I wanna do is get the angle rates commands which are the output of the angles controller. At the end my aim is to plot the roll rate and pitch rate (commanded Vs actual).