whoenig/crazyflie_ros #174: Flying with LPS  and differenz to Hover.py

stikkx, 2020-02-28:
Hello whoenig, 
could you tell me the main difference between this script and the Hover.py ? And how can i use this with the LPS instead of vicon ? 

My problem is that the CF crashes short after takeoff, but with the Hover.py they fly very nice. 

Thanks for your help and the work on crazyflie!

whoenig, 2020-02-28:
Which script is causing a crash for you?

With the LPS everything runs on-board, so there is no need to send position/pose information from the basestation. Thus, only the launch file should differ, but all the scripts should work with any tracking system.

stikkx, 2020-03-02:
All crazyswarm scripts run unstable. Often a part of the CF simply crashes during the flight. For example at https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/scripts/niceHover.py . The CF swings during the flight

Is it possible to use the test_high_level.py (or crazyflie.py) without using vicon ? It is expected as third parameter.

Is it correct that with the hover.py the CFs only start if a maximum of 3 CFs are used ?

stikkx, 2020-03-02:
i read more about crazy_ros and ros the topics. If i am right the third agument is only the cf topic ? So i can use /cf1 for the first cf ?

stikkx, 2020-03-09:
So i reconfigure the PID controller from Crazyswarm, now it is much better. But only on ToDA2.

closed 2020-03-09 by stikkx