whoenig/crazyflie_ros #176: Unable to echo /crazyflie/imu data

kashishdhal, 2020-04-02:
Hello @whoenig,

I am able to successfully launch hover_vicon.launch (please see below) as specified on your github. However, when I echo /crazyflie/imu or battery or pressure, I don't see any data. I made sure to source devel/setup.bash

Please let me know if you would be able to help me with the same. Note, I don't have a joystick but I don't know if that will make a difference in imu or pressure topics.

<pre><font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~</b></font>$ roslaunch crazyflie_demo hover_vicon.launch x:=0 y:=0 z:=0.5
... logging to /home/gnclab/.ros/log/40dd63d2-72fc-11ea-a41b-1c697a0794e3/roslaunch-system76-pc-16294.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
<font color="#CC0000">WARNING: disk usage in log directory [/home/gnclab/.ros/log] is over 1GB.</font>
<font color="#CC0000">It&apos;s recommended that you use the &apos;rosclean&apos; command.</font>

/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/loader.py:412: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  data = yaml.load(text)
<font color="#FCE94F">xacro.py is deprecated; please use xacro instead</font>
<b>started roslaunch server http://system76-pc:34579/</b>

SUMMARY
========

PARAMETERS
 * /crazyflie/controller/PIDs/X/integratorMax: 0.1
 * /crazyflie/controller/PIDs/X/integratorMin: -0.1
 * /crazyflie/controller/PIDs/X/kd: 20.0
 * /crazyflie/controller/PIDs/X/ki: 2.0
 * /crazyflie/controller/PIDs/X/kp: 40.0
 * /crazyflie/controller/PIDs/X/maxOutput: 10.0
 * /crazyflie/controller/PIDs/X/minOutput: -10.0
 * /crazyflie/controller/PIDs/Y/integratorMax: 0.1
 * /crazyflie/controller/PIDs/Y/integratorMin: -0.1
 * /crazyflie/controller/PIDs/Y/kd: -20.0
 * /crazyflie/controller/PIDs/Y/ki: -2.0
 * /crazyflie/controller/PIDs/Y/kp: -40.0
 * /crazyflie/controller/PIDs/Y/maxOutput: 10.0
 * /crazyflie/controller/PIDs/Y/minOutput: -10.0
 * /crazyflie/controller/PIDs/Yaw/integratorMax: 0.0
 * /crazyflie/controller/PIDs/Yaw/integratorMin: 0.0
 * /crazyflie/controller/PIDs/Yaw/kd: -20.0
 * /crazyflie/controller/PIDs/Yaw/ki: 0.0
 * /crazyflie/controller/PIDs/Yaw/kp: -200.0
 * /crazyflie/controller/PIDs/Yaw/maxOutput: 200.0
 * /crazyflie/controller/PIDs/Yaw/minOutput: -200.0
 * /crazyflie/controller/PIDs/Z/integratorMax: 1000.0
 * /crazyflie/controller/PIDs/Z/integratorMin: -1000.0
 * /crazyflie/controller/PIDs/Z/kd: 6000.0
 * /crazyflie/controller/PIDs/Z/ki: 3500.0
 * /crazyflie/controller/PIDs/Z/kp: 5000.0
 * /crazyflie/controller/PIDs/Z/maxOutput: 60000.0
 * /crazyflie/controller/PIDs/Z/minOutput: 10000.0
 * /crazyflie/controller/frame: /vicon/gncQuad/gn...
 * /crazyflie/controller/worldFrame: world
 * /crazyflie/crazyflie_add/enable_logging: False
 * /crazyflie/crazyflie_add/enable_logging_battery: True
 * /crazyflie/crazyflie_add/enable_logging_imu: True
 * /crazyflie/crazyflie_add/enable_logging_magnetic_field: True
 * /crazyflie/crazyflie_add/enable_logging_packets: True
 * /crazyflie/crazyflie_add/enable_logging_pose: False
 * /crazyflie/crazyflie_add/enable_logging_pressure: True
 * /crazyflie/crazyflie_add/enable_logging_temperature: True
 * /crazyflie/crazyflie_add/pitch_trim: 0
 * /crazyflie/crazyflie_add/roll_trim: 0
 * /crazyflie/crazyflie_add/tf_prefix: crazyflie
 * /crazyflie/crazyflie_add/uri: radio://0/80/2M
 * /crazyflie/crazyflie_add/use_ros_time: True
 * /crazyflie/joy/dev: /dev/input/js0
 * /crazyflie/joystick_controller/use_crazyflie_controller: True
 * /crazyflie/pose/name: goal
 * /crazyflie/pose/rate: 30
 * /crazyflie/pose/x: 0
 * /crazyflie/pose/y: 0
 * /crazyflie/pose/z: 0.5
 * /robot_description: &lt;?xml version=&quot;1....
 * /rosdistro: melodic
 * /rosversion: 1.14.3
 * /vicon/datastream_hostport: 192.168.10.1:801
 * /vicon/stream_mode: ClientPull
 * /vicon/tf_ref_frame_id: /world

NODES
  /crazyflie/
    baselink_broadcaster (tf/static_transform_publisher)
    controller (crazyflie_controller/crazyflie_controller)
    crazyflie_add (crazyflie_driver/crazyflie_add)
    joy (joy/joy_node)
    joystick_controller (crazyflie_demo/controller.py)
    pose (crazyflie_demo/publish_pose.py)
  /
    crazyflie_server (crazyflie_driver/crazyflie_server)
    rviz (rviz/rviz)
    vicon (vicon_bridge/vicon_bridge)

auto-starting new master
<b>process[master]: started with pid [16307]</b>
<b>ROS_MASTER_URI=http://localhost:11311</b>

<b>setting /run_id to 40dd63d2-72fc-11ea-a41b-1c697a0794e3</b>
<b>process[rosout-1]: started with pid [16318]</b>
started core service [/rosout]
<b>process[crazyflie_server-2]: started with pid [16325]</b>
<b>process[crazyflie/crazyflie_add-3]: started with pid [16326]</b>
<b>process[crazyflie/joy-4]: started with pid [16332]</b>
<b>process[crazyflie/joystick_controller-5]: started with pid [16338]</b>
[ INFO] [1585623834.224473439]: wait_for_service /add_crazyflie
[ INFO] [1585623834.225827631]: found /add_crazyflie
[ INFO] [1585623834.228366335]: Adding radio://0/80/2M as crazyflie with trim(0.000000, 0.000000). Logging: 1, Parameters: 0, Use ROS time: 1
<b>process[crazyflie/controller-6]: started with pid [16344]</b>
<font color="#CC0000">[ERROR] [1585623834.230447030]: Couldn&apos;t open joystick /dev/input/js0. Will retry every second.</font>
<b>process[crazyflie/pose-7]: started with pid [16347]</b>
<b>process[crazyflie/baselink_broadcaster-8]: started with pid [16350]</b>
<b>process[vicon-9]: started with pid [16355]</b>
<b>process[rviz-10]: started with pid [16362]</b>
[ INFO] [1585623834.266429805]: Connecting to Vicon DataStream SDK at 192.168.10.1:801 ...
[ INFO] [1585623834.267349363]: .
[ INFO] [1585623834.268865711]: CF Console: SYS: ----------------------------
[ INFO] [1585623834.271229262]: CF Console: SYS: Crazyflie 2.1 is up and running!
[ INFO] [1585623834.273392815]: CF Console: SYS: Production release 2020.02
[ INFO] [1585623834.276974643]: CF Console: SYS: I am 0x20383347344D500800250035 and I have 1024KB of flash!
[ INFO] [1585623834.278490793]: CF Console: CFGBLK: v1, verification [OK]
[ INFO] [1585623834.279885338]: CF Console: DECK_CORE: 0 deck(s) found
[ INFO] [1585623834.282406961]: CF Console: IMU: BMI088 Gyro I2C connection [OK].
[ INFO] [1585623834.284764807]: CF Console: IMU: BMI088 Accel I2C connection [OK]
[ INFO] [1585623834.287125852]: CF Console: IMU: BMP388 I2C connection [OK]
[ INFO] [1585623834.289723674]: CF Console: ESTIMATOR: Using Complementary (1) estimator
[ INFO] [1585623834.292115265]: CF Console: CONTROLLER: Using PID (1) controller
[ INFO] [1585623834.294487187]: CF Console: MTR-DRV: Using brushed motor driver
[ INFO] [1585623834.295911004]: CF Console: EEPROM: I2C connection [OK].
[ INFO] [1585623834.307399908]: CF Console: STAB: Wait for sensor calibration...
[ INFO] [1585623834.308824721]: CF Console: SYS: Free heap: 8216 bytes
[ INFO] [1585623834.310125405]: CF Console: STAB: Ready to fly.
[ INFO] [1585623834.311002275]: Requesting parameters...
[ INFO] [1585623834.367665112]: Found variables in cache.
<b>[crazyflie/crazyflie_add-3] process has finished cleanly</b>
<b>log file: /home/gnclab/.ros/log/40dd63d2-72fc-11ea-a41b-1c697a0794e3/crazyflie-crazyflie_add-3*.log</b>
[ INFO] [1585623834.663208415]: Requesting memories...
[ INFO] [1585623834.700935269]: Memories: 8
[ INFO] [1585623834.739220622]: Ready...
[ INFO] [1585623834.739249062]: Elapsed: 0.475502 s
[INFO] [1585623834.818719]: found update_params service
[INFO] [1585623834.819596]: waiting for emergency service
[INFO] [1585623834.821553]: found emergency service
[INFO] [1585623834.822548]: waiting for land service
[INFO] [1585623834.824954]: found land service
[INFO] [1585623834.825984]: waiting for takeoff service
[INFO] [1585623834.828023]: found takeoff service
[ INFO] [1585623835.267563366]: ... connected!
[ INFO] [1585623835.267649953]: Setting Stream Mode to ClientPull: Success
[ INFO] [1585623835.267715562]: Axis Mapping: X-Forward Y-Left Z-Up
[ INFO] [1585623835.267798269]: Version: 1.3.0
[ INFO] [1585623835.267864272]: setting up grab_vicon_pose service server ... 
[ INFO] [1585623835.269357908]: setting up segment calibration service server ... 
<font color="#C4A000">[ WARN] [1585623835.273258602]: grab frame returned false</font>
<font color="#C4A000">[ WARN] [1585623835.277218383]: gncQuad occluded, not publishing... </font>
[ INFO] [1585623835.288111332]: creating new object gncQuad/gncQuad ...
<font color="#C4A000">[ WARN] [1585623835.290802863]: unable to load zero pose for gncQuad/gncQuad</font>
[ INFO] [1585623835.290888505]: ... done, advertised as &quot; vicon/gncQuad/gncQuad&quot;</pre>

whoenig, 2020-04-02:
It looks like `enable_logging` is set to False in the launch file, which would prevent it from creating any ROS topics. You do not need a joystick for operation, it is just convenient for emergency shutoff etc. I hope that helps!

kashishdhal, 2020-04-03:
Hello @whoenig ,

Thanks for the reply. When I do rostopic list, I am able to see all the topics but rostopic echo does not return any result.

I modified the launch file as per your suggestion and still could not echo the topics. Please let me know what shoudl I do next?

Here is the output from the launch after modification of the launch file:

<pre><font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~</b></font>$ roslaunch crazyflie_demo hover_vicon.launch 
... logging to /home/gnclab/.ros/log/19989308-755b-11ea-a41b-1c697a0794e3/roslaunch-system76-pc-9974.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
<font color="#CC0000">WARNING: disk usage in log directory [/home/gnclab/.ros/log] is over 1GB.</font>
<font color="#CC0000">It&apos;s recommended that you use the &apos;rosclean&apos; command.</font>

/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/loader.py:412: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  data = yaml.load(text)
<font color="#FCE94F">xacro.py is deprecated; please use xacro instead</font>
<b>started roslaunch server http://system76-pc:37301/</b>

SUMMARY
========

PARAMETERS
 * /crazyflie/controller/PIDs/X/integratorMax: 0.1
 * /crazyflie/controller/PIDs/X/integratorMin: -0.1
 * /crazyflie/controller/PIDs/X/kd: 20.0
 * /crazyflie/controller/PIDs/X/ki: 2.0
 * /crazyflie/controller/PIDs/X/kp: 40.0
 * /crazyflie/controller/PIDs/X/maxOutput: 10.0
 * /crazyflie/controller/PIDs/X/minOutput: -10.0
 * /crazyflie/controller/PIDs/Y/integratorMax: 0.1
 * /crazyflie/controller/PIDs/Y/integratorMin: -0.1
 * /crazyflie/controller/PIDs/Y/kd: -20.0
 * /crazyflie/controller/PIDs/Y/ki: -2.0
 * /crazyflie/controller/PIDs/Y/kp: -40.0
 * /crazyflie/controller/PIDs/Y/maxOutput: 10.0
 * /crazyflie/controller/PIDs/Y/minOutput: -10.0
 * /crazyflie/controller/PIDs/Yaw/integratorMax: 0.0
 * /crazyflie/controller/PIDs/Yaw/integratorMin: 0.0
 * /crazyflie/controller/PIDs/Yaw/kd: -20.0
 * /crazyflie/controller/PIDs/Yaw/ki: 0.0
 * /crazyflie/controller/PIDs/Yaw/kp: -200.0
 * /crazyflie/controller/PIDs/Yaw/maxOutput: 200.0
 * /crazyflie/controller/PIDs/Yaw/minOutput: -200.0
 * /crazyflie/controller/PIDs/Z/integratorMax: 1000.0
 * /crazyflie/controller/PIDs/Z/integratorMin: -1000.0
 * /crazyflie/controller/PIDs/Z/kd: 6000.0
 * /crazyflie/controller/PIDs/Z/ki: 3500.0
 * /crazyflie/controller/PIDs/Z/kp: 5000.0
 * /crazyflie/controller/PIDs/Z/maxOutput: 60000.0
 * /crazyflie/controller/PIDs/Z/minOutput: 10000.0
 * /crazyflie/controller/frame: /vicon/gncQuad/gn...
 * /crazyflie/controller/worldFrame: world
 * /crazyflie/crazyflie_add/enable_logging: True
 * /crazyflie/crazyflie_add/enable_logging_battery: True
 * /crazyflie/crazyflie_add/enable_logging_imu: True
 * /crazyflie/crazyflie_add/enable_logging_magnetic_field: True
 * /crazyflie/crazyflie_add/enable_logging_packets: True
 * /crazyflie/crazyflie_add/enable_logging_pose: False
 * /crazyflie/crazyflie_add/enable_logging_pressure: True
 * /crazyflie/crazyflie_add/enable_logging_temperature: True
 * /crazyflie/crazyflie_add/pitch_trim: 0
 * /crazyflie/crazyflie_add/roll_trim: 0
 * /crazyflie/crazyflie_add/tf_prefix: crazyflie
 * /crazyflie/crazyflie_add/uri: radio://0/80/2M
 * /crazyflie/crazyflie_add/use_ros_time: True
 * /crazyflie/joy/dev: /dev/input/js0
 * /crazyflie/joystick_controller/use_crazyflie_controller: True
 * /crazyflie/pose/name: goal
[... 56 lines cut ...]
[INFO] [1585884473.464464]: found emergency service
[INFO] [1585884473.465377]: waiting for land service
<font color="#C4A000">[WARN] [1585884473.467375]: wait_for_service(/crazyflie/land): failed to contact, will keep trying</font>
<font color="#CC0000">[crazyflie_server-2] process has died [pid 10008, exit code -6, cmd /home/gnclab/catkin_ws/devel/lib/crazyflie_driver/crazyflie_server __name:=crazyflie_server __log:=/home/gnclab/.ros/log/19989308-755b-11ea-a41b-1c697a0794e3/crazyflie_server-2.log].</font>
<font color="#CC0000">log file: /home/gnclab/.ros/log/19989308-755b-11ea-a41b-1c697a0794e3/crazyflie_server-2*.log</font>
[ INFO] [1585884473.914310212]: ... connected!
[ INFO] [1585884473.914422459]: Setting Stream Mode to ClientPull: Success
[ INFO] [1585884473.914521716]: Axis Mapping: X-Forward Y-Left Z-Up
[ INFO] [1585884473.914640461]: Version: 1.3.0
[ INFO] [1585884473.914767153]: setting up grab_vicon_pose service server ... 
[ INFO] [1585884473.916741515]: setting up segment calibration service server ... 
<font color="#C4A000">[ WARN] [1585884473.919794355]: grab frame returned false</font>
<font color="#C4A000">[ WARN] [1585884473.922739034]: aslRover1 occluded, not publishing... </font>
<font color="#C4A000">[ WARN] [1585884473.922839279]: gncQuad occluded, not publishing... </font>
[ INFO] [1585884473.933036509]: creating new object aslRover1/aslRover1 ...
[ INFO] [1585884473.933252202]: creating new object gncQuad/gncQuad ...
<font color="#C4A000">[ WARN] [1585884473.936435123]: unable to load zero pose for aslRover1/aslRover1</font>
[ INFO] [1585884473.936551007]: ... done, advertised as &quot; vicon/aslRover1/aslRover1&quot; 
<font color="#C4A000">[ WARN] [1585884473.938421950]: unable to load zero pose for gncQuad/gncQuad</font>
[ INFO] [1585884473.938513883]: ... done, advertised as &quot; vicon/gncQuad/gncQuad&quot; 
[INFO] [1585884474.074684]: wait_for_service(/crazyflie/land): finally were able to contact [rosrpc://system76-pc:44611]
[INFO] [1585884474.077905]: found land service
[INFO] [1585884474.080997]: waiting for takeoff service
[INFO] [1585884474.087241]: found takeoff service
</pre>

kashishdhal, 2020-04-03:
Hello @whoenig,

I just noticed in my output **[crazyflie_server-2] process has died** which means it's not able to find radio now (i don't know why?)

Scan command line returns the following:

<pre><font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~</b></font>$ rosrun crazyflie_tools scan
radio://0/80/2M
</pre>

In the cfclient I am able to connect to the following address: 0xE7E7E7E7E7

My launch file looks like the following:

 <arg name="uri" default="radio://0/80/2M/0xE7E7E7E7E7" />
  <arg name="frame" default="/vicon/gncQuad/gncQuad" />
  <arg name="joy_dev" default="/dev/input/js0" />
  <arg name="x" default="0" />
  <arg name="y" default="0" />
  <arg name="z" default="0.5" />

Please  let me know where I am going wrong?

whoenig, 2020-04-16:
The process seemed to have died because of `Could not find gyro.x in log toc!`. Which firmware version are you using?

kashishdhal, 2020-04-16:
Hello @whoenig ,

Thanks for the reply. I am using the latest firmware, I guess it was 2020.02.

I also noticed that whenever my IMU parameter is set to True, it gives error in gyro.x.

Please help!

whoenig, 2020-04-16:
This logging variable still exists in the latest firmware (see https://github.com/bitcraze/crazyflie-firmware/blob/master/src/modules/src/stabilizer.c#L613-L617), so the issue must come from somewhere else.

1. Try deleting the TOC cache files "~/.ros/log*.csv"
2. Re-run and post the the output of the newly created cache file. Alternatively, you can use `rosrun crazyflie_tools listLogVariables --uri radio://0/80/2M/0xE7E7E7E7E7` and post the output of that command.

kashishdhal, 2020-04-16:
Hello @whoenig,

Here is the output after deleting the log file:

<pre><font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~</b></font>$ roslaunch crazyflie_demo hover_vicon.launch 
... logging to /home/gnclab/.ros/log/cc27d664-8014-11ea-a41b-1c697a0794e3/roslaunch-system76-pc-22919.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
<font color="#CC0000">WARNING: disk usage in log directory [/home/gnclab/.ros/log] is over 1GB.</font>
<font color="#CC0000">It&apos;s recommended that you use the &apos;rosclean&apos; command.</font>

/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/loader.py:412: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  data = yaml.load(text)
<font color="#FCE94F">xacro.py is deprecated; please use xacro instead</font>
<b>started roslaunch server http://system76-pc:40113/</b>

SUMMARY
========

PARAMETERS
 * /crazyflie/controller/PIDs/X/integratorMax: 0.1
 * /crazyflie/controller/PIDs/X/integratorMin: -0.1
 * /crazyflie/controller/PIDs/X/kd: 20.0
 * /crazyflie/controller/PIDs/X/ki: 2.0
 * /crazyflie/controller/PIDs/X/kp: 40.0
 * /crazyflie/controller/PIDs/X/maxOutput: 10.0
 * /crazyflie/controller/PIDs/X/minOutput: -10.0
 * /crazyflie/controller/PIDs/Y/integratorMax: 0.1
 * /crazyflie/controller/PIDs/Y/integratorMin: -0.1
 * /crazyflie/controller/PIDs/Y/kd: -20.0
 * /crazyflie/controller/PIDs/Y/ki: -2.0
 * /crazyflie/controller/PIDs/Y/kp: -40.0
 * /crazyflie/controller/PIDs/Y/maxOutput: 10.0
 * /crazyflie/controller/PIDs/Y/minOutput: -10.0
 * /crazyflie/controller/PIDs/Yaw/integratorMax: 0.0
 * /crazyflie/controller/PIDs/Yaw/integratorMin: 0.0
 * /crazyflie/controller/PIDs/Yaw/kd: -20.0
 * /crazyflie/controller/PIDs/Yaw/ki: 0.0
 * /crazyflie/controller/PIDs/Yaw/kp: -200.0
 * /crazyflie/controller/PIDs/Yaw/maxOutput: 200.0
 * /crazyflie/controller/PIDs/Yaw/minOutput: -200.0
 * /crazyflie/controller/PIDs/Z/integratorMax: 1000.0
 * /crazyflie/controller/PIDs/Z/integratorMin: -1000.0
 * /crazyflie/controller/PIDs/Z/kd: 6000.0
 * /crazyflie/controller/PIDs/Z/ki: 3500.0
 * /crazyflie/controller/PIDs/Z/kp: 5000.0
 * /crazyflie/controller/PIDs/Z/maxOutput: 60000.0
 * /crazyflie/controller/PIDs/Z/minOutput: 10000.0
 * /crazyflie/controller/frame: /vicon/crazyflie/...
 * /crazyflie/controller/worldFrame: world
 * /crazyflie/crazyflie_add/enable_logging: True
 * /crazyflie/crazyflie_add/enable_logging_battery: True
 * /crazyflie/crazyflie_add/enable_logging_imu: True
 * /crazyflie/crazyflie_add/enable_logging_magnetic_field: True
 * /crazyflie/crazyflie_add/enable_logging_packets: True
 * /crazyflie/crazyflie_add/enable_logging_pose: False
 * /crazyflie/crazyflie_add/enable_logging_pressure: True
 * /crazyflie/crazyflie_add/enable_logging_temperature: True
 * /crazyflie/crazyflie_add/pitch_trim: 0
 * /crazyflie/crazyflie_add/roll_trim: 0
 * /crazyflie/crazyflie_add/tf_prefix: crazyflie
 * /crazyflie/crazyflie_add/uri: radio://0/80/2M/0...
 * /crazyflie/crazyflie_add/use_ros_time: True
 * /crazyflie/joy/dev: /dev/input/js0
 * /crazyflie/joystick_controller/use_crazyflie_controller: True
 * /crazyflie/pose/name: goal
 * /crazyflie/pose/rate: 30
 * /crazyflie/pose/x: 0
 * /crazyflie/pose/y: 0
 * /crazyflie/pose/z: 0.5
[... 416 lines cut ...]
tdoa3.cc (float)
tdoa3.tof (uint16)
tdoa3.tdoa (float)
outlierf.bucket0 (int32)
outlierf.bucket1 (int32)
outlierf.bucket2 (int32)
outlierf.bucket3 (int32)
outlierf.bucket4 (int32)
outlierf.accLev (float)
outlierf.errD (float)
motion.motion (uint8)
motion.deltaX (int16)
motion.deltaY (int16)
motion.shutter (uint16)
motion.maxRaw (uint8)
motion.minRaw (uint8)
motion.Rawsum (uint8)
motion.outlierCount (uint8)
oa.front (uint16)
oa.back (uint16)
gyro.z (float)
health.motorVarXM1 (float)
oa.right (uint16)
health.motorVarXM2 (float)
health.motorVarYM2 (float)</pre>

kashishdhal, 2020-04-17:
In another run I got:

<pre>[ INFO] [1587083586.987459511]: Requesting Logging variables...
[ INFO] [1587083587.029726883]: Found variables in cache.
terminate called after throwing an instance of &apos;std::runtime_error&apos;
  what():  Could not find baro.pressure in log toc!
<font color="#CC0000">[crazyflie_server-2] process has died [pid 31766, exit code -6, cmd /home/gnclab/catkin_ws/devel/lib/crazyflie_driver/crazyflie_server __name:=crazyflie_server __log:=/home/gnclab/.ros/log/00ea1e74-8043-11ea-a41b-1c697a0794e3/crazyflie_server-2.log].</font>
<font color="#CC0000">log file: /home/gnclab/.ros/log/00ea1e74-8043-11ea-a41b-1c697a0794e3/crazyflie_server-2*.log</font>
<font color="#C4A000">[WARN] [1587083587.228487]: wait_for_service(/crazyflie/update_params): failed to contact, will keep trying</font>
</pre>

whoenig, 2020-04-17:
My guess is that this is related to the safelink, but I'll need to try to reproduce this. You can disable the safelink [here](https://github.com/whoenig/crazyflie_cpp/blob/master/include/crazyflie_cpp/Crazyflie.h#L15) [followed by `catkin_make` to rebuilt].

kashishdhal, 2020-04-17:
Thanks @whoenig , it worked!

What is the function of this feature that we disabled? Should we worry about it or should not bother for now?

whoenig, 2020-04-17:
Thanks for the update! There are two communication modes: "safelink" and "non-safelink". Both have the exact same feature set, so you don't need to worry. "Safelink" is a bit newer, but apparently not mature yet. I'll roll-back the default version to not use safelink anymore for the time being.

kashishdhal, 2020-04-17:
Okay thanks! I'm glad that it would help others too. 

Also, I was wondering if it is possible that we can add a .c file inside MCU for our custom applications. I just purchased a debugger and Arm JTAG. Please let me know.

closed 2020-04-17 by kashishdhal

whoenig, 2020-04-17:
I disabled safelink by default. For your custom application: Yes that is entirely possible (you don't even need the debugger, although the debugger is nice to find issues in your custom code). You can probably get best help at forum.bitcraze.io on how to add your custom code.

kashishdhal, 2020-04-17:
Oka thanks @whoenig

kashishdhal, 2020-05-18:
Hello @whoenig ,

I am not getting Covariance data in IMU topic, any clue why? Is it disabled somewhere?

Regards,
Kashish Dhal

commit 7094e314 (2020-04-17): Update crazyflie_cpp

Fixes #176.