whoenig/crazyflie_ros #181: Continuous signal with crazyflie_ros

eduardo-del, 2020-05-26:
Hello guys

I would like to know if you can help me with something,
I am working with crazyflie_ws in the crazyflie_ros folder https://github.com/whoenig/crazyflie_ros, specifically there is a file named pose.publish.py https://github.com/whoenig/crazyflie_ro ... sh_pose.py
and this generates the desired points for x, y and z, but these are only constants,
I would like to know if I can enter a variant signal in time as it is a signal sine,

thank you very much.

eduardo-del, 2020-05-27:
I try to do a sine signal with the next pose.publish.py file, I modificad this file in the lines

 `
   t = rospy.get_time()
   x = 0
   y = 0
   z = 0.3*math.sin(2*3.1416*0.05*t) + 0.6
`

`
#!/usr/bin/env python

import rospy
import math
import tf
from geometry_msgs.msg import PoseStamped

if __name__ == '__main__':
    rospy.init_node('publish_pose', anonymous=True)
    worldFrame = rospy.get_param("~worldFrame", "/world")
    name = rospy.get_param("~name")
    r = rospy.get_param("~rate")
    t = rospy.get_time()
    x = 0
    y = 0
    z = 0.3*math.sin(2*3.1416*0.05*t) + 0.6

    msg = PoseStamped()
    msg.header.seq = 0
    msg.header.stamp = rospy.Time.now()
    msg.header.frame_id = worldFrame
    msg.pose.position.x = x
    msg.pose.position.y = y
    msg.pose.position.z = z
    quaternion = tf.transformations.quaternion_from_euler(0, 0, 0)
    msg.pose.orientation.x = quaternion[0]
    msg.pose.orientation.y = quaternion[1]
    msg.pose.orientation.z = quaternion[2]
    msg.pose.orientation.w = quaternion[3]

    pub = rospy.Publisher(name, PoseStamped, queue_size=1)

    while not rospy.is_shutdown():
        msg.header.seq += 1
        msg.header.stamp = rospy.Time.now()
        pub.publish(msg)
	rate.sleep()

`

But really i haven't good result, maybe some one did some like this, can you help me?
thank you

eduardo-del, 2020-05-27:
I see my topic  crazyflie/goal on rosbag and
this only have a constat like 
`z_goal = 0.6`

whoenig, 2020-05-27:
Your approach is correct, but make sure you update your message inside the main while loop (right before `pub.publish(msg)`. 

Please post code as file, or inside the code environment on github (you can use the "Preview" tab to check if what you post gets rendered correctly before commenting).

eduardo-del, 2020-05-27:
Thank you for the answer @whoenig and for the advice :) 
i will tell you if this is working.

I have another quesiton, can you help me?

Do you think that I may the same with demo1.py file  [https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_demo/scripts/demo1.py](url) ?

`#x   ,   y,   z, yaw, sleep`
` [0.0 , 0.0, 0.3*math.sin(2*3.1416*0.05*t), 0, 60]`

whoenig, 2020-05-27:
Sure, you can create an appropriate array using either numpy or Python list comprehension.

eduardo-del, 2020-05-27:
Thank you so much @whoenig 
I will tray all this :)

eduardo-del, 2020-05-27:
HI @whoenig

Again with other question
Can I write a line with the acceleration of the z desired  in crazyflie2.yaml file [https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_controller/config/crazyflie2.yaml](url)

for example in crazyflie2.yaml file

`
  Z:

    kp: 5100.0
    kd: 6900.0
    ki: 3500.0
    g: 9.81 
    ed: -0.3*2*3.1416*0.05*2*3.1416*0.05*math.sin(2*3.1416*0.05*t)
    m: 0.032
    minOutput: 10000.0
    maxOutput: 60000.0
    integratorMin: -1000.0
    integratorMax: 1000.0
`

eduardo-del, 2020-05-28:
Hi @whoenig  I try the next code with publish_pose.py file

`

    while not rospy.is_shutdown():
        msg.header.seq += 1
        msg.header.stamp = rospy.Time.now()
	    msg = PoseStamped()  # add line
	    msg.pose.position.z = z  # add line
        pub.publish(msg)
	rate.sleep()

    
when I add two lines my joystick don't do anything
my crazyflie don't receive commands like crazyflie/cmd_vel 
however my nodes exist

eduardo-del, 2020-05-28:
I checked my crazyflie/cmd_vel and this is cero,
i will check all the lines on my file 
thank you @whoenig

whoenig, 2020-05-28:
* yaml files do not allow equations
* For your code, make sure you actually change the value of local variable `z` in every loop iteration.

In general, I recommend learning Python well first, then learn ROS in simulation only, and finally switch to real hardware.

eduardo-del, 2020-05-28:
Thank you for the answer @whoenig  :)

aviad719, 2020-06-15:
Hii @whoenig .
I am new to Python and to ROS and i want to do a project in ROS using Crazyflie.
can you recommend on relevant tutorials ??

whoenig, 2020-06-17:
In general I recommend first learning Python, then learning ROS in simulation, and afterwards picking up projects with real hardware. For both Python and ROS are good books available, which is perhaps easier/more structured than following tutorials online.